micropython/micropython #12105: Pico: sdcard.py timeout waiting for v2 card

pollo1308, 2023-07-27:
give me this issue with kingston sd card 32 gb somebody help plis

dpgeorge, 2023-07-28:
Please describe your hardware set up, the version of the code you are using, the code that you run, and the output from running.

Flipje1955, 2023-07-28:
@pollo1308: had the same experience with 32GB cards of a variety of brands, using the official driver. Could not figure out what was wrong (am the first one to admit I am not in expert in the domain) but observed the issue only appeared with the 32GB type cards. 
I use 16GB cards since with the same sw and have no issues. Platform rp2040 of various brand boards. MP 1.19 / 1.20 incl intermediates (nightly builds).

pollo1308, 2023-07-31:
my bad the vcc pin of the micro sd conect to vbus pin, not to 3.3v

closed 2023-08-02 by pollo1308

La89688, 2024-10-08:
Same experience. 32GB (Samsung Evo plus) didn't work either 5V and 3.3V. 16GB (Verbatim premium) works with 5V VSYS, but not working with 3V3(OUT) on Pi Pico.

peterhinch, 2024-10-09:
Is the SPI bus shared with other hardware?

La89688, 2024-10-09:
No, on this SPI is only SDcard module.

RYoder97, 2024-10-23:
So, I understand that the issue may be the power connection and whether you use VBUS or 3V3 OUT, but what is recommended to be done to connect a Pico to Thonny AT THE SAME TIME? My PC probably isn't giving the Pico 5V via USB, so can I keep my Pico plugged into my PC while powering it from another source?

b-blake, 2025-06-01:
Same error here with two different cards an two different card interfaces,
8GB Sandisk SD card & 32GB Samsung EVO Select SD Mini card.

Bruce