micropython/micropython #12108: ZLIB has gone in V1.20

Tangerino, 2023-07-28:
In V1.20, `zlib` was replaced by `deflate`, so one remark and one request

- When we remove a library, it implies to make existing code to stop working. Instead, it would be better to make new code to implement new features and recompilation id needed to remove 'old' features.

- How to I re-enable `zlib` support since I don't have bandwidth to fix the user code now?

Thanks in advance

dpgeorge, 2023-07-28:
You must be using an unstable nightly release, because v1.20 did not remove `zlib`.  If you are using unstable releases then you need to expect there to be some issues like this.

> * How to I re-enable `zlib` support

You can install it from `micropython-lib`, as documented in http://docs.micropython.org/en/latest/library/zlib.html

You could also freeze this module if you want to include it directly in the firmware (that may be a better option for your use case).

Tangerino, 2023-07-28:
Yes, it is a nightly release.
Thank you for the quick reply.

closed 2023-07-28 by Tangerino

jimmo, 2023-07-31:
@Tangerino  Some extra notes:

> When we remove a library, it implies to make existing code to stop working. Instead, it would be better to make new code to implement new features and recompilation id needed to remove 'old' features.

This is definitely the approach in general. Honestly it's pretty rare to "replace" a feature, and this example with compression was a bit of a special case. Whenever compilation is required it's a barrier for most users, so it's hard to make forward progress if we cannot enable new features by default.

We considered freezing the zlib module from micropython-lib by default, but then we're committing to doing that forever, and not that many people use compression, so it's extra code size for a small benefit.

> In V1.20, zlib was replaced by deflate,

As Damien already mentioned, it's 1.21 that does this. But I've often thought the way we name the nightly builds is confusing, so I raised https://github.com/micropython/micropython/issues/12127 to follow-up on this.