micropython/micropython #12113: AES CCM Mode

radu022003, 2023-07-29:
I would like to have the possibility to perform encryption using the aes ccm mode on a raspberry pi pico w.
This encryption is required for BTHome project.
I have successfully compiled the micropython with user c module examples and an example for native module as mpy, but I would still need an example in order to create a .mpy module using mbedtls ccm.
Thanks in advance.

Carglglz, 2023-07-29:
@radu022003 
>  I would still need an example in order to create a .mpy module using mbedtls ccm.
Thanks in advance.

check the develop branch at [mpy-mbedtls](https://github.com/Carglglz/mpy-mbedtls/blob/develop/) , here is the [AES CCM bit](https://github.com/Carglglz/mpy-mbedtls/blob/a35f229bab61b630d268323f07c93e9b3d3f71fe/mbedtls/mod_mbedtls.c#L620C11-L620C11) and here is the [bthome_v2_encryption](https://github.com/Carglglz/mpy-mbedtls/blob/develop/example/bthome_v2_encryption.py) example 👍🏼

radu022003, 2023-08-02:
here can be found a working example of bthome https://github.com/radu022003/bthome_v2_pico_w
Many thanks @Carglglz

jcea, 2024-12-11:
I am interested in this too.