micropython/micropython #12124: Support for ESP32 Quad and Octal SPI on SPI2 and SPI3

shantzis62, 2023-07-30:
I have an ESP32S3 project that requires a high bandwidth between the MCU and an external device. The only feature blocking the use of micropython is its lack of support for ESP32's multi-line SPI (Quad and Octal). I looked over the SPI code in `ports/esp32` and it looks pretty doable. I'm considering adding it and sending a PR. Before I do so, though, I'm wondering how I should expose it in a module.  I think the most sensible thing would be to create a separate `esp32.QSPI` class (and perhaps an `esp32.OctalSPI` class).

I'd be interested in hearing what people think.

Thanks!

uraich, 2023-08-24:
Yes, very interested! I tried to get the esp32-camera driver to run on the esp32s3 and I get problems when trying to configure the camera, at the moment the driver tries to allocate the framebuffer in PSRAM.
When I allocate the buffer in SRAM everything works fine.

echo-lalia, 2024-04-19:
I'm currently looking for quad spi support for an [esp32 s3 WROOM](https://github.com/profi-max/JC4827W543_4.3inch_ESP32S3_board/tree/main), which includes a display for use with quad-spi. 

I'm very interested in this feature as well! I want to write a driver to use this display module in MicroPython, but I need QSPI working first. 

My thinking is that the QSPI feature could just be added to the SPI feature for supported ports. The reason is, that this seems to be essentially how it works in [esp-idf](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_master.html#spi-master-on-spi1-bus). So, theoretically, I think it would just require passing on a few extra args when in use.

In fact, if I understand this correctly (and I absolutely might not 😅) it looks like the [esp32 hw spi code](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_hw_spi.c) is just automatically assigning the QUADWP and QUADHD pins to -1. Perhaps all that would be needed would be a couple optional keywords for those pins? (For quad-spi at least?).

```
    spi_bus_config_t buscfg = {
        .miso_io_num = self->miso,
        .mosi_io_num = self->mosi,
        .sclk_io_num = self->sck,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1
    };
```

echo-lalia, 2024-04-19:
I'm currently looking for quad spi support for an [esp32 s3 WROOM](https://github.com/profi-max/JC4827W543_4.3inch_ESP32S3_board/tree/main), which includes a display for use with quad-spi. 

I'm very interested in this feature as well! I want to write a driver to use this display module in MicroPython, but I need QSPI working first. 

My thinking is that the QSPI feature could just be added to the SPI feature for supported ports. The reason is, that this seems to be essentially how it works in [esp-idf](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_master.html#spi-master-on-spi1-bus). So, theoretically, I think it would just require passing on a few extra args when in use.

In fact, if I understand this correctly (and I absolutely might not 😅) it looks like the [esp32 hw spi code](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_hw_spi.c) is just automatically assigning the QUADWP and QUADHD pins to -1. Perhaps all that would be needed would be a couple optional keywords for those pins? (For quad-spi at least?).

```
    spi_bus_config_t buscfg = {
        .miso_io_num = self->miso,
        .mosi_io_num = self->mosi,
        .sclk_io_num = self->sck,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1
    };
```

Btw, if anyone would be willing to share some working (even if unofficial or janky) code for getting qspi in mp, I'd be very grateful 😊

Usonaki, 2024-08-22:
> Btw, if anyone would be willing to share some working (even if unofficial or janky) code for getting qspi in mp, I'd be very grateful 😊

I have some janky code that works. It won't let me add .c and .h files.