micropython/micropython #12130: Stream.readinto() returned before EOF encountered

reiyawea, 2023-07-31:
# basic information
 - firmware: esp32-20230426-v1.20.0.bin
 - version information: MicroPython v1.20.0 on 2023-04-26; ESP32 module with ESP32
 - board: generic ESP32-WROOM-32E module (ESP32 D0WD chip, without PSRAM)
# issue
`Stream.readinto()` returned too early, before EOF is encountered.
# sample code
```
import uasyncio as asyncio

buffer = bytearray(16384)

async def foo(reader, writer):
    length = await reader.readinto(buffer)
    print('%d bytes received'%length)
    writer.close()
    await writer.wait_closed()

loop = asyncio.get_event_loop()
loop.create_task(asyncio.start_server(foo, '0.0.0.0', 5555))
loop.run_forever()
```
When PC sends 8192 bytes (before closing the connection), `reader.readinto(buffer)` returns only 4320 only (`length` being 4320).
I think `Stream.readinto()` should not return until EOF encountered (i.e. when peer closes the connection).

peterhinch, 2023-08-01:
My understanding of `StreamReader.readinto` is that it returns the number of bytes available at the time when it is called. If it waited for EOF it would be impossible to use with hardware interfaces such as I2S (see [example code](https://github.com/peterhinch/micropython-micro-gui/blob/d6cc79e7143b547d0de5b35973b856b64706e97a/gui/demos/audio.py#L244)) where the data needs to be consumed as it is produced. 

It therefore needs to be called in a loop as in the example. When EOF is reached, zero bytes are returned, enabling a loop exit.

dpgeorge, 2023-08-01:
@peterhinch is correct: `readinto(buf)` will read *at most the length of the buffer*.  It may read less than that.  See the docs: http://docs.micropython.org/en/latest/library/asyncio.html#asyncio.Stream.readinto

You may want to use `readexactly()`.

closed 2023-08-01 by dpgeorge

reiyawea, 2023-08-02:
@dpgeorge Thanks for your explanation. I will try other ways as workaround.
But consider `socket.readinto(buf[, nbytes])`, which reads bytes into the `buf`. When `nbytes` is not specified, it reads all data available from the socket until EOF.
I thought `Stream.readinto` should have had the same feature.

dpgeorge, 2023-08-03:
> But consider `socket.readinto(buf[, nbytes])`, which reads bytes into the `buf`. When `nbytes` is not specified, it reads all data available from the socket until EOF.

This is not true.  Sockets can also return less data than requested (although they do try to read as much as possible).  This behaviour is documented.