micropython/micropython #12140: When the bytearray() function allocates large amounts of memory, it causes the machine to run slower

iBigDane, 2023-08-01:
I use micropython 1.19.1,use esp32-wrover-e whith 8MB SPIRAM.
I want to store the recording information of I2S in bytearray().
run the code
`
buf = bytearray(512*1000)
`
my micropython become very slower,Even if I just print a letter,print("a")
I tried running the code like this
```
a = bytearray(64000)
b = bytearray(64000)
c = bytearray(64000)
d = bytearray(64000)
e = bytearray(64000)
f = bytearray(64000)
g = bytearray(64000)
h = bytearray(64000)
```
This remains the case

iBigDane, 2023-08-01:
I used c to allocate 512KB of memory and then called it on micropython without slowing micropython down
`buf = gc_alloc(512*1000, false);`

The complete code is as follows
```
STATIC mp_obj_t machine_record_record(mp_obj_t self_in, mp_obj_t rec_time) {
    (void)self_in;
    const char *filename = "rec.pcm";
    mp_obj_t f_args[2] = {
        mp_obj_new_str(filename, strlen(filename)),
        MP_OBJ_NEW_QSTR(MP_QSTR_wb),
    };
    mp_obj_t file = mp_vfs_open(MP_ARRAY_SIZE(f_args), &f_args[0], (mp_map_t *)&mp_const_empty_map);

    len = 128000 * mp_obj_get_int(rec_time);
    buf = gc_alloc(len, false);
    int count = 0;
    mp_hal_delay_ms(1000);
    mp_printf(&mp_plat_print,"start record\n");
    size_t bytes_read;
    
    i2s_read(I2S_NUM, (char *)buf, len, &bytes_read, 100);
    mp_printf(&mp_plat_print,"end record\n");
    int errcode;
    int len_w = mp_stream_rw(file , buf, len, &errcode, MP_STREAM_RW_WRITE);

    if (errcode != 0 && len_w!=len) {
        mp_raise_OSError(errcode);
        memset(buf, 0, len);
        file=NULL;
        mp_printf(&mp_plat_print,"Write %s error!\n\r",filename);
    }
	
    free(buf);
    mp_stream_close(file);
	
    return mp_obj_new_int(1);
}
```

iBigDane, 2023-08-01:
[image]

robert-hh, 2023-08-01:
When I run:
```
import time
a = bytearray(64000)
b = bytearray(64000)
c = bytearray(64000)
d = bytearray(64000)
e = bytearray(64000)
f = bytearray(64000)
g = bytearray(64000)
h = bytearray(64000)
buf=bytearray(512*1000)
```
and then execute:
```
s=time.time();print(a);print(time.time()-s)
# or
s=time.time();print(b);print(time.time()-s)
# or
s=time.time();print(c);print(time.time()-s)
```
Each of these take ~22 seconds. At a serial speed of 115200 and the total transferred number of bytes as 4 * 64000 for each array, this is the serial transmit time (4 * 64000 / 11520)

Printing a string like in your example happens immediately. So something strange is happening at your set-up.

Board ESP32 with a WROVER module.
MicroPython v1.20.0-311-gea1a5e43d on 2023-07-23; ESP32 module (spiram) with ESP32

iBigDane, 2023-08-01:
[image]
I burned the official 1.20.0 version of the firmware, and it was still very slow when running
I haven't changed any Settings

robert-hh, 2023-08-01:
Please try the most recent daily build. These have changed to IDF 5.x.

iBigDane, 2023-08-01:
[v1.20.0-327-gd14ddcbdb (2023-07-27) .bin](https://micropython.org/resources/firmware/esp32spiram-20230727-unstable-v1.20.0-327-gd14ddcbdb.bin)
I use the lmost recent daily build and the phenomenon is the same

peterhinch, 2023-08-01:
It sounds like broken hardware to me. Here are my results on an ESP32 with SPIRAM:
```python
MicroPython v1.20.0-283-g7d66ae603 on 2023-07-14; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from time import ticks_ms, ticks_diff
=== t = ticks_ms()
=== a = bytearray(512*1000)
=== t1 = ticks_ms()
=== print("a")
=== t2 = ticks_ms()
=== print(f"Alloc {ticks_diff(t1, t)}ms print {ticks_diff(t2, t1)}ms")
=== 
a
Alloc 144ms print 1ms
>>> 
```

iBigDane, 2023-08-01:
> And in your example the print("a") takes extremely long? If yes, something may be broken at you set-up. If it's not the board, it something about the way you connect to it. Which terminal emulator do you use? If it is Thonny, try something simple like mpremote, PuTTy or screen.

> I used thonny, thonny version is 4.0.2, I used PuTTy as you suggested, The previous program worked fine,Thank you so much

closed 2023-08-01 by dpgeorge