micropython/micropython #12153: Add Thread ID return to _thread.start_new_thread function

pablogventura, 2023-08-03:
Description:
Currently, in MicroPython, the `_thread.start_new_thread` function allows starting a new thread but does not provide a direct way to obtain the ID of the newly created thread. This feature request aims to add Thread ID return to the `start_new_thread` function, similar to CPython, to facilitate tracking and managing the created threads.

Proposal:
Modify the `_thread.start_new_thread` function to return the ID of the newly created thread. This would allow developers to have a more effective way of tracking threads in execution and overall improve the threading experience in MicroPython.

Example of usage:
```python
import _thread

def my_thread_func():
    # Thread's code here
    pass

# Start the thread and get the thread ID
thread_id = _thread.start_new_thread(my_thread_func, ())

# Print the thread ID
print("Thread ID:", thread_id)
```

Benefits:

Eases identification and management of threads created using start_new_thread.
Allows better control and monitoring of running threads.
Improves compatibility with existing code that utilizes this functionality in CPython.

Considerations:
It is essential to consider the performance implications and backward compatibility while implementing this feature. However, since CPython already provides this functionality, adding it to MicroPython would be beneficial for users who work with threads and desire to maintain common code between both implementations.

I am willing to implement this improvement and work on a pull request for the MicroPython repository if you find it useful.

Thank you.

dlech, 2023-08-03:
Here is a patch for the unix port that we've been carrying for a while: https://github.com/pybricks/micropython/commit/afe519d7e00d2ab1578ec8a5173530801fd92056

We have a use case where we are using signals to interrupt syscalls that are blocking a specific thread and so this feature is essential in order to get the thread id while avoiding race conditions.

pablogventura, 2023-08-03:
Thank you very much for your reply! I am working on ESP32. Do you think it could be ported easily? Anyway, I'm curious if there is some underlying reason for not letting start_new_thread return the id in the original implementation.

dlech, 2023-08-03:
I suspect it was omitted to save a bit on code size. The [original PR](https://github.com/micropython/micropython/pull/2165) doesn't really give any details.

I went ahead a made a PR to implement it since I was already half way there.

pablogventura, 2023-08-07:
Thanks you a lot! I hope it will be merge soon!

closed 2023-09-03 by dpgeorge