micropython/micropython #12154: Build on Debian fails

g0730n, 2023-08-03:
I keep getting this error, while trying to build for Pi-Pico, using my laptop with Debian 10.13:

Just as a test I am building the master branch of micropython found here on github with no modifications. I have updated all build tools, and following instructions to a T.

Attempting to build on my pi zero to see if it works there. Have tried everything I can find in search results for similar issues to no avail.

> make
[ -e build-PICO/Makefile ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO -DMICROPY_BOARD_DIR=/home/gordon/Desktop/pi0-pico/pico/micropython/ports/rp2/boards/PICO
PICO_SDK_PATH is /home/gordon/Desktop/pi0-pico/pico/micropython/lib/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Build type is MinSizeRel
PICO target board is pico.
Using board configuration from /home/gordon/Desktop/pi0-pico/pico/micropython/lib/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.7 (found version "3.7.3") found components:  Interpreter 
TinyUSB available at /home/gordon/Desktop/pi0-pico/pico/micropython/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
CMake Warning at /home/gordon/Desktop/pi0-pico/pico/micropython/lib/pico-sdk/src/rp2_common/pico_btstack/CMakeLists.txt:14 (message):
  PICO_BTSTACK_PATH specified but content not present.

>CMake Warning at /home/gordon/Desktop/pi0-pico/pico/micropython/lib/pico-sdk/src/rp2_common/pico_lwip/CMakeLists.txt:14 (message):
  PICO_LWIP_PATH specified but content not present.

>Including custom CMake file /home/gordon/Desktop/pi0-pico/pico/micropython/lib/pico-sdk/src/common/pico_base/generate_config_header.cmake
Found User C Module(s): 
ELF2UF2 will need to be built
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gordon/Desktop/pi0-pico/pico/micropython/ports/rp2/build-PICO
make -s -C build-PICO || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
Scanning dependencies of target BUILD_FROZEN_CONTENT
make[3]: *** No rule to make target 'genhdr/mpversion.h', needed by 'genhdr/qstr.i.last'.  Stop.
make[2]: *** [CMakeFiles/Makefile2:73: CMakeFiles/BUILD_FROZEN_CONTENT.dir/all] Error 2
make[1]: *** [Makefile:84: all] Error 2
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:46: all] Error 1

g0730n, 2023-08-03:
Build just successfully completed on Pi-Zero, using exact same parameters as on my Debian machine.

dlech, 2023-08-03:
It sounds like submodules did not get checked out. `make submodules`

g0730n, 2023-08-03:
> It sounds like submodules did not get checked out. `make submodules`

I did do this before building. Will try again when I'm home from work. Using the same exact steps on my pi zero it worked flawlessly. Have tried 3 times on my laptop.

dlech, 2023-08-03:
You might need to do `make clean` as well.

g0730n, 2023-08-04:
Okay, trying again. Here's what I tried:

`sudo apt-get update`
`sudo apt-get upgrade`
`git clone https://github.com/micropython/micropython.git --branch master`
`cd micropython/mpy-cross`
`make`
`cd ..`
`cd ports/rp2`
`make clean`
`make submodules`
`make`

Still getting the same exact error. Used the same commands to build on Pi, so know it's not me it's something with my laptop.

andrewleech, 2023-08-04:
It could quite likely be an old version of cmake causing the issue, debian always has "stable" (aka old) versions of packages installed by apt-get, even on latest debian. You might have better luck with `sudo apt remove cmake; pip3 install -U cmake`

sosi-deadeye, 2023-08-04:
If it helps, Debian Bookworm on x86_64 (12) has this version of `cmake`:
```
esp32@miranda:~/micropython/ports/rp2$ cmake --version
cmake version 3.25.1
```

On a RP0W2 (Debian Bullseye, 11, aarch64):
```
deadeye@rpi0w2:~ $ apt-cache show cmake
Package: cmake
Version: 3.18.4-2+rpt1+rpi1+deb11u1
Architecture: arm64
```

Trying compilation on RP0W2:
```
git clone https://github.com/micropython/micropython.git
cd micropython
echo "Compiling mpy"
make -C mpy-cross -j 4

echo "Getting submodules for PICO"
make -C ports/rp2 BOARD=PICO submodules
echo "Compiling for PICO"
make -C ports/rp2 BOARD=PICO -j 4

echo "Getting submodules for PICO_W"
make -C ports/rp2 BOARD=PICO_W submodules
echo "Compiling for PICO_W"

# changed -j to 1 to prevent unresponsive RP0W2
make -C ports/rp2 BOARD=PICO_W -j 1
```

After installing `cmake` and `gcc-arm-none-eabi` I was able to compile the firmware.

It built the firmware for `PICO`, but got unresponsive during compilation of the firmware for `PICO_W`. So I changed `-j 4` to `-j 1`.

Firmware for PICO_W is still compiling ...

g0730n, 2023-08-04:
I am installing a new debian/python install on a VM to see if I can get a work around. I tried uninstalling cmake, then reinstalling it via pip3, but was running into errors.

So I think either my cmake or my python/pip install on laptop is corrupted. When I setup this laptop a few years ago I did not use any virtual environments for python and I think something got messed up in the past year or so. I have another laptop with a fresh install of debian and python3 so I could try on there as well.

g0730n, 2023-08-04:
> 

Interesting... I reinstalled cmake, and used your commands (not in VM, on laptop) building PICO failed with same error, build for PICO_W worked fine.

On virtuabox: New Debian install, installed cmake, python3.11, and any other dependencies for building. Both PICO and PICO_W built just fine.