micropython/micropython #12164: ports/stm32: lwip submodule not present, builds failing

renestraub, 2023-08-04:
# Description

When trying to compile STM32 boards with Ethernet support lwip sources are missing. 

`make submodule` initialises the following modules, but not lwip. Builds that depend on lwip fail.

``` sh
straub@Mac-mini-2 stm32 % make V=1 submodules
Updating submodules: lib/libhydrogen lib/stm32lib lib/micropython-lib
git submodule sync ../../lib/libhydrogen ../../lib/stm32lib ../../lib/micropython-lib
git submodule update --init ../../lib/libhydrogen ../../lib/stm32lib ../../lib/micropython-lib
Submodule 'lib/libhydrogen' (https://github.com/jedisct1/libhydrogen.git) registered for path '../../lib/libhydrogen'
Submodule 'lib/micropython-lib' (https://github.com/micropython/micropython-lib.git) registered for path '../../lib/micropython-lib'
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path '../../lib/stm32lib'
Cloning into '/Users/straub/prj/lwip_test/lib/libhydrogen'...
Cloning into '/Users/straub/prj/lwip_test/lib/micropython-lib'...
Cloning into '/Users/straub/prj/lwip_test/lib/stm32lib'...
Submodule path '../../lib/libhydrogen': checked out '5c5d513093075f7245ea522101b17c50aa579af2'
Submodule path '../../lib/micropython-lib': checked out 'c113611765278b2fc8dcf8b2f2c3513b35a69b39'
Submodule path '../../lib/stm32lib': checked out '928df866e4d287ebc3c60726151513ebee609128'
```

The missing module(s) can be populated by running the relevant git command manually. 

``` sh
straub@Mac-mini-2 lwip_test % git submodule init
Submodule 'lib/asf4' (https://github.com/adafruit/asf4) registered for path 'lib/asf4'
Submodule 'lib/axtls' (https://github.com/micropython/axtls.git) registered for path 'lib/axtls'
Submodule 'lib/btstack' (https://github.com/bluekitchen/btstack.git) registered for path 'lib/btstack'
Submodule 'lib/cyw43-driver' (https://github.com/georgerobotics/cyw43-driver.git) registered for path 'lib/cyw43-driver'
Submodule 'lib/fsp' (https://github.com/renesas/fsp.git) registered for path 'lib/fsp'
Submodule 'lib/libffi' (https://github.com/atgreen/libffi) registered for path 'lib/libffi'
Submodule 'lib/lwip' (https://github.com/lwip-tcpip/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/mynewt-nimble' (https://github.com/micropython/mynewt-nimble.git) registered for path 'lib/mynewt-nimble'
Submodule 'lib/nrfx' (https://github.com/NordicSemiconductor/nrfx.git) registered for path 'lib/nrfx'
Submodule 'lib/nxp_driver' (https://github.com/hathach/nxp_driver.git) registered for path 'lib/nxp_driver'
Submodule 'lib/pico-sdk' (https://github.com/raspberrypi/pico-sdk.git) registered for path 'lib/pico-sdk'
Submodule 'lib/tinyusb' (https://github.com/hathach/tinyusb) registered for path 'lib/tinyusb'
Submodule 'lib/wiznet' (https://github.com/andrewleech/wiznet_ioLibrary_Driver.git) registered for path 'lib/wiznet5k'
straub@Mac-mini-2 lwip_test % git submodule update
Cloning into '/Users/straub/prj/lwip_test/lib/asf4'...
Cloning into '/Users/straub/prj/lwip_test/lib/axtls'...
Cloning into '/Users/straub/prj/lwip_test/lib/btstack'...
Cloning into '/Users/straub/prj/lwip_test/lib/cyw43-driver'...
Cloning into '/Users/straub/prj/lwip_test/lib/fsp'...
Cloning into '/Users/straub/prj/lwip_test/lib/libffi'...
Cloning into '/Users/straub/prj/lwip_test/lib/lwip'...
Cloning into '/Users/straub/prj/lwip_test/lib/mynewt-nimble'...
Cloning into '/Users/straub/prj/lwip_test/lib/nrfx'...
Cloning into '/Users/straub/prj/lwip_test/lib/nxp_driver'...
Cloning into '/Users/straub/prj/lwip_test/lib/pico-sdk'...
Cloning into '/Users/straub/prj/lwip_test/lib/tinyusb'...
Cloning into '/Users/straub/prj/lwip_test/lib/wiznet5k'...
Submodule path 'lib/asf4': checked out '84f56af13292d8f32c40acbd949bde698ddd4507'
Submodule path 'lib/axtls': checked out '531cab9c278c947d268bd4c94ecab9153a961b43'
Submodule path 'lib/btstack': checked out '77e752abd6a0992334047a48038a5a3960e5c6bc'
Submodule path 'lib/cyw43-driver': checked out '8ef38a6d32c54f850bff8f189bdca19ded33792a'
Submodule path 'lib/fsp': checked out 'e78939d32d1ccea9f0ba8bb42c51aceffd386b9b'
Submodule path 'lib/libffi': checked out 'e9de7e35f2339598b16cbb375f9992643ed81209'
Submodule path 'lib/lwip': checked out '6ca936f6b588cee702c638eee75c2436e6cf75de'
Submodule path 'lib/mynewt-nimble': checked out '42849560ba7906f023f61e5f7ff3709ba2c1dfca'
Submodule path 'lib/nrfx': checked out '7a4c9d946cf1801771fc180acdbf7b878f270093'
Submodule path 'lib/nxp_driver': checked out 'fa5a554c7944d2a196626f8d3631e44943f9abcc'
Submodule path 'lib/pico-sdk': checked out '6a7db34ff63345a7badec79ebea3aaef1712f374'
Submodule path 'lib/tinyusb': checked out '868f2bcda092b0b8d5f7ac55ffaef2c81316d35e'
Submodule path 'lib/wiznet5k': checked out '0803fc519ad7227e841287fb3638d6c8b2f111a1'
```

Is this intended? It would be more convenient if these modules would be initialised by `make submodules` in ports/stm32.

# Steps to reproduce

``` sh
git clone git@github.com:micropython/micropython.git lwip_test
cd lwip_test
cd mpy-cross
make
cd ..
cd ports/stm32
make submodules
make BOARD=STM32F769DISC

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-STM32F769DISC/genhdr
GEN build-STM32F769DISC/genhdr/pins.h
GEN build-STM32F769DISC/genhdr/pybcdc.inf
GEN build-STM32F769DISC/genhdr/pybcdc_inf.h
GEN build-STM32F769DISC/genhdr/pllfreqtable.h
GEN stmconst build-STM32F769DISC/modstm_qstr.h
GEN build-STM32F769DISC/genhdr/mpversion.h
GEN build-STM32F769DISC/genhdr/qstr.i.last
../../extmod/modcryptolib.c:64:10: fatal error: mbedtls/aes.h: No such file or directory
   64 | #include <mbedtls/aes.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
../../extmod/modhashlib.c:35:10: fatal error: mbedtls/version.h: No such file or directory
   35 | #include "mbedtls/version.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
../../extmod/modlwip.c:42:10: fatal error: lwip/init.h: No such file or directory
   42 | #include "lwip/init.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
mpnetworkport.c:40:10: fatal error: lwip/netif.h: No such file or directory
   40 | #include "lwip/netif.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
network_lan.c:34:10: fatal error: lwip/netif.h: No such file or directory
   34 | #include "lwip/netif.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
../../extmod/modssl_mbedtls.c:41:10: fatal error: mbedtls/platform.h: No such file or directory
   41 | #include "mbedtls/platform.h"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
../../extmod/network_lwip.c:35:10: fatal error: lwip/init.h: No such file or directory
   35 | #include "lwip/init.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
In file included from ../../shared/netutils/dhcpserver.c:38:
../../shared/netutils/dhcpserver.h:29:10: fatal error: lwip/ip_addr.h: No such file or directory
   29 | #include "lwip/ip_addr.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
```

# Environment / Version

* MacOS Ventura 13.5

```
* 2dcd74543 - (8 months ago) py/gc: Speed up incremental GC cycles by tracking the last used block. - Damien Tournoud (HEAD -> master, origin/master, origin/HEAD)
* 70c564324 - (3 days ago) extmod/modssl_mbedtls: Reference SSLContext from SSLSocket. - Jim Mussared
* 01c758e26 - (8 days ago) unix/README: Fix Markdown link markup. - Brett Cannon
```

sosi-deadeye, 2023-08-04:
To get the right `submodules`, you need to set the environment variable `BOARD`.

```
esp32@miranda:~/micropython/ports/stm32$ make BOARD=STM32F769DISC submodules
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/mbedtls lib/lwip lib/libhydrogen lib/stm32lib lib/micropython-lib
Synchronizing submodule url for '../../lib/libhydrogen'
Synchronizing submodule url for '../../lib/lwip'
Synchronizing submodule url for '../../lib/mbedtls'
Synchronizing submodule url for '../../lib/micropython-lib'
Synchronizing submodule url for '../../lib/stm32lib'
```

Then compile:
`make -j 16 BOARD=STM32F769DISC`

I do not get any error:
```
LINK build-STM32F769DISC/firmware.elf
   text    data     bss     dec     hex filename
 489704     144   88516  578364   8d33c build-STM32F769DISC/firmware.elf
GEN build-STM32F769DISC/firmware.bin
GEN build-STM32F769DISC/firmware.hex
GEN build-STM32F769DISC/firmware.dfu
```

renestraub, 2023-08-04:
Thanks for the clarification. However this is not correctly documented on https://docs.micropython.org/en/latest/develop/gettingstarted.html

```
$ cd ports/stm32
$ make submodules
$ make BOARD=<board>
```

dpgeorge, 2023-08-04:
Fixed by 7fad499d1e4d31ffdd4a93a61a9c604189636e3c

closed 2023-08-04 by dpgeorge