micropython/micropython #12172: Uart multiplying messages when configuration is changed

yamenk-gribaudo, 2023-08-05:
Uart sending previously sended messages when running `init()` with another rx-tx configuration. Minimal example:

```
from machine import UART
import time

uart = UART(1, 115200, bits=8, parity=None, stop=1, timeout=40)

while True:
    print("working")
    time.sleep_ms(100)
    uart.init(rx=4, tx=15)
    time.sleep_ms(100)
    uart.write("uart 1\n")
    
    time.sleep_ms(100)
    uart.init(rx=16, tx=17)
    time.sleep_ms(100)
    uart.write("uart 2\n")
    
    time.sleep_ms(100)
    uart.init(rx=18, tx=5)
    time.sleep_ms(100)
    uart.write("uart 3\n")
```

I expect that if I connect something to the uart 1, I should revieve something like: 
```
uart 1
uart 1
uart 1
...
```

But instead I see:
```
uart 1
uart 2
uart 3
uart 1
...
```

I also try to flush, deinit and redefine the uart each time, but got the same problem. So it seems a bug in the micropython uart implementation.

I could work on it if someone points me in the right direction.
Thanks,

robert-hh, 2023-08-05:
You did not mention the board. Is it a ESP32?

robert-hh, 2023-08-05:
What do you get at the other pin's you have chose for output. You might get the same. To stop the output at a TX pins and to resolve a possible conflict at the input pins, you have to disable the rx, tx pins switching input and output to another pin pair. Just declare them as GPIO input (rx) and output(tx) pins using machine.Pin. Set the tx to high.

yamenk-gribaudo, 2023-08-05:
> You did not mention the board. Is it a ESP32?

Sorry, the board is generic esp32 like [this one ](data:image/jpeg;base64,/9j/4AAQSkZJRgABAQAAAQABAAD/2wCEAAoHCBQUFBgUFRUYGBgaGxoaGBsbGxsaGx0bHBsZGxsaGSEbIC0kHB0tHxodJTcmKi4xNDQ0GyM6PzoyPy0zNDEBCwsLEA8QGBISGjMmIyEzMTMzNTMxMTEzMzMzMzMzMzMzMTMzMzEzMTEzMzMxMzExMzMzMzMzMzExMzMxPj4zM//AABEIAMcA/QMBIgACEQEDEQH/xAAcAAEAAQUBAQAAAAAAAAAAAAAABgECBAUHAwj/xAA7EAACAQMCBAQDBgUEAgMBAAABAgMAESEEEgUGMUETIlFhMnGBI0JSkaGxFGLB4fAHcoLRJDOSsvFT/8QAGQEBAQEBAQEAAAAAAAAAAAAAAAECAwQF/8QAIREBAQEAAgEFAQEBAAAAAAAAAAERAjEhAxJBUXFhIgT/2gAMAwEAAhEDEQA/AOzUpSgUpSgUpSgUpSgUpSgUpXhqdQkal5GVFGSzEAD6mg9qVhjiUPhibxU8Mi4fcNpHz6VoYucoSxJssW+SMNfzbo1DMxUZ22v74qaJXSsfS6mOVFeNldWF1ZTcEexFZFUKUpQKUpQKUpQKUpQKUpQKUpQKUpQKUpQKUpQKUpQKUpQKVQmopx7nzRaS6mTxJBjZHZiD6Mb7V+pvUtwSutTxrmHS6QXnlVPRb3Y/JRk1yDj/APqbrJ7pDbTp/L5nI92Ix9APnUHmlZ2LMxZj1LEkn5k5NTzVx07j/wDq07XXRxbB/wD0ksW/4oDYfUn5VzzinGdRqW3TSu5/mOB8lGB9BWFFGzsFVSzHoALk/QVJ+F8lTSWMpEa+nxP+XQfU/SlydiNeM1gu5toNwtzYN6gdAfetkeFa1o3maNypYuxPxsT1axNyOvbvXROF8vafT5SMF/xt5m+np9K2oFYvqT4i4iXBedX05gihIcuI1ZXYKgJYrZvwHaVuf5b5Bz1zhnGIp7qjLvX4kuCRYlSR6rcEXHpXKuM8pxyuJYgFcMC6fCsgvcjp5WI79M1G9JxOQa3e6PAyOSkYYr4e7aDfuykKMdCc+x3OUqPo2lQnljnmPUSeBJ5ZAoYMP/W3YqT9xwcWODi2TYTatIUpSgUpSgUpSgUpSgUpSgUpSgUpSgUpSgUqlY2o1SKhcuqr2YkWv7eualuD0nmRFLuwVRksxAAHqScCoJx7/VDSw3XTqZ3GL/BGD8yLt9Bb3rl/MvG9RqnJknMqAnZbCCxP3VwD79a0JapPKpJx7nTWau4eQoh+4l0X5Gxu31JqNms/hPB5tUxWJQbW3MTZRe9rn6Hp6VM+GckxpmZjI34RdUv+5/Sls4nmoHo9DJK22NGc+wwPmeg+tSzhnI5+Kd7fyJ1+rH+g+tTWGFI1CIqovYAAD8hVxYAX/wC6xef017WPw3hsUK7Y41T3A8x+ZOTWcWC5NY5kY/CLe/f9elerQjqc1i37VTeDlc5/zFXeGQNzH6etVRwM2x61dPGxyDg/p+vSsjzWYL0T97/qKweK8Hj1S7jdJFuEkXDrcEHI6ggnB/Ssrwrj39xb9qrH5Ot/6/WruFjl83D30Dv4wvj7J7EqzXBBHa4tcqx6X61OOR+fnUvHMGaIZjGXdF6lQ3WQAZCkbrA2JwK288CSo0ciBlOCD0P9x6ioPxvl+bShn043xkhr+YyR2N7gA2YXAN+otXXjz3tnHdNDrY5kWSN1dGF1ZTcH/o+3UVl1xPlnX6vQ6RZ7oI2d3W5XbIAPMjgZVj1DZOLdK6ly/wAwRayNXQ7WZQxQkFgOh6HIvi4+tjitst1SlKoUpSgUpSgUpSgUpSgUpSgUpSgifMvM50s0cZCojqS0rglA33VO04B7n8r1zPX8NGu1rx6fU70VGlvdnRTYM6oL4O4m30ro/OWrgV0h1C3SVSAT0BUjr+YqCajhp4YTLHCJ4mZZFYOyOhXIBK9V9vl865y5fLXwhTtd2VyqyAlS5+B7YtJbofRx/wAr9RjzQlSQQVYfEp6i/T5g9QRgiptwnga6rRaqZ40Er/aRsQSqxofOFNx5s/oPlUV0ySrGviIPCJZI5Wt9mwJ+Bs+TcpBUi3W1jmums1ZwbiE+nfxImAFwrKxGx73IV1vkYOe3Yg11HhXFo9SCB5JFF3jJuR/Mp++n8w6XzbqYnr4lliBQh2ih08aFV8jPIfDcFiMgMqnDWBYn2rTqXgmEYdg6MbAMPEVl65XFrDqPlXO5f1vp0yVGvRENwMZ797Vq+CcwpPtjlKrIcKwsEfsB6JJ7dD2t0reGHPcGufSrFcG4FgRgXq7wyfY9z/b+lVJsbfUki5+QqniZGT79D+1Sj0AA62qryAC/X9q8XQNZu3vVqt2GamC97keleO4DFr16sl+uKvVR2FqDwYM3U2Hyr2jW2KFa8NTqo4hulcIPc2NNp4R7mPk9JT4kPlcXJjvZGva5A6Kxt8j39a0Wu4lFp/Ci0vjJKlizSAI0cl/MEPUqRgjKnFbDi3Pii66dNx/G3T6VE5JJtWzSM4MgBNuhKjsv4mF729K7cdzyzXbeA87RSbEnZUZztRiRtZgAdrHAViCCOxvi3SpnXy7w7iTrKkjqHCMpEY8oJW1h9a6Xyxz00csemkbxEI2lmYb0Zb5B6OCLXHYhrYwNsur0rF0erSVA8bBlPQj2wQfQg4I7VlVQpSlApSlApSlApSlAqhqtUoIX/qBy8+siUxkb03WU/eDWuAexx8q59wfj0ui+w1UbbBhlcXZR2t7W9PpXbdRHeoDzVy1Hq5gjEoxU2dQCcZG4Y3DNZrXaOcwcEkMbS6GV/Cky8aOQpv18oNv+6weI8W0p4eulMis0dgiBHWS58zmTcLbg/ocg1jafiM/DtR4dyV3FWU9GH4h2FxW31mig1ZXUQ7Q6kEo2Rf8ABIOtvfqPes9fhrScb4fJwwwqzSSoyo7IdyKr+VgBYm9j6974rPXwdcqgv5mknmcxggpePeoLsuPMvqD7VXUcVl1k8Oll021ImDyZeR3XeGb/AHjqo633epry5s4ZLDrv/DjZPFAIhjJjLAjoQpBGDY9O96Zqy40pLx7RKVVpUV0a67XVum+3wMRbPSpXwTmUp9nPfapChjdnT0D93X3+Ie4wNLHq45I5o3UI4jSAQhWeRTHJ5uoNvKT2sLVi6/SHTNJhzBHI0SyEr4iOovYjBdMg2sLXxU/lXPp1HaCAwIIIBVgbgg9wRgg+vSvJ1yRb9s1B+CccfTlVw8bXYi/lN8loj91/UHB79iJvpNVHKgkjbcpNj2INr7WHVWt29Mi4zWbMFxXNz+XWr0AHQWrWaddWZ2LvD4GdoVW3+24nGP6/lfruMwwizuC34R1rI2Vqw9fxGKEEyuBbt1P5VDOK84SsdsQCL75J/LpWgn1Tyg3IsG+JjZFJt1JP/Z+danD7Nb/i/PEjXSBNg/E2W+gqJy+NqCzsXktlmPwqPUk+VRXv/CDaZFBkUFQznyRgm9rL8b/CewH6VtG0JZ5IpHDKkHjIp+zRboGUqitYm9xfNbkk6ZauDRRHYoJkkY/CDsjXIABc5Yn+UW968zMSnhvIiIGLBOi7rWJvkk2A9avbiDER7cNGpAJO7LG9wLWFu1YLxg9etbSrdSUPwHcLdbEZ+ufzt8qyeE6uOFrurEWwF23v/wAhgdrixHasnhHBJNS22NkBzcM20+wGO/Y9PcVharSPG5jkUo69VOCP7e9VE45C4/MNUSHZUdGuDbY7otwXHZyo+IWz1vXWuDcfg1VxG43qFLxk+ZQwBBt3BB6/MYIIr5rj1DpfaSAQQRci9wV7exqVcn66GFJHeV0cLuDKb+fd5BYAlRbBJxdhRX0HSoRyfzsmpS0xVGD7Fe4CydApI+4xuBboT0OQKmwoitKUoFKUoFKUoFKUoLWFaHiGnPjo1j3/AGNSCvKRalWILqeFxyvPHIitcAqSAWU2K7kJHlPSucycO1Ojfx1BCHy7wRYm58rC974P9DXZn0BWR3xYrj1ve9R7V6QPpZEYXG5v33Vjpe0a02og4jGAfstTH8DqdrX9Q1xb/L+teHAZ/wCA1Eh17SM77PDmJZ7bSTY7mvnGL9u9YfHuXjpQs0LOQ2T0umLixHUdc1sOD8ch1sQ02sC3PlV7DJv69j7flTM8waheHNxDVTamNzCqg7GAsXZR1OcAKQTk4tXhwPiw8RI5gkbJK0kjSF3aRyhjI81zfpjJNsCtvxZJ+GxNEqJNp2LMjEAtGzLtLAj2+nYikfLarw19RKoeacCRZCCSg3WADjoTYMfn3pP9RZWkk4ZJHHG0as4kibUTRsVUIBI0e6P4SjYHqc5r24ZxJ4mE0Mm5TZbt0wf/AFzr+x+qkVg6PXzJExmRiJY3gjnctsVS1yF6/eXtbN7mtjxSNSZ9XC4QoY40WMKY5AyHfcWG74e3S9TMV58x81zyP4aWjW3msb3NzgE9rWrTwaZ287MFU/fkJAObeW+W+lVk1RkkRo4wjFVuzfaWYXLMoYWHrkE4r30u1oNQXkjaRdm0uwMgsw7m9lsbWH5YrXtZ166DRsZCgj84RnBmDbSFUuGRF+K4BALEjIxWboeGNJ/CyhXdnkfcxswVE8MhlQLtRckXAtkVrdXzKiyxyQIxKwrGwkAIwpF7jJIv1Fr2tWX/AKf68xzs8j7UCsu3ooLlSWteyjyAXtbOSMXzzt48bZ8NccvKRKo+WV8OeORwPFk3ow6oASUBPrc5tatZPy8UkOo4hKksUaKFVPIzvawFlsTa3rkn0BqdyRhl3x2Pe3Y/L0rScWmh2eHOwYP5dnbP+da8Hpf9XO8utevn6HH2945hLpGmLS6eM7L3KIS/hj0c2x9QAe3SsNHBwcGplxXVR6LT/wANpX2uw+22tvZicWLKAFx6dja3Wtbr+X1TTJO5Mcz3K6ezM+ASf5rbRexF1vkmvpx4GiScxsGDFSMgjBrfTcxx6uEpqEUulgki38S3sLWK+xP9o3odDLOT4cZe3VuiL/uY4H51uNNwFdqL4n2rSOr2V9iKoXrdQb3PXA/K5o1Bq1RY3GMWxjHcH2qszhSRYkgkW7+X5fKsk8MmEmyTbH9kJRuZdpRgCtmBIub4HrjFBstDx5I4xHtLEY2WGxg3xh85BxY9iO1dF5P5taGFEmdpV85vfcyIp6Li7bRkqfNa5FxYHjw9az4ddI1oyyqv4gg3kWCgYtfF+4vc3JoPprR6uOZFkjZXRhdWU3BrJri/L/H5tHDE7MqIwcRo2EdUC2LdT5sqGHRlPUGup8G41HqUDL5W2hmQkbgD0ODYj3FQbWlKVQpSlApSlAq1quq00Hk63rU6/QqI3VF63J/rW6tVjLSzU1DNfpyI4rjIKXB/LNQ7m7lsvJfToASlygsoNja69Beuo8T0J [... cut ...] QKUpQKUpQKUpQKUpQKUpQKoRVaUGPPDuUjpetPqdBsiI6nPT3v/AN1IK8Z4Q4KnoalhEN1nDkm0/hSX2uo6GxFrEEfUCuY8S4DqdFIZA1lB2pIjWJJyBa9x8Oe3zrtfFNMEjAXoMfpWv12hSRPDkUMrrkH6fkfes9NduI8Q5g1silP4mQJe7gOVJ7XJGfpe3tXQ/wDS2C2j23vaRv1Cn+tWwf6bw+IWeV2S4Ij+HHozA3I+QFdB4Vw6OJAkaKijsosP/wBrUZWwaQmthFpgK9lW1XVcRQLVaUoFKUoK0pSiqGlDSgrSlKBSlKBSlKBSlKBSlKBSlKBSlKBSlKBSlKCx0BFiLitfxKAnaVF7Yx72rZ0oNS0O2QD5VslQCvGbT7mVr9LX+hvWVQUpVaUFKVWlBSlVpQKUpQUNKGlBWlKUClKUClKUClKUClKUClKUClKUClKUClKUClKUClKUClKUClKUClKUClKUFDSq0oFKUoP/2Q==)

yamenk-gribaudo, 2023-08-05:
Thanks for the quick response,

> What do you get at the other pin's you have chose for output. You might get the same. 
I get the same, yes. Its sending all the messages to all devices

>To stop the output at a TX pins and to resolve a possible conflict at the input pins, you have to disable the rx, tx pins switching input and output to another pin pair. Just declare them as GPIO input (rx) and output(tx) pins using machine.Pin. Set the tx to high.

I will try that and post my result here.

yamenk-gribaudo, 2023-08-05:
> To stop the output at a TX pins and to resolve a possible conflict at the input pins, you have to disable the rx, tx pins switching input and output to another pin pair. Just declare them as GPIO input (rx) and output(tx) pins using machine.Pin. Set the tx to high.

wow, this worked. May I ask what was happening?  To my understanding, only the hardware UARTs have the capacity to save message. Where are the messages being saved after I deinit the uart?

robert-hh, 2023-08-05:
The UARt does not store any messages. It's more simple. The ESP32 has an internal kind of switchboard, which connects the signals of modules like UART or GPIO to the pins of the chip. You can connect e.g. the UART tx signal to several pins. That happens here. Connecting tx to pin 17 does not remove the previously made connection to pin 15. That is done by connection Pin15 to GPIO.

yamenk-gribaudo, 2023-08-05:
Thanks, very clear. For the ones ending here in the future, I end uo with something like this:
```
def send_msg(rx, tx, msg):
    uart.init(rx=rx, tx=tx)
    uart.write(str(msg) + "\n")
    while uart.txdone() is False:
        time.sleep_ms(1)
    p = Pin(tx, Pin.OUT)
```

robert-hh, 2023-08-05:
Do not forget to detach the rx pin. While it seems fine to connect UART TX to several pins, connection RX to several pins leads to an undefined state, where several external outputs are connected together.

yamenk-gribaudo, 2023-08-05:
Something like this?
```
def send_msg(rx, tx, msg):
    uart.init(rx=rx, tx=tx)
    uart.write(str(msg) + "\n")
    while uart.txdone() is False:
        time.sleep_ms(1)
    Pin(tx, Pin.OUT)
    Pin(rx, Pin.OUT)
```

robert-hh, 2023-08-05:
`Pin(rx, Pin.IN)`
Otherwise the output of the ESP32 would drive against the output of the connected serial device.