micropython/micropython #12173: Welcome (Please read before posting)

Cognisant1, 2023-08-05:
### Discussed in https://github.com/orgs/micropython/discussions/9111

<div type='discussions-op-text'>

<sup>Originally posted by **jimmo** August 26, 2022</sup>
## Welcome to the MicroPython forum, hosted on GitHub Discussions! :snake: :octocat: 

### :wave: Getting started
Here are some useful tips:
- Anyone can post and reply to discussions, you just need a GitHub account.
- Please search before posting new issues.
- You can use [GitHub Flavoured Markdown](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/about-writing-and-formatting-on-github) to format your posts. In particular use `` ```py `` ... `` ``` `` around your code snippets to get syntax highlighting, see the section below.
- When describing a problem please include information about which version of MicroPython, where you downloaded the firmware from, and which hardware/board you're using. If there's wiring involved, please include a photo.
- When asking your question, please describe what you've already tried. Include code snippets where possible!
- Please do not post answers generated using LLM tools like ChatGPT. Their knowledge of MicroPython (as distinct to Python) is fairly limited and the answers tend to be misleading.

There's also a huge wealth of knowledge at the [previous MicroPython forum](https://forum.micropython.org/).

### :bucket: Categories

If you have a question, please use one of the "Ports" or "Q&A" categories (depending on whether your question is specific to a particular port/board or more generally about MicroPython). For more open-ended discussion see the "General" and "Ideas" categories. We'd also love to hear about your projects too, see the "Show and tell" category.

If you're not sure if something is a bug, it's fine to create a post here. The moderators can transfer the post into an issue if necessary.

### :gear: Code formatting

[GitHub Flavoured Markdown](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/about-writing-and-formatting-on-github) uses [two different ways to format code](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code).

The first is `print("inline formatting")` which is where a code snippet is inline in a paragraph. This uses a *single backtick* around the code -- e.g. `` `print("inline formatting")` ``. This is useful for very short expressions and referencing variable/function names.

For blocks of code, use *triple backticks* with an optional language specifier. For example

<pre>
```py
while True:
  pin.toggle()
  await asyncio.sleep_ms(100)
```
</pre>

which will produce the following:

```py
while True:
  pin.toggle()
  await asyncio.sleep_ms(100)
```

*In almost all cases, if you write your code (or paste it in from your editor), select it, then click the `<>` button in the toolbar (or press Ctrl-E), then the GitHub editor will insert the right number of backticks for you.*

### :mailbox_with_mail: Mail filtering

If you are subscribed to the [micropython repo](https://github.com/micropython/micropython) and would like to filter discussions separately from issues and PRs, you can match on the message-id which will contain `"micropython/micropython/repo-discussions"`.

For example in GMail, the following two filters will put Issues & PRs into one label, and Discussions in another. Use the "Includes the words" box to enter the query:

`listid:micropython.micropython.github.com -"micropython/micropython/repo-discussions"`
&nbsp; &nbsp; :arrow_right: Archive, Label: micropython
`listid:micropython.micropython.github.com "micropython/micropython/repo-discussions"`
&nbsp; &nbsp; :arrow_right: Archive, Label: micropython/discussions

### :heart: Community Guidelines

Please see the [MicroPython Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) as well as the [GitHub Community Guidelines](https://docs.github.com/en/site-policy/github-terms/github-community-guidelines).</div>

Cognisant1, 2023-08-05:
I can’t seem to access the EXT_GPIO pins on Nano RP2040. It says “expecting a regular GPIO Pin”. Can anyone help?

robert-hh, 2023-08-05:
EXT_GPIO pins are only available at the RP2040 PICO_W. You can access them there through the Pin.cpu class, or as string argument to the Pin method like Pin("EXT_GPIO0"). EXT_GPIO0 can as well be addresses as "LED", e.g. Pin("LED").

robert-hh, 2023-08-05:
b.t.w: Please delete your first post, copying Jimmo's post. It seemed to be made in error.

jimmo, 2023-08-05:
@Cognisant1 To raise issues against MicroPython please use the [Issue Tracker](https://github.com/micropython/micropython/issues/new/choose).

As @robert-hh has explained above, EXT_GPIO pins are only on boards where additional pins are provided via some interface (such as the WiFi chip on the Pico W).

closed 2023-08-05 by jimmo

Cognisant1, 2023-08-06:
ThanksSorry for raising an issue in the wrong place but thank you so much for responding. I thought the Arduino nano RP2040 did have these extended pins via the Nina module in the Board (see https://github.com/micropython/micropython/blob/67fac4ebc53db6337008ba06df7932faec80f57c/ports/rp2/boards/ARDUINO_NANO_RP2040_CONNECT/pins.csv?plain=1)Sent from my iPhone

Closed #12173 as completed.

robert-hh, 2023-08-06:
Yes, indeed, I forgot that one. They are listed in pins.csv as and contained in the Pin.cpu and Pin.board classes. Same rule as for the PICO_W for using them.

Cognisant1, 2023-08-06:
So sorry to be a bother,  but I’m using Thonny with the nightly build of .uf2 and when I try Pin(‘EXT_GPIO3’) or Pin (36) I get the response invalid pin. Don’t know why Sent from my iPad

robert-hh, 2023-08-06:
That may be a Thonny problem (again). I just uploaded the most recent nightly build, using the simple method by
- Downloading the file from https://micropython.org/download/ARDUINO_NANO_RP2040_CONNECT/
- Pushing reset twice.
- Waiting until the `RPI-RP2` drive appears,
- copying the .uf2 file to the device and
- Waiting until the `RPI-RP2` drive disappears.

Then I get in REPL using a simple terminal like Putty or mpremote with Ctrl-B the version prompt. And can use the ext pins.
```
MicroPython v1.20.0-327-gd14ddcbdb on 2023-07-27; Arduino Nano RP2040 Connect with RP2040
Type "help()" for more information.
>>> from machine import Pin
>>> p36=Pin('EXT_GPIO3')
>>> type(p36)
<class 'Pin'>
```

Rule of Thonny: If something fails when using Thonny, try the same NOT using Thonny.

Cognisant1, 2023-08-06:
This is very helpful. When I try exactly what you’ve written it works fine. But when I add adc = ADC(p36) the error is ValueError: expecting a regular GOIO PinSent from my iPhone

<class 'Pin'>

Rule of Thonny: If something fails when using Thonny, try the same NOT using Thonny.

robert-hh, 2023-08-06:
The EXT_GPIOxx pins are not pins of the RP2, but pins of the NINAW10 ESP32 based WiFi module, and their operation requires communicating with the ESP32. Therefore not all functions of a RP2 pin may be available. If it rejects use for ADC, then it this is not implemented. Looking at the code, that seems indeed not to be supported. Although it could easily be implemented, since it is already supported by the NINAW10 module.

Cognisant1, 2023-08-06:
Any way I can implement it in micropython. I don’t particularly want to go back to Arduino. I was enjoying micropython!!Sent from my iPhone

Cognisant1, 2023-08-07:
Hi RobertThis is a very valuable email conversation. I’m sorry I put it in the wrong place. But I’m fascinated to hear that the ADC on the NINAW10 module might be relatively easy to implement. Could you possibly convert my query into a request for the ADC implementation to be achieved? Best wishesAdrianSent from my iPhone

robert-hh, 2023-08-07:
>  I’m sorry I put it in the wrong place. But I’m fascinated to hear that the ADC on the NINAW10 module might be relatively easy to implement. 

ADC on the NINA module is already implemented. Only it is not used by the machine.ADC class of the RP2. That connection is missing. Maybe someone from Arduino can contribute it.