micropython/micropython #12191: Pico W will not connect on WiFi

LordInchcape2, 2023-08-08:
I have followed several forum threads on Pico W running MicroPython not connecting on WifI without a clear resolution.
My Pico W will get the list of available networks but will not connect.
If I run the Pico W in the Arduino Gui it works perfectly. Connects every time.
So I know the problem is not my router or the hardware of the Pico W. It must be the MicroPython support.
Any help gratefully received.

LordInchcape2, 2023-08-08:
I had something very similar. 
But I have just tried the exact code example you sent and it will still not connect.

robert-hh, 2023-08-08:
Which Micropython version is installed?

LordInchcape2, 2023-08-08:
MicroPython v1.20.0 on 2023-04-26   Raspberry Pi Pico W with RP2040

robert-hh, 2023-08-08:
You could try a recent nightly build, just in case there were changes to that topic.

LordInchcape2, 2023-08-08:
Thanks I will try that. I will also try micropython on one of my ESP32's.

LordInchcape2, 2023-08-09:
Apologies. A major case of finger trouble!!
The password on some of my scripts had picked up a stray character.,

closed 2023-08-09 by LordInchcape2