micropython/micropython #12216: ports/unix: please implement machine.unique_id()

paravoid, 2023-08-11:
What triggered this was my attempt to use [micropython-mqtt](https://github.com/peterhinch/micropython-mqtt/) from the Unix port, in order to run some tests with my code before moving this to an actual board.

The module uses a few things that are not implemented in the Unix port, and probably needs to be ported itself.

One of the issues, however, is `machine.unique_id()`, which seems... basic enough for the Unix port to have it.

One strategy could be to use gethostid(), which is deprecated in most Unix systems, as it's a 32-bit integer (not very unique). Another one could be to use `/etc/machine-id`, a UUID, which exists in -at least- modern Linux systems.

A search for /etc/machine-id yields:
https://codesearch.debian.net/search?q=%2Fetc%2Fmachine-id&literal=1
Most seem to be falling back to /var/lib/dbus/machine-id.

andrewleech, 2023-08-11:
At the moment the Unix port doesn't have any kind of machine module. I've spoken to a few people about starting a (pure python) machine module for Unix with mostly mocked or faked peripherals for use in unit testing, though things like this would certainly make sense there too.

paravoid, 2023-08-12:
I see a few things implemented already in modmachine.c, like machine.idle() etc. Do you mean that it wouldn't make sense to implement this kind of thing in C, but instead add a pure Python module for such purposes?

andrewleech, 2023-08-12:
Oh wow, turns it I missed that addition... yeah all of a sudden there is a machine module on Unix port. There wasn't one let time I started a new project.
```
MicroPython v1.20.0-dirty on 2023-05-14; linux [GCC 12.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import machine
>>> dir(machine)
['__class__', '__name__', '__dict__', 'PinBase',
'Signal', 'idle', 'mem16', 'mem32', 'mem8', 'time_pulse_us']
```

Regardless, yeah I'd think it's better to add these kinds of things in a python extension package rather than C. It is possible to have extension package in micropython-lib that add to builtins (except for a couple of special builtins that can't be extended).

jonnor, 2024-07-13:
On Linux, there is a FreeDesktop standard for machine identifier, via a `/etc/machine-id` file - ref https://www.freedesktop.org/software/systemd/man/latest/machine-id.html
That will cover most Linux systems from the last years, probably? So that could easily be read in, be it in the machine module, or in some Python code. There are of course a range of other options, ref http://0pointer.de/blog/projects/ids.html

Since 2023 it seems to also be available in FreeBSD (https://cgit.freebsd.org/src/commit/?id=17333d92643d). However it may be good to still support fallback to the DBus machine ID, which was used previously. `/var/lib/dbus/machine-id`

For Mac OS, there seems to be a ioreg command that one should use. Ref https://apple.stackexchange.com/a/342043
So that could be called using `os.system()`.