micropython/micropython #12221: Reading memory with mem32 can give negative values?

mjaspers2mtu, 2023-08-13:
On my esp32 running MicroPython v1.20 I get strange behavior from machine.mem32[_address_]

when reading blocks of memory I sometimes come across negative values.
here is some simple code that reads from rtc memory, and shows the bug:
```py
import machine
for addr in range(0x50000000, 0x500000FF, 4):
    print(machine.mem32[addr])
```
result from above program:
```py
982129596
-598711246
-1442082364
-1561223426
1761193693
2024300204
-1877603854
611949399
...
```
even after converting the memory integer values to hex, the negative remains, and the hex value is incorrect, I found this returns the correct memory value in hex:
```py
addrval=mem32[addr] if mem32[addr]>0 else mem32[addr] & 0xffffffff
```
My bug report is this:
Reading memory should not return a negative value, I think this could be fixed by assigning the correct variable type.
eg: unsiged_long vs long

dlech, 2023-08-13:
I think this is normal for Python. It doesn't know about signed vs. unsigned integers so always treats them as signed. If we tried to return unsigned here, then ports that use `MICROPY_LONGINT_IMPL_NONE` would not be able to work.

You can simplify your workaround to `addrval = mem32[addr] & 0xffffffff` to avoid reading twice.

mjaspers2mtu, 2023-08-13:
Hello dlech,

Thank you for the reply and improvement of my code (tested and works)!
Is it correct that doing this mask operation of all 32 bits and putting it in a new variable changes the data type? (to not be signed anymore)
Further mem16 and mem8 do not show this behavior.

Further, I do not want to break any code of other ports, 
But... it really does not feel correct that I would need to do **any** arithmetic operations when inspecting memory.
Is this only esp related?

dlech, 2023-08-13:
> Is it correct that doing this mask operation of all 32 bits and putting it in a new variable changes the data type?

No, it doesn't change the data type, it is `int` in any case. But it does create a new object if the original was negative (or more strictly speaking, if the original does not fit in MP_SMALL_INT).

> Is this only esp related?

This will happen on any 32-bit port.