micropython/micropython #12228: Consider renaming `MP_STATE_PORT`

jimmo, 2023-08-15:
Before `MP_REGISTER_ROOT_POINTER` was added, a port could define additional root pointers in its mpconfigport.h. It made sense to refer to them via a distinct macro, hence `MP_STATE_PORT`, even though all ports alias that to `MP_STATE_VM`.

Now the `MP_REGISTER_ROOT_POINTER` feature can be used to implement root pointers for modules that mpstate.h does not know about, such as modules in extmod, or user c module. We could also consider moving some config-optional stuff that's currently in mpstate.h to use `MP_REGISTER_ROOT_POINTER`. But things defined with `MP_REGISTER_ROOT_POINTER` are no longer "port". So for example, in `modbluetooth_nimble.c` we use `MP_STATE_PORT(bluetooth_nimble_root_pointers)`, but the only thing that's "port" about them is that bluetooth can be enabled by the port/board.

So I see three options:
1. Remove `MP_STATE_PORT` and just change everything to use `MP_STATE_MEM`.
2. Replace `MP_STATE_PORT` with `MP_STATE_ROOT_POINTER` and have that aliased to `MP_STATE_VM` in a common location (i.e. not ports' `mpconfigport.h`).
3. Same as (2) but move the generated root pointers into their own struct, and add a fourth field (between `thread` and `vm`) on `mp_state_ctx_t`, and have `MP_STATE_ROOT_POINTER` access that specifically.

(My vote is for (3), these variables are no more "vm" than they are "port", and should find a new home )

FWIW, This came up as a question on Discord because the embed port does not define `MP_STATE_PORT`.

jimmo, 2023-08-15:
cc @dlech

dlech, 2023-08-15:
Number 3 sounds good. 

I guess we don't really need `MP_STATE_PORT` for anything that is not a root pointer since the code can just use a local static global variable for that, so seems fine to remove it.

If we are lucky, it might even reduce code size a bit by having a smaller offset to some of the fields.

jimmo, 2023-09-13:
See https://github.com/micropython/micropython/pull/12436 which implements option (3).

@dlech sadly this did not lead to a code size decrease (the opposite it seems). So maybe I will revert the change to put it into a separate structure (but at least in the meantime that change let us find the cases where `MP_STATE_VM` was being used for `MP_REGISTER_ROOT_POINTER` fields).