micropython/micropython #12253: No more GENERIC_SPIRAM board?

Tangerino, 2023-08-17:
Where is the GENERIC_SPIRAM board definition (ESP32)?
It was merged into another board type?

```bash
ARDUINO_NANO_ESP32   GENERIC_S3           LOLIN_S2_PICO        UM_FEATHERS2         UM_TINYPICO          deploy_c3.md         manifest.py          sdkconfig.base       sdkconfig.spiram_sx
GENERIC              LILYGO_TTGO_LORA32   M5STACK_ATOM         UM_FEATHERS2NEO      UM_TINYS2            deploy_s2.md         manifest_test.py     sdkconfig.ble        sdkconfig.usb
GENERIC_C3           LOLIN_C3_MINI        OLIMEX_ESP32_POE     UM_FEATHERS3         UM_TINYS3            deploy_s3.md         pins_prefix.c        sdkconfig.spiram
GENERIC_S2           LOLIN_S2_MINI        SIL_WESP32           UM_PROS3             deploy.md            make-pins.py         sdkconfig.240mhz     sdkconfig.spiram_oct
```

Tangerino, 2023-08-17:
```
MicroPython v1.20.0-328-g01c758e26-dirty on 2023-07-31
```

DvdGiessen, 2023-08-17:
They were merged in 2fbf42d3ad974b7de12823bb5a86df8a08d917ce; now instead of being a separate board they're a "variant" of the generic board. You can select these with the [`BOARD_VARIANT` parameter](https://github.com/micropython/micropython/blob/a18d62e06727e0c424da1f43f80a5b0cb76bcc04/ports/esp32/Makefile#L43).

We should probably update the [README section](https://github.com/micropython/micropython/tree/master/ports/esp32#building-the-firmware) to reflect this.

sosi-deadeye, 2023-08-17:
I fixed may build scripts, because of this change.
The `variants` are stored in `ports/$PORT/$BOARD/board.json`
There are also shell scripts in `tools/autobuild`

Code to generate make commands:
```python
#!/usr/bin/env python3
import json
from collections.abc import Generator
from pathlib import Path

# micropyton is in my home directory
MICROPYTHON = Path.home() / "micropython"
PORTS = MICROPYTHON / "ports"

def get_ports() -> Generator[Path, None, None]:
    for port in sorted(PORTS.iterdir()):
        if port.is_dir() and port.joinpath("boards").exists():
            yield port

def get_boards(port: Path) -> Generator[Path, None, None]:
    for board in sorted(port.joinpath("boards").iterdir()):
        if board.is_dir():
            if board.exists() and board.is_dir():
                yield board

def get_variants(board: Path) -> list[str]:
    try:
        with board.joinpath("board.json").open() as fd:
            return json.load(fd).get("variants", [])
    except FileNotFoundError:
        return []

def make_command(port: Path, board: Path, variant: str | None = None) -> None:
    cmd = cmd = f"make -C {port} -j 8 BOARD={board.name}"
    if variant:
        cmd += f" BOARD_VARIANT={variant}"
        output_dir = port.joinpath(f"build_{board.name}_{variant.upper()}")
    else:
        output_dir = port.joinpath(f"build_{board.name}")

    cmd += f" BUILD={output_dir}"
    sub = f"{cmd} submodules"

    print(sub)
    print(cmd)

for port in get_ports():
    for board in get_boards(port):
        make_command(port, board)
        for variant in get_variants(board):
            # skipping idf3
            if variant == "idf3":
                continue
            make_command(port, board, variant)
    print()

```

Example for esp32 GENERIC with SPIRAM.
(the project is in /home/esp32/micropython)

```
make -C /home/esp32/micropython/ports/esp32 -j 8 BOARD=GENERIC BOARD_VARIANT=spiram BUILD=/home/esp32/micropython/ports/esp32/build_GENERIC_SPIRAM submodules
make -C /home/esp32/micropython/ports/esp32 -j 8 BOARD=GENERIC BOARD_VARIANT=spiram BUILD=/home/esp32/micropython/ports/esp32/build_GENERIC_SPIRAM
```

Tangerino, 2023-08-18:
Sorry but not clear what need to be done to compile for GENERIC_SPIRAM.
This is a kind of change that breaks stuff...

Tangerino, 2023-08-18:
That's what I get. The board has 2MB of RAM

```
MicroPython v1.20.0-379-ga18d62e06 on 2023-08-18; Generic ESP32 module with SPIRAM with ESP32
Type "help()" for more information.
>>> gc.mem_free()
51520
>>> gc.mem_alloc()
14368
>>> 
```

robert-hh, 2023-08-18:
That's the confusing result of one of the recent PRs. gc.mem_free() does not show all memory. micropython.mem_info shows more. And even if mem_free() just shows about 50 k, you can allocate more RAM, up to the physical max. The available RAM will be expanded on request.

Tangerino, 2023-08-19:
Confusing and IMO a regression. I hope it will be fixed. Thank you very much.

closed 2023-08-19 by Tangerino

jimmo, 2023-08-21:
> We should probably update the [README section](https://github.com/micropython/micropython/tree/master/ports/esp32#building-the-firmware) to reflect this.

@DvdGiessen We did, but it's in the subsequent PR which hasn't been merged yet. https://github.com/micropython/micropython/pull/12240

Note that once that PR is merged, it will be

`make BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM`

> Confusing and IMO a regression.

This will of course all be in the v1.21 release notes. And #12240 includes a notice when you try and use the old "GENERIC" board name.

We're also trying to make this _less_ confusing for the S2/S3 boards by just not having a spiram variant at all, the one build just works for both types. Unfortunately this isn't possible for the original ESP32.

It wasn't our intention to have a several-day gap between merging the two halves of this work (#12088 and #12240) but life (and a weekend) got in the way.

> I hope it will be fixed.

It's not clear what you mean here... what do you need "fixed" ?

We often have to make changes to improve consistency, but unfortunately this by its nature means that we have to break the "inconsistent" part.

> That's the confusing result of the recent PR https://github.com/micropython/micropython/pull/12141.

@robert-hh This will definitely take some explaining, but this is a very important feature to make ESP32 support SSL and not be unnecessarily slowed down by large PSRAM. Definitely open to suggestions on further improvements we can make to the API...
I'm not sure it makes sense for `gc.mem_free()` to include un-assigned RAM though.