micropython/micropython #12262: OSError: can't set AP config on ESP8266

inovatorius, 2023-08-19:
Good day!

Error is raised under these circumstances:

1. Enable STA interface and connect to AP.
2. Enable AP interface and set AP channel via `config(channel=13)`
3. `OSError: can't set AP config`

Error is not raised when STA is not connected to AP. Currently, I do not have a short version of my code that can reliably reproduce this error.

MicroPython 1.20.0 on ESP2866.

peterhinch, 2023-08-20:
The following sequence (derived from [this](https://github.com/peterhinch/micropython-mqtt/blob/17782357473e28ce29c16223960e541d9cbfa215/mqtt_as/esp32_gateway/nodes/link.py#L46)) should work:
```python
import network
from time import sleep_ms

sta = network.WLAN(network.STA_IF)
sta.active(True)
while not sta.active():
    sleep_ms(100)
ap = network.WLAN(network.AP_IF)
ap.active(True)
while not ap.active():
    sleep_ms(100)
sta.connect(*credentials)
while not sta.isconnected():
    sleep_ms(100)
```

inovatorius, 2023-08-21:
> The following sequence (derived from [this](https://github.com/peterhinch/micropython-mqtt/blob/17782357473e28ce29c16223960e541d9cbfa215/mqtt_as/esp32_gateway/nodes/link.py#L46)) should work: [...]

Indeed, you have to wait for the interface to become `active()` after you use `active(True)`. Thank you!

closed 2023-08-21 by inovatorius