micropython/micropython #12270: ESP32C3 no output on serial after flashing finished.

weiyshay, 2023-08-20:
Not sure what I am messing up with. It seems that the board hangs and no response to 'mpremote repl' after a flashing.

Commands:

  idf.py -D MICROPY_BOARD=GENERIC_C3 clean
  idf.py -D MICROPY_BOARD=GENERIC_C3 build
  idf.py -p /dev/ttyACM0 flash
  idf.py -p /dev/ttyACM0 monitor

Output:

esp32-dev-micropython#idf.py -p /dev/ttyACM0 monitor
/data/2023/src/esp32c3/micropython/idf/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: monitor
Running idf_monitor in directory /data/2023/src/esp32c3/micropython/micropython/ports/esp32
Executing "/root/.espressif/python_env/idf5.1_py3.10_env/bin/python /data/2023/src/esp32c3/micropython/idf/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/micropython.elf -m '/root/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/data/2023/src/esp32c3/micropython/idf/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...

--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

I am trying to build micropython and flash it to an ESP32C3 board while it seems that there is
no output/REPL on the serial port. I have successfully verified my env by:

```
1. Run the 'hello_world' demo in espressif via 'idf.py build flash'.   
2. Flashing the official mp image 'esp32c3-20230426-v1.20.0.bin' and run some demos via mpremote.
```

My env:
```
- Ubuntu 22.04
- idf: v5.1
- Micropython: 
commit a18d62e06727e0c424da1f43f80a5b0cb76bcc04 (HEAD -> master, origin/master, origin/HEAD)
Author: robert-hh <robert@hammelrath.com>
Date:   Wed Aug 16 08:41:40 2023 +0200
```

Flashing cmd & output:
```
  idf.py -D MICROPY_BOARD=GENERIC_C3 clean
  idf.py -D MICROPY_BOARD=GENERIC_C3 build
  idf.py -p /dev/ttyACM0 flash
  idf.py -p /dev/ttyACM0 monitor
  
esp32-dev-micropython#idf.py -D MICROPY_BOARD=GENERIC_C3 flash
/data/2023/src/esp32c3/micropython/idf/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: flash
Serial port /dev/ttyS0
Connecting.......................
/dev/ttyS0 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP32-C3
Running ninja in directory /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build
Executing "ninja flash"...
[1/225] Performing build step for 'bootloader'
[1/1] cd /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/bootloader/esp-idf/esptool_py && /root/.espressif/python_env/idf5.1_py3.10_env/bin/python /data/2023/src/esp32c3/micropython/idf/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/bootloader/bootloader.bin
Bootloader binary size 0x3fb0 bytes. 0x4050 bytes (50%) free.
[2/223] cd /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/esp-idf/main_esp32c3 && /usr/bin/cmake -E make_directory /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/genhdr && /root/.espressif/python_env/idf5.1_py3.10_env/bin/python /data/2023/src/esp32c3/micropython/micropython/py/makeversionhdr.py /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/genhdr/mpversion.h
[3/10] cd /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/esp-idf/main_esp32c3 && /root/.espressif/python_env/idf5.1_py3.10_env/bin/python /data/2023/src/esp32c3/micropython/micropython/tools/makemanifest.py -o /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/frozen_content.c -v MPY_DIR=/data/2023/src/esp32c3/micropython/micropython -v MPY_LIB_DIR=/data/2023/src/esp32c3/micropython/micropython/lib/micropython-lib -v PORT_DIR=/data/2023/src/esp32c3/micropython/micropython/ports/esp32 -v BOARD_DIR=/data/2023/src/esp32c3/micropython/micropython/ports/esp32/boards/GENERIC_C3 -b /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build -f-march=xtensawin --mpy-tool-flags= /data/2023/src/esp32c3/micropython/micropython/ports/esp32/boards/manifest.py
[4/5] cd /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/esp-idf/esptool_py && /root/.espressif/python_env/idf5.1_py3.10_env/bin/python /data/2023/src/esp32c3/micropython/idf/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/partition_table/partition-table.bin /data/2023/src/esp32c3/micropython/micropython/ports/esp32/build/micropython.bin
micropython.bin binary size 0x179dc0 bytes. Smallest app partition is 0x1f0000 bytes. 0x76240 bytes (24%) free.
[4/5] cd /data/2023/src/esp32c3/micropython/idf/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/data/2023/src/esp32c3/micropython/idf/esp-idf -D "SERIAL_TOOL=/root/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/data/2023/src/esp32c3/micropython/idf/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32c3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;--no-stub;write_flash;@flash_args" -D WORKING_DIRECTORY=/data/2023/src/esp32c3/micropython/micropython/ports/esp32/build -P /data/2023/src/esp32c3/micropython/idf/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32c3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 micropython.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.dev1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 48:31:b7:35:cd:bc
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00010000 to 0x00189fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...
Took 0.10s to erase flash block
Writing at 0x00000000... (6 %)
Writing at 0x00000400... (12 %)
Writing at 0x00000800... (18 %)
Writing at 0x00000c00... (25 %)
Writing at 0x00001000... (31 %)
Writing at 0x00001400... (37 %)
Writing at 0x00001800... (43 %)
Writing at 0x00001c00... (50 %)
Writing at 0x00002000... (56 %)
Writing at 0x00002400... (62 %)
Writing at 0x00002800... (68 %)
Writing at 0x00002c00... (75 %)
Writing at 0x00003000... (81 %)
Writing at 0x00003400... (87 %)
Writing at 0x00003800... (93 %)
Writing at 0x00003c00... (100 %)
Wrote 16384 bytes at 0x00000000 in 0.5 seconds (250.6 kbit/s)...
Hash of data verified.
Erasing flash...
Took 2.33s to erase flash block
Writing at 0x00010000... (0 %)
Writing at 0x00010400... (0 %)
Writing at 0x00010800... (0 %)
Writing at 0x00010c00... (0 %)
Writing at 0x00011000... (0 %)
Writing at 0x00011400... (0 %)
Writing at 0x00011800... (0 %)
Writing at 0x00011c00... (0 %)
Writing at 0x00012000... (0 %)
Writing at 0x00012400... (0 %)
Writing at 0x00012800... (0 %)
Writing at 0x00012c00... (0 %)
Writing at 0x00013000... (0 %)
Writing at 0x00013400... (0 %)
Writing at 0x00013800... (0 %)
Writing at 0x00013c00... (1 %)
...
Writing at 0x00188c00... (99 %)
Writing at 0x00189000... (99 %)
Writing at 0x00189400... (99 %)
Writing at 0x00189800... (99 %)
Writing at 0x00189c00... (100 %)
Wrote 1548288 bytes at 0x00010000 in 49.7 seconds (249.4 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.03s to erase flash block
Writing at 0x00008000... (33 %)
Writing at 0x00008400... (66 %)
Writing at 0x00008800... (100 %)
Wrote 3072 bytes at 0x00008000 in 0.1 seconds (260.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done

  
```

weiyshay, 2023-08-20:
Chat in discord:
https://discord.com/channels/574275045187125269/952412628926472273

robert-hh, 2023-08-20:
Commenting out the line `boards/GENERIC_C3/sdkconfig.c3usb` in the file `ports/esp32/boards/GENERIC_C3/mpconfigboard.cmake` allows to build a firmware with REPL at UART1. These are the type b) boards. Seems like it requires a variant for these boards.

robert-hh, 2023-08-20:
A modified  mpconfigboard.cmake could be:
```
set(IDF_TARGET esp32c3)

set(SDKCONFIG_DEFAULTS
    boards/sdkconfig.base
    boards/sdkconfig.ble
)

if(NOT MICROPY_BOARD_VARIANT STREQUAL "uart")
    set(SDKCONFIG_DEFAULTS
        ${SDKCONFIG_DEFAULTS}
    boards/GENERIC_C3/sdkconfig.c3usb
    )
endif()
```

And then the variant for UART can be built & uploaded with:
`make BOARD=GENERIC_C3 BOARD_VARIANT=uart -j5 deploy`

weiyshay, 2023-08-20:
Thank you very much for your quick response, Robert! It works.

I am not sure if this fix will impact the other options in 'sdkconfig.c3usb' in particular for the 'CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y'. 

I tried to add the option like below but unfortunately it doesn't work. Any practice? 

```
diff --git a/ports/esp32/boards/GENERIC_C3/sdkconfig.c3usb b/ports/esp32/boards/GENERIC_C3/sdkconfig.c3usb
index d9e7c7f61..81d643c01 100644
--- a/ports/esp32/boards/GENERIC_C3/sdkconfig.c3usb
+++ b/ports/esp32/boards/GENERIC_C3/sdkconfig.c3usb
@@ -3,5 +3,5 @@ CONFIG_ESP32C3_BROWNOUT_DET=y
 CONFIG_ESP32C3_BROWNOUT_DET_LVL_SEL_7=
 CONFIG_ESP32C3_BROWNOUT_DET_LVL_SEL_4=y
 CONFIG_ESP32C3_BROWNOUT_DET_LVL=4
-CONFIG_ESP_CONSOLE_UART_DEFAULT=
+CONFIG_ESP_CONSOLE_UART_DEFAULT=y
 CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y

```

robert-hh, 2023-08-20:
Better keep the issue open, since it must be fixed.

jimmo, 2023-08-21:
If I understand correctly, then https://github.com/micropython/micropython/pull/12240#issuecomment-1685573338 should fix this?

closed 2026-08-14 by dpgeorge