micropython/micropython #12289: V 1.20.0 will not build on Arch Linux

adrianblakey, 2023-08-23:
1.  Download the source distro.   
2. cd mpy-cross   
3. make  

```

...
CC ../py/stackctrl.c
../py/stackctrl.c: In function ‘mp_stack_ctrl_init’:
../py/stackctrl.c:32:32: error: storing the address of local variable ‘stack_dummy’ in ‘mp_state_ctx.thread.stack_top’ [-Werror=dangling-pointer=]
   32 |     MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
../py/stackctrl.c:31:18: note: ‘stack_dummy’ declared here
   31 |     volatile int stack_dummy;
      |                  ^~~~~~~~~~~
In file included from ../py/runtime.h:29,
                 from ../py/stackctrl.c:27:
../py/mpstate.h:296:23: note: ‘mp_state_ctx’ declared here
  296 | extern mp_state_ctx_t mp_state_ctx;
      |                       ^~~~~~~~~~~~
cc1: all warnings being treated as errors
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [../py/mkrules.mk:83: build/py/stackctrl.o] Error 
```

dlech, 2023-08-23:
Likely duplicate of #11464.

dpgeorge, 2023-08-23:
Please use latest master, or see https://github.com/micropython/micropython/issues/11464#issuecomment-1543042388

closed 2023-08-23 by dpgeorge