micropython/micropython #12299: asyncio: MicroPython vs. CPython create_task() garbage collection behavior

paravoid, 2023-08-25:
(Not sure if I should label this a docs bug, feature request, or discussion. Apologies if I missed the mark.)

CPython exhibits a bit of a counterintuitive behavior, which requires users to store the result of `asyncio.create_task()`, otherwise face the danger of the task being garbage collected.

Will McGugan of Textual blogged about this here:
https://textual.textualize.io/blog/2023/02/11/the-heisenbug-lurking-in-your-async-code/

Vincent Bernat reported this as a documentation bug against CPython, https://github.com/python/cpython/issues/88831 last year, and subsequently [the CPython documentation](https://docs.python.org/3/library/asyncio-task.html) was adjusted to say:
> Important
> Save a reference to the result of this function, to avoid a task disappearing mid-execution. The event loop only keeps weak references to tasks. A task that isn’t referenced elsewhere may get garbage collected at any time, even before it’s done. For reliable “fire-and-forget” background tasks, gather them in a collection:
```python
background_tasks = set()

for i in range(10):
    task = asyncio.create_task(some_coro(param=i))

    # Add task to the set. This creates a strong reference.
    background_tasks.add(task)

    # To prevent keeping references to finished tasks forever,
    # make each task remove its own reference from the set after
    # completion:
    task.add_done_callback(background_tasks.discard)
```

Is MicroPython exhibiting the same behavior? If so, should the documentation be adjusted?

1. The MicroPython documentation [provides an example](https://docs.micropython.org/en/latest/library/asyncio.html) that does **not** store a reference, basically what CPython warns users not to do. Is this a misleading example or a non-issue in MicroPython? Should the example be CPython-friendly anyway though? Should the documentation explain this difference?
2. Even if one wanted to be better-safe-than-sorry and write portable code, the CPython-recommended code above does not work under MicroPython, resulting in a `TypeError: unsupported type for __hash__: 'Task'`. Should `__hash__` be added for Tasks? Or should the documentation mention some other way to do this?

Not a MicroPython bug per se, but also worth mentioning: @peterhinch's (excellent!) [async tutorial](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md) also has examples where Tasks are not stored. At one point it's mentioned that "_[t]he .create_task method returns the Task instance which **may** be saved for status checking or cancellation_" (emphasis mine). May, or should?

peterhinch, 2023-08-25:
I have added [this note](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#note) to the tutorial pending definitive confirmation from the maintainers.

jimmo, 2023-08-30:
> Is this a misleading example or a non-issue in MicroPython?

It is both of these things :) We should be providing examples that are also good/valid CPython, but yes, MicroPython does not have this problem because of what Peter said, (and because it does not support weak references).

Related to this, CPython's reference counting also enables CPython to do immediate detection of un-waited tasks when the task goes out of scope.

> resulting in a TypeError: unsupported type for __hash__: 'Task'. Should __hash__ be added for Tasks?

Tasks do implement `__hash__`, but if you're using the optimised Task implementation, I think this only got implemented in ~May when we added implicit hash by default for all built-in types. So you might find this is already fixed in the latest builds (and if it isn't please let us know because that would be surprising).

peterhinch, 2023-08-30:
@jimmo Thanks for that. I will amend the sample scripts in the tutorial to save task references for the CPython workround.

I can confirm that tasks can now be added to a `set`.

paravoid, 2024-01-15:
Status update, as of today, 2024-01-15, tested with MicroPython v1.22.1:
- Tasks are now hashable and able to be added to a set(), as also reported above.
- `task.add_done_callback(background_tasks.discard)` (from the CPython doc) does not work, as `add_done_callback()` is not yet implemented.
- https://docs.micropython.org/en/latest/library/asyncio.html has not been adjusted to be CPython-friendly, by providing example code that stores tasks to a set.
- @peterhinch's async tutorial has been updated to add a CPython compatibility note, also linking to this issue (thanks!)