micropython/micropython #12311: docs: How to use io module

yamenk-gribaudo, 2023-08-27:
Hey, I'm using micropython 1.20 with a generic esp32 devkit board. I have read the [io module](https://docs.micropython.org/en/latest/library/io.html) documentation and I dont understand how it work. 

First: It seems like it doesnt matter the argument passed to the io.StringIO class:
```
import io
a = io.StringIO(1)
a.write("hola")
```

I would expect this operation fails because the buffer should be size 1, but not only it doesnt fail, but I was able to allocate the whole word and print it using `a.getvalue()`.

Second: Why the `a.read()` is just printing an empty string? It seems I can use the `write()` method but I cant use the `read()` method. Altough I wa able to get the value inside the buffer using the `getvalue()` method.

Third: How can I flush the buffer? Seems like the `flush()` do not work. I'm using it like this:
```
import io
a = io.StringIO(1)
a.write("hola")
a.flush()
print(a.getvalue())
```
But this prints the "hola".

Thanks in advance for any answers to my questions. And if there is way to update the docs in https://docs.micropython.org I could work on that,

peterhinch, 2023-08-27:
See [CPython docs](https://docs.python.org/3/library/io.html#io.StringIO) for args. Passing an int does not define the buffer size.

rkompass, 2023-08-27:
You need to `a.seek(0)` before reading, otherwise your position is at the end of the written text and an `a.read()`  tries to read from there.
`a.tell()` will tell the position. `a.getvalue()` always returns all of the buffer written, even several times.

yamenk-gribaudo, 2023-08-27:
@peterhinch thanks for the answer. What is the default size of the buffer then? Does reusing it change it size? If so, how can I avoid memory fragmentation issues in the case where I have to use a buffer like object?

yamenk-gribaudo, 2023-08-27:
@rkompass thanks for the answer, I was able to use `read()` without a problem now. Do you know how can I erase the value of the object? And what the `flush()` method is supposed to do?

rkompass, 2023-08-28:
You may overwrite parts of the buffer, just seek to the position you want to write at. I am not sure about flush(). It should be just there to be compatible to a file. It is needed for a file because after a write there the changes may not be on the disk yet. With the io this is not an issue.
The main idea is to have something that may be a replacement for a file, although it's in memory, as stated in the docs, which are perhaps not very extensive at this point. So it's not a buffer or list, where you can remove the first half of it.
CPython docs recommend to select desired content and write it in to a new io object and deleting the old, if necessary to shrink the object.

Having that clear you might mark this issue as closed or suggest a set of explanations and/or example code and do a PR for the docs or formulate an enhancement request.

closed 2023-08-29 by yamenk-gribaudo