micropython/micropython #12317: pll_cap_ext 10

MATTYGILO, 2023-08-28:
Using:
ESP32S3 OCTAL SPIRAM
ESP IDF 4.4

I run this code and it works most times, sometimes when the board is reset using `` it repeatedly prints `pll_cap_ext 10` and I unable to connect to the board via BLE:

```
ble = ubluetooth.BLE()
ble.active(True)  # `pll_cap_ext 10` is printed hundreds of times whilst this is run
```

Then when I unplug the device from power and plug back in it works again

The error `pll_cap_ext 10` means that the device is failing to broadcast to wifi, I assume this occurs during BLE as the use simmilar resources etc.

MATTYGILO, 2023-08-29:
My device now get's frozen when running `ble.active(False)`.

It freezes with this red output `E (629860) NimBLE: ble_hs_stop: failed to terminate connection; rc=2`

MATTYGILO, 2023-09-05:
I am having similar problems to these threads: 

[https://forum.micropython.org/viewtopic.php?f=18&t=12696&p=71159#p71159](https://forum.micropython.org/viewtopic.php?f=18&t=12696&p=71159#p71159)

jonnor, 2024-08-24:
Does this issue still happen on MicroPython 1.23 - with the releases on the Download page? If so, can you provide complete code to reproduce?

projectgus, 2024-08-27:
Looks like this was fixed in ESP-IDF v4.4.4 ([bug report in Chinese](https://forum.esp32.com/viewtopic.php?t=33875)), so shouldn't be present in the current release which is several versions newer now.

closed 2024-08-27 by projectgus