micropython/micropython #12325: A very weird Issue in the esp32s3 port caused by recent update

nspsck, 2023-08-28:
It is very hard to describe what the actual issue is... And I could not locate where the Issue lies. This probably happened just last week, because at the time of #12237, almost the same code could be excuted repeatedly.

Basically, if you run the following code 2 times in a row in Thonny
```python
#Code
from micropython import mem_info

mem_info()
b = bytearray(150 * 1024)
mem_info()

#Output
stack: 688 out of 15360
GC: total: 64000, used: 2416, free: 61584, max new split: 8257536
 No. of 1-blocks: 35, 2-blocks: 7, max blk sz: 18, max free sz: 3836
stack: 688 out of 15360
GC: total: 217856, used: 155344, free: 62512, max new split: 8126464
 No. of 1-blocks: 23, 2-blocks: 5, max blk sz: 9600, max free sz: 3839
```
you will notice, that **the first time it goes very fast and the second time it takes a really long time for it to complete**. This behavior can be observed on a lot boards **using N16R8 module**. Like: LOLIN S3, LOLIN S3 Pro, WeAct Studio esp32s3 N16R8, Official ESP32-S3-DevKitM-1.

The same piece of code causes some boards (Lilygo T-Display S3 AMOLED) to hard reset automatically. 

For the testing, I always used the native USB.

To build the firmware I used:
```shell
make BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT USER_C_MODULES=#A display driver written in C
```
Note: I tried simplely delete the line in [this file](https://github.com/micropython/micropython/commit/326dfd2a8579828c4159d965c6a41883a2a90029), I also tried to set it to `0`.
```
#define MICROPY_HW_ENABLE_UART_REPL (1)
```
Neither approach had any effects on this Issue.

projectgus, 2023-08-30:
Hi @nspsck ,

This sounds like it might be the same as this issue - https://github.com/thonny/thonny/issues/2886 

However, just in case, there are some recent esp32 memory allocation changes as described in https://github.com/orgs/micropython/discussions/12316 . This shouldn't cause more than a very short delay (much less than a second) between iterations, though. (Previously, the whole PSRAM could fill up before any garbage collection pass - meaning no garbage collection pauses would happen until it filled up and then one long garbage collection pause would happen. Now, running code like this will trigger frequent smaller and much shorter garbage collection pauses.)

nspsck, 2023-08-30:
Hi projectgus,

> This sounds like it might be the same as this issue - thonny/thonny#2886

Thank you for your response. I can confirm (well I hope...) that this is actually the case. As both using `screen` and the `repl` in thonny with the following code causes no issue what so ever.
```python
import machine

b = bytearray(1024 * 100)

machine.soft_reset()
```
> However, just in case, there are some recent esp32 memory allocation changes as described in https://github.com/orgs/micropython/discussions/12316 . This shouldn't cause more than a very short delay [...]

Would there still be a possibility, that this is a `gc` related problem, if something like
```python
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
b = bytearray(1024 * 100)
```
runs perfectly? The code is "at one go".

peterhinch, 2023-08-30:
I tested this using mpremote with the latest firmware and couldn't replicate the problem. It does look like Thonny is the culprit.
```
MicroPython v1.20.0-396-g1dedb65e6 on 2023-08-24; FeatherS3 with ESP32-S3
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from micropython import mem_info
=== 
=== mem_info()
=== b = bytearray(150 * 1024)
=== mem_info()
=== 
stack: 672 out of 15360
GC: total: 64000, used: 1840, free: 62160, max new split: 8257536
 No. of 1-blocks: 23, 2-blocks: 3, max blk sz: 18, max free sz: 3836
stack: 672 out of 15360
GC: total: 217856, used: 154960, free: 62896, max new split: 8126464
 No. of 1-blocks: 13, 2-blocks: 3, max blk sz: 9600, max free sz: 3836
>>> 
```

nspsck, 2023-08-30:
> I tested this using mpremote with the latest firmware and couldn't replicate the problem. It does look like Thonny is the culprit.

Ye, Thonny is most likely the cause for this problem. Time to bothering them!

closed 2023-08-30 by nspsck