micropython/micropython #12337: New machine.CAN driver

projectgus, 2023-08-30:
## Summary

Planned cross-platform `machine.CAN` driver for Micropython.

## Background

MicroPython currently only has one driver for a CAN bus controller, the [pyb.CAN](https://docs.MicroPython.org/en/latest/library/pyb.CAN.html) driver on STM32. Various folks have contributed functional and useful pull requests with CAN drivers for other ports. Many implement a `machine.CAN` interface. Quite sensibly, these drivers have all based their new APIs on the `pyb.CAN` driver. These efforts are hugely appreciated 🙏.

@jimmo, @dpgeorge, and I discussed this recently. We all agree MicroPython should have a cross-platform `machine.CAN` interface. None of us want to use the `pyb.CAN` API for this:

* The conventions used by `pyb.CAN` don't match the best conventions of `machine` APIs.
* It's not very ergonomic, compared to libraries such as [CircuitPython canio](https://docs.circuitpython.org/en/latest/shared-bindings/canio/index.html) or [python-can](https://python-can.readthedocs.io/en/stable/index.html).
* Some functions assume behaviour of the ST CAN hardware peripheral that is not common across platforms. Trying to adapt these to each port in turn is messy and difficult for both users and maintainers.

## Goals

MicroPython should have a `machine.CAN` interface which is:

* Similar to other cross-platform `machine` APIs.
* Relatively easy to use.
* Still fast and light on resource usage.
* Easy to implement on new hardware platforms.
* Easy to extend for use with `asyncio`.
* Easy to integrate with higher-level CAN-based driver layers such as canopen.

Also:

* The interface should be easy for the existing un-merged driver PRs to adapt to.

## Plan

We plan to do the following:

1. [ ] Write a `machine.CAN` API as a docs PR, discuss with other maintainers and the community, and merge it.
2. [ ] Implement common cross-port interface helpers as `extmod/machine_can.c` (similar to `extmod/machine_pwm.c`, etc).
3. [ ] Create a compliant `machine.CAN` implementation on STM32 (this will be a thin wrapper around `pyb.CAN`, which will remain for legacy compatibility.)
4. [ ] Work with the existing PR authors as needed/wanted to merge `machine.CAN` drivers for new ports.

GitHub sponsor funds have been allocated for the first three steps, and I plan to work on it. Currently other work (USB device driver functionality) is the priority, but this is the next priority after that.

This issue post will be updated with overall status of this effort.

IhorNehrutsa, 2023-08-30:
I am ready to correct **port/esp32: Add CAN(TWAI) driver** #12331 to this API.

kwagyeman, 2023-08-30:
Awesome!

kwagyeman, 2023-08-30:
Some comments on Transmit and Receive:

I previously worked at Embark Trucks doing a lot of their embedded system design. I have seen how to use CAN effectively and have thoughts on what works well and what does not.

CAN technically is a message buffer interface system. Generally, most embedded systems do not actually use FIFO based approaches when sending or receiving CAN data. Typically, the message buffer allocation for RX/TX buffers is fixed and pre-allocated by your code. The processor then determines when message buffers have new data and it reads that and processes it.

However, this kind of interface is very foreign when compared to Ethernet/WiFi/UART which all have stream based approaches. Given this, most users don't know what to do with message buffers and will be confused on how to use them correctly if they have never used CAN before.

So, the FIFO based stream interface makes CAN easier to use for beginners. Given this, I support that we should use a fifo based interface over a RAW message buffer based interface.

Moving, send/receive are easy for users to understand. However, it's important that any CAN driver that's implemented respect message ordering with FIFOs. E.g. on my IMXRT port recv() messages that pass through the CAN filters will be in order given what was sent on the bus. For transmit, I explicitly re-use the same message buffers for transmit if the CAN id matches in-order to keep transmit message ordering too. I am pointing out this requirement as if transmit order is not kept it makes it impossible to perform any CAN message protocols (which exist). The STM32 PYB.CAN port does not implement TX message ordering for it's basic CAN (non-fd) transmit method (Message ordering between different CAN ids is generally not required).

kwagyeman, 2023-08-30:
As for masks on filters.

This is very rarely used. Typically, you are just looking for a few IDs that you have filters setup for. I would drop support for masking message filters. We never used this in practice on any of our MCUs. You either had a message you were looking for or not. Same for a range of IDs.

obdevel, 2023-09-01:
Could we consider how the API might support external CAN controllers as well ? I currently maintain my own MP driver for the ubiquitous MCP2515 SPI chip but I'd be amenable to changing the interface to meet the new standard, particularly if it meant my application code were more portable to architectures with on-chip CAN. I also maintain a number of Arduino libraries for various CAN implementations. The lowest common denominator abstract interface is init, available, receive and send. The CircuitPython canio interface/base class seems similar.

andrewleech, 2023-09-01:
@obdevel external controllers seem relevant to me too, surely a common CAN interface should be able to support them too! 

Could you share some details about the public API you currently have as a point of comparison? 

I'm personally invested in promoting an api that matches desktop `python-can` simply to make supporting other existing higher level can libraries, but I'm not experienced with CAN so looping people who do actively use it into the conversation makes sense to me.

kwagyeman, 2023-09-01:
I'd recommend to never use the MCP2515. It actively randomly corrupts CAN packets. If you are building any application in the world using that part will lead to intermittent failures happening. It may be possible to write a driver such that this doesn't happen... but, I repeatedly ran into that chip failing to register packets sent on the CAN bus to the host MCU and also saw events where it would corrupt nominal data on the bus during transmit.

obdevel, 2023-09-01:
> @obdevel external controllers seem relevant to me too, surely a common CAN interface should be able to support them too!
>
> Could you share some details about the public API you currently have as a [...]

From my experience maintaining Arduino libraries for the past few years, the underlying implementations - whether on-chip or external - are so different that any API is going to be pretty abstract. As I said, it's at the level of init(), send(), checkavailable() and receive().  I currently support ESP32, Pico PIO, SAMC and SAME, as well as the external MCP2515.

The Adafruit CircuitPython canio API is documented here: https://docs.circuitpython.org/en/latest/shared-bindings/canio/index.html, although I don't particularly like it ;)

I'm happy to work with whatever API the project comes up with. For an external SPI controller, it would probably be a a pure-python implementation, rather than baked into the interpreter, but I'm happy to be challenged on that.

obdevel, 2023-09-01:
> I'd recommend to never use the MCP2515. It actively randomly corrupts CAN packets. If you are building any application in the world using that part will lead to intermittent failures happening. It [...]

That is not my experience over the past 6 or 7 years, although many of the existing driver implementations are fairly poor. You need to make sure you service interrupts in a timely fashion as it only has two hardware reception buffers.

The MCP2515 does have one big advantage when considering the hobbyist and maker community ... it's the only external CAN transceiver chip available in a through-hole package. All others are small surface-mount packages which are beyond the capability of most people.

kwagyeman, 2023-09-01:
Could be that. I always grabbed a driver from online.

obdevel, 2023-09-04:
I forgot to mention before that there is a RP2040 PIO implementation of a CAN controller, written in C (https://github.com/KevinOConnor/can2040). I created a thin Arduino wrapper library last year and it works very well. There are only very minor deviations from the CAN spec. Our application project has felt comfortable enough to standardise on this.

projectgus, 2023-09-05:
Thanks everyone for the input here, it's really helpful to see the scope of what people are thinking about Re: Micropython and CAN and to learn from collective experience. It will be a little while before there's much more movement from me on this, but I'm very keen!

> I'm happy to work with whatever API the project comes up with. For an external SPI controller, it would probably be a a pure-python implementation, rather than baked into the interpreter, but I'm happy to be challenged on that.

Yes, I agree this is a good goal - to make it possible to implement a driver using an external controller whose API matches the `machine.CAN` one (i.e. can instantiate the driver object and then pass it in place of a `machine.CAN` instance, subject to whatever restrictions the controller has).

It would be great if these are eventually contributed as drivers in micropython-lib (so people can `mip install` them), written in pure Python. This all seems to me like it should be viable.

kentindell, 2023-09-17:
I already implemented a CAN API for MicroPython and would be happy to see it in the upstream. It's fairly comprehensive, as you can see from the documentation for it here:

https://github.com/kentindell/canhack/raw/master/CANPico/docs/CANPico%20MicroPython%20SDK%20reference%20manual.pdf

Cheatsheet for the API here: https://canislabs.com/downloads/can%20mp%20sdk%20cheatsheet.pdf

The code uses a C API for the drivers, and these have their own repository and they build from a submodule in my MicroPython fork.

The C API is structured to allow different hardware to be supported. I have old code lying around for the STM32 bxCAN that drives it properly (i.e. a priority software queue for 32 frames where the drivers keep the highest priority frames in the three transmit buffers).

It would be possible to fit some of the old CAN hardware like the MCP2515 to this model.

The C API drivers for the MCP251xFD are here:

https://github.com/kentindell/canis-can-sdk

The drivers handle the latest silicon errata where a race condition in the silicon can corrupt data.

I will accept pull requests into this repo for other controller and board drivers if anyone wants to add more: my intention is that this C API can become useful in its own right. The main binding of the drivers is to the RP2040 with the CANPico pinouts, but other hardware can be bound. See more on the CANPico here: https://canislabs.com/canpico/

Whatever you decide to do you ABSOLUTELY MUST NOT introduce priority inversion into the drivers. I have a blog post on that issue here: https://kentindell.github.io/2020/06/29/can-priority-inversion/

kwagyeman, 2023-09-17:
Yes, priority inversion is terrible. Ran into that on our trucks at Embark. Don't use a TX fifo.

In the NXP driver I wrote... we have 32+ message buffers for TX data and you don't have the same pressure on the transmit buffers as with the STM32. So, each message as have it's own buffer and then when you send you just re-use the same buffer to keep order.

kentindell, 2023-09-17:
> you don't have the same pressure on the transmit buffers as with the STM32

The bxCAN is designed according to same requirements as the older Motorola msCAN, which was designed according to the Volvo Car Corporation's requirements, which were set according to my requirements (very long story). The three transmit buffers arbitrate internally based on CAN ID, and an "buffer empty" interrupt handler is used to fill a free buffer from a priority ordered queue in software in the host memory. If a new frame with higher priority (lower ID) than anything in the 3 transmit buffers is generated then the lowest priority of the three is aborted (the bxCAN has a nice register field to say which one). The empty interrupt then will fill the free buffer with the highest priority frame. If the empty interrupt came from a "transmitted" (rather than "aborted") then the original frame is removed from the host side buffer.

The reason there are three slots and not two is that it avoids a race condition when the abort happens at the same time as another frame was transmitted, which would then leave no frame to enter into arbitration after the three CAN IFS bits (for complex reasons to do with latency timing analysis it's important to always enter the highest priority ready frame into arbitration).

For old controllers, like the one in the ESP32 (based on an old Philips CAN controller) or the MCP2515 (with a useless 2-bit priority field) then the race can't be avoided, so a single buffer should be used, with an abort when it needs to be replaced with a higher priority frame. It definitely should not be driven like a UART FIFO.

On the receive side, it's common to drive CAN with a receive interrupt (there's nearly always at least a 2-entry FIFO. There's reasonable time to process an incoming frame provided interrupts are not locked out for a long time elsewhere in the firmware. It's very important not to drop frames: the CAN protocol provides atomic broadcast in hardware thanks  its error handling, and dropping frames breaks that atomicity. For a CAN bus at 1Mbit/sec the time between two frames can be as short as 47 microseconds. That's quite tight, so with my bxCAN drivers on the PyBoard I made the interrupt the highest priority.

Another common receive mechanism for CAN is to have a pre-allocated buffer in hardware according to ID (not for the bxCAN, but for multi buffer devices like the MCP251xFD) and the hardware places an incoming frame into the matching buffer, which can then be polled at a slow rate (since the next frame with that ID will typically be many milliseconds before arriving). My CAN API doesn't support this model, mostly to support a wide range of controllers via the same mechanism (receive interrupt). If there is enough interest, this alternative no-interrupt buffer scheme could be added to the API.

projectgus, 2023-09-20:
Hi @kentindell, thanks for joining the discussion - it's much appreciated! I was planning to ask you to review the new API anyway once it was posted, but you're here already. :laughing: 

Regarding the C API you maintain, do you have any thoughts on the viability of writing controller drivers for external controllers in Python rather than C? The main advantage of Python drivers is that they're straightforward to install via `mip`, rather than depending on the build configuration of the MicroPython firmware.

My gut feeling is that it will be OK, any time critical processing can still be done in a hard interrupt handler and I/O (CAN and SPI) should become bottlenecks well before Python becomes one. (That said, it is still fiddly to write a performant driver in pure Python without too many memory allocations.)

kentindell, 2023-09-21:
I hadn't considered at all writing drivers in Python, but now you mention it I've been given it some thought.

The drivers have some tight timing constraints: the minimum time for a CAN frame to go on the bus is 47 bit times (a remote frame or 0 byte frame). So in the worst-case, the drivers must pull the frame out of the receive buffer (or fill the transmit buffer) before the bus has sent a 47 bit frame - or 47 microseconds at 1Mbit/sec. Some fixed delay for starting an ISR can be tolerated due to the receive FIFO being at least two deep (and on the MCP25xFD it can be programmed to be deeper), but the per-frame overhead (SPI + CPU) has to meet this deadline. As you said, the SPI delays are common to all languages.

So, can Python comfortably hit this 47 microsecond deadline? What happens if garbage collection is taking place when the interrupt is raised? Is garbage collection interruptible by Python ISRs? I guess yes, since Python ISRs are already possible.

If Python can start fast enough, and the dynamic memory allocation can be resolved (pre-allocated structures?) then in theory I could see it working. I guess for some devices you'd still need C for performance (like the bxCAN on the PyBoard) but maybe there can be a way to have either C or Python drivers, against a consistent Python API.

projectgus, 2023-09-26:
> If Python can start fast enough, and the dynamic memory allocation can be resolved (pre-allocated structures?) then in theory I could see it working. I guess for some devices you'd still need C for [...]

This assessment sounds right to me, thanks for laying out the worst-case deadline requirement.

On most ports it's possible to write "hard" interrupt handlers in Python ([some docs here](https://docs.micropython.org/en/latest/reference/isr_rules.html)), which must not allocate but otherwise should run with similar latency to a C ISR.

For built-in CAN peripherals inside a particular micro, I think it's reasonable and likely these will be implemented 100% or nearly 100% in C. The Python driver use case is more for the external controllers (MCP251x, etc), which could be interfaced to any port.

kwagyeman, 2023-10-09:
Hi can we get some progress on this? We need to merge in support for this module soon...

projectgus, 2023-10-09:
Hey @kwagyeman,

I understand this is a hassle for the OpenMV product release and that it's be very desirable for you folks to have full upstream support including a CAN driver.

For now, the situation in the first post of the issue remains true - this is blocked behind the USB device driver work. It will be a while until there's visible progress, and I suspect even once a draft API is available then will remain a draft for a while longer as trade-offs are being talked through, details checked, early versions of actual drivers are implemented and tested, etc. I also don't want the process to take too long, but getting a good long-term result will require some of that back and forth. I don't foresee any way around that, unfortunately.

In terms of your product schedule, is there a way to work around missing upstream CAN support (and/or defined `machine.CAN` API) in the interim? Does maintaining a fork (for now) or an out of tree driver work as a stop-gap?

kwagyeman, 2023-10-10:
We can, I finished the driver already, it's just that it will be the wrong long-term API. Only a few people will probably need it so we can just give me what I have right now... and then switch later.

projectgus, 2023-10-11:
Sounds good. Hopefully switching will be an option before too long.

kentindell, 2023-10-11:
It's really important that upstream support is done properly, and avoids the common CAN bear traps (eg priority inversion). I'd prefer to wait for that so that at the end I don't need to maintain a fork and can put future efforts into enhancing the upstream.

kwagyeman, 2023-10-28:
Any updates on this?

winnie-xzh, 2023-11-24:
Looking forward to it very much.

projectgus, 2023-12-07:
Thanks everyone for your patience. An initial draft machine.CAN API is up, please take a look and leave any comments.

kwagyeman, 2023-12-07:
Where is the link to the draft?

Tomblarom, 2024-02-02:
As a typical "python user" with basic hw-knownledge, that has the interest to use CAN on micropython, I was captivated by reading where this implementation started, https://github.com/micropython/micropython/issues/5087 and how it formed over the past years, only to be finally merged soon! I really take my hat off to your work and look forward to being a user of your fruits soon. I hope my words motivate you a little!

I'm planning to build a tiny dual-CAN-device using off-the-shelf parts and programming in micropython. :)

AndrewD, 2024-09-12:
Is there a plan to restart the effort on this? It would be a great feature for a project I have planned with an imxrt.

sveinungkb, 2025-03-05:
I was excited seeing @Tomblarom's comment from February, but seems like it was wrong February. What is remaining to get this merged? In particular for esp32.

IhorNehrutsa, 2025-03-05:
@sveinungkb 
I corrected the basic parameters **port/esp32: Add CAN(TWAI) driver** #12331 according to this 'machine.CAN' API.

jantman, 2025-05-25:
I'm also very interested. Is there anything I can do to help this along?