micropython/micropython #12349: v1.20.0 (2023-04-26) crashes ESP32 on LILYGO TTGO LORA32 as soon as Bluetooth is used

antirez, 2023-08-31:
This bug was here for 1 year, however I thought that it could be solved before the official release, but now I see that maybe this was never reported befofre.

So, if you use LILYGO devices with 1.20.0 and use BLE code like that:

https://github.com/antirez/freakwan/blob/main/bt.py

The ESP32 device will immediately crash (restart or alike). This bug did never happen with the old stable version, 1.19.1.
This was tested in multiple devices. Basically everybody trying [FreakWAN](https://github.com/antirez/freakwan) with 1.20.0 has this issue, and we have to tell people to install 1.19.1 to work with it.

The issue is always reproducible. However if you want a self contained Python file creating the issue without the rest of FreakWAN, just ask me. I believe this bug is critical because it does not allow to use the latest MicroPython build with very popular ESP32 devices.

In want to use this last line to say thank you for the wonderful project that MicroPython is. Thanks.

dpgeorge, 2023-08-31:
Thanks for the report.

I do not have a LILYGO LoRa board to test this on, but it sounds like it's not related to LoRa (only BLE) so the bug should be reproducible on any ESP32 board.  Is that true?

Testing v1.20.0 (2023-04-26) on a ESP32 chip (TinyPico board) I can do a BLE GAP scan without any issue.

Note that the most recent ESP32 builds (nightly builds found at https://micropython.org/download/ESP32_GENERIC/) are built with IDF 5.0.2, and maybe these work now on the LILYGO LoRa board.

Can I ask you:
- What exactly is the crash, the dump/trace back log?
- Are you able to reproduce the issue on a non LoRa board?  Eg a standard Espressif WROVER/WROOM board.
- Can you provide code that shows the issue using only BLE?  The bt.py file linked above does not actually run anything or start BLE, it just sets up some classes (or maybe that's already crashing??)
- Can you please try the latest nightly builds and see if they still have the same crash.

> In want to use this last line to say thank you for the wonderful project that MicroPython is. Thanks.

You are welcome!  Thanks for the kind feedback.

jimmo, 2023-08-31:
One major change for BLE on ESP32 between v1.19 and v1.20 was the change to use synchronous events. 5dbb822ca4a809ac5cb4513afb0411b4eb8dc3cf

I notice that freakwan primarily uses asyncio (yay!), but your BLE uart code is not async. asyncio is not at all thread safe, and BLE callbacks happen in scheduler context and so are not safe to interact with asyncio code at all. You basically have to use `asyncio.ThreadSafeFlag` to make this safe.

That said, I don't know for sure whether this could explain the crash you're seeing though, but all bets are off w.r.t. asyncio and threads. The easiest way to solve this is with `aioble` -- https://github.com/micropython/micropython-lib/tree/master/micropython/bluetooth/aioble. I haven't written an aioble UART example yet although it should be fairly straightforward to adapt the existing temp sensor example.

As Damien said, if you can provide a minimal standalone repro that would be very helpful to confirm this.

jimmo, 2023-08-31:
Made a correction to the above, just to clarify what I meant by interacting with asyncio code and show an example.

Btw, random aside, just while looking for that example of create_task I noticed in `networking.py` this comment: `as uasyncio does not support awaiting sockets`. You should be able to use a StreamReader wrapping your socket and `await sr.read()`.

antirez, 2023-10-19:
Sorry for being silent about this issue. I jumped here just to say that indeed ESP32 nightly build work in the LILYGO LORA32. However the ssd1306 module is missing in this default build, so it is not possible to use it for FreakWAN unfortunately.

robert-hh, 2023-10-19:
You can always install it manually by copying the file ssd1306.py to the board's file system.

antirez, 2023-10-19:
Thanks @robert-hh, I thought it was a hard-coded C library, compiled with MicroPython itself.

robert-hh, 2023-10-19:
No. It is Python code, embedded into the firmware as so called `frozen bytecode`. The source is at https://github.com/micropython/micropython-lib/tree/e025c843b60e93689f0f991d753010bb5bd6a722/micropython/drivers/display/ssd1306

antirez, 2023-10-19:
That's cool. I'll try. LILYGO devices being able to use the standard ESP32 build is a great news. The only problem is that my project, FreakWAN, is at the limit of the memory usage, so frozen bytecode is a more effective way compared to embedding the display code. But my guess is that's little code, or I can find ways to implement a smaller driver.

boardmain, 2023-12-30:
same problem with latest version

antirez, 2024-01-31:
I too see the same issues, even without using any asynchronous code. Even just that code I'm putting here in the message after some time will brick the device serial and I need to power off/on the device to talk again with it. Latest MicroPython code, ESP32-S3. To reproduce just use mpremote run file.py, connect with a Serial UART client (one easy to compile and run can be found inside FreakWAN), write some test, then cltr+C from the REPL, run again a couple of time until the device freezes.

Note that this example is not able to crash the device immediately but it gets bricked after running it a few times, but in the actual code I have it crashes immediately, but still not able to distill it alone.

```python
import bluetooth
import struct
from micropython import const

_ADV_TYPE_FLAGS = const(0x01)
_ADV_TYPE_NAME = const(0x09)
_ADV_TYPE_UUID16_COMPLETE = const(0x3)
_ADV_TYPE_UUID32_COMPLETE = const(0x5)
_ADV_TYPE_UUID128_COMPLETE = const(0x7)
_ADV_TYPE_UUID16_MORE = const(0x2)
_ADV_TYPE_UUID32_MORE = const(0x4)
_ADV_TYPE_UUID128_MORE = const(0x6)
_ADV_TYPE_APPEARANCE = const(0x19)
_IRQ_CENTRAL_CONNECT = const(1)
_IRQ_CENTRAL_DISCONNECT = const(2)
_IRQ_GATTS_WRITE = const(3)

_FLAG_WRITE = const(0x0008)
_FLAG_NOTIFY = const(0x0010)

_UART_UUID = bluetooth.UUID("6E400001-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_TX = (
    bluetooth.UUID("6E400003-B5A3-F393-E0A9-E50E24DCCA9E"),
    _FLAG_NOTIFY,
)
_UART_RX = (
    bluetooth.UUID("6E400002-B5A3-F393-E0A9-E50E24DCCA9E"),
    _FLAG_WRITE,
)
_UART_SERVICE = (
    _UART_UUID,
    (_UART_TX, _UART_RX),
)

def pack_adv_data(adv_type, value):
    return struct.pack("BB", len(value) + 1, adv_type) + value

def pack_adv_service(uuid):
    b = bytes(uuid)
    if len(b) == 2:
        return pack_adv_data(_ADV_TYPE_UUID16_COMPLETE, b)
    elif len(b) == 4:
        return pack_adv_data(_ADV_TYPE_UUID32_COMPLETE, b)
    elif len(b) == 16:
        return pack_adv_data(_ADV_TYPE_UUID128_COMPLETE, b)

# Generate a payload to advertise as primary payload
def gen_advertising_payload():
    payload = bytearray()
    payload += pack_adv_data(_ADV_TYPE_FLAGS, struct.pack("B", 0x06))
    payload += pack_adv_service(_UART_UUID)
    return payload

class BLEUART:
    def __init__(self, ble, name="mpy-uart", rxbuf=100):
        self._ble = ble
        self._ble.active(True)
        self._ble.irq(self.irq_handler)
        ((self._tx_handle, self._rx_handle),) = self._ble.gatts_register_services((_UART_SERVICE,))
        self._connections = set()
        self._rx_buffer = bytearray()
        self._handler = None
        self._payload = gen_advertising_payload()
        self._advertise()

    def set_callback(self, handler):
        self._handler = handler

    def irq_handler(self, event, data):
        # Track connections so we can send notifications.
        print("IRQ",data)

        if event == _IRQ_CENTRAL_CONNECT:
            conn_handle, _, _ = data
            self._connections.add(conn_handle)
            print("Connected", conn_handle)
        elif event == _IRQ_CENTRAL_DISCONNECT:
            conn_handle, _, _ = data
            if conn_handle in self._connections:
                self._connections.remove(conn_handle)
            # Start advertising again to allow a new connection.
            self._advertise()
            print("Disconnected", conn_handle)
        elif event == _IRQ_GATTS_WRITE:
            conn_handle, value_handle = data
            if conn_handle in self._connections and value_handle == self._rx_handle:
                self._rx_buffer += self._ble.gatts_read(self._rx_handle)
                if self._handler:
                    self._handler()

    def _advertise(self, interval_us=200000):
        self._ble.gap_advertise(interval_us, adv_data=self._payload)

    def read(self, sz=None):
        if not sz:
            sz = len(self._rx_buffer)
        result = self._rx_buffer[0:sz]
        self._rx_buffer = self._rx_buffer[sz:]
        return result

if  __name__ == "__main__":
    import time

    def receive_callback():
        data = uart.read().decode()
        print("Data from BLE",data)
        # uart.print(data)

    ble = bluetooth.BLE()
    uart = BLEUART(ble)
    uart.set_callback(receive_callback)
    while True:
        print("Runnig")
        time.sleep(1)

```

antirez, 2024-02-01:
I got a guru meditation stack trace:

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4009d601  PS      : 0x00060533  A0      : 0x8009d1b3  A1      : 0x3ffce6b0  
A2      : 0x00000000  A3      : 0x0000000a  A4      : 0x00000800  A5      : 0x00000c00  
A6      : 0x00000000  A7      : 0x00060523  A8      : 0x3fff27fc  A9      : 0x3ffe4364  
A10     : 0x00001c20  A11     : 0x00000028  A12     : 0x3ffe4364  A13     : 0x3ffe4388  
A14     : 0x3ffe43b8  A15     : 0x0000000b  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000000c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4009d5fe:0x3ffce6b0 0x4009d1b0:0x3ffce6d0 0x4008227c:0x3ffce6f0 0x40082517:0x3ffce710 0x40086984:0x3ffce730 0x40125f7a:0x3ffce750 0x40124cd1:0x3ffce770 0x40124db8:0x3ffce790 0x400f3414:0x3ffce7e0 0x400e9fa1:0x3ffce800 0x400df2db:0x3ffce820 0x400e67f1:0x3ffce850 0x400e68b5:0x3ffce870 0x40085635:0x3ffce890 0x400df26c:0x3ffce930 0x400e67f1:0x3ffce9a0 0x400e68b5:0x3ffce9c0 0x400e4c70:0x3ffce9e0 0x400e4aad:0x3ffcea40 0x400e67f1:0x3ffcea60 0x40085585:0x3ffcea80 0x400df26c:0x3ffceb20 0x400e67f1:0x3ffceb80 0x400e6806:0x3ffceba0 0x400f470e:0x3ffcebc0 0x400f48a8:0x3ffcec50 0x400d8092:0x3ffceca0

ELF file SHA256: 9b0c9ceeb137b944

Rebooting...
```

Carglglz, 2024-02-01:
@antirez this may be of help: https://github.com/tve/esp32-backtrace

I've just decoded the backtrace 
```
BT-0: 0x4009d5fe is in tlsf_malloc (/home/micropython/esp-idf-v5.0/components/heap/tlsf/tlsf.c:567).
BT-1: 0x4009d1b0 is in multi_heap_malloc_impl (/home/micropython/esp-idf-v5.0/components/heap/multi_heap.c:207).
BT-2: 0x4008227c is in heap_caps_malloc_base (/home/micropython/esp-idf-v5.0/components/heap/heap_caps.c:75).
BT-3: 0x40082517 is in heap_caps_calloc (/home/micropython/esp-idf-v5.0/components/heap/heap_caps.c:479).
BT-4: 0x40086984 is in bt_osi_mem_calloc (/home/micropython/esp-idf-v5.0/components/bt/porting/mem/bt_osi_mem.c:27).
BT-5: 0x40125f7a is in esp_nimble_hci_init (/home/micropython/esp-idf-v5.0/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:422).
BT-6: 0x40124cd1 is in esp_nimble_init (/home/micropython/esp-idf-v5.0/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:95).
BT-7: 0x40124db8 is in nimble_port_init (/home/micropython/esp-idf-v5.0/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:166).
BT-8: 0x400f3414 is in mp_bluetooth_init (/home/micropython/micropython-autobuild/extmod/nimble/modbluetooth_nimble.c:623).
BT-9: 0x400e9fa1 is in bluetooth_ble_active (/home/micropython/micropython-autobuild/extmod/modbluetooth.c:295).
BT-10: 0x400df2db is in fun_builtin_var_call (/home/micropython/micropython-autobuild/py/objfun.c:119).
BT-11: 0x400e67f1 is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:713).
BT-12: 0x400e68b5 is in mp_call_method_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:729).
BT-13: 0x40085635 is in mp_execute_bytecode (/home/micropython/micropython-autobuild/py/vm.c:1042).
BT-14: 0x400df26c is in fun_bc_call (/home/micropython/micropython-autobuild/py/objfun.c:273).
BT-15: 0x400e67f1 is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:713).
BT-16: 0x400e68b5 is in mp_call_method_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:729).
BT-17: 0x400e4c70 is in mp_obj_instance_make_new (/home/micropython/micropython-autobuild/py/objtype.c:349).
BT-18: 0x400e4aad is in type_call (/home/micropython/micropython-autobuild/py/objtype.c:999).
BT-19: 0x400e67f1 is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:713).
BT-20: 0x40085585 is in mp_execute_bytecode (/home/micropython/micropython-autobuild/py/vm.c:957).
BT-21: 0x400df26c is in fun_bc_call (/home/micropython/micropython-autobuild/py/objfun.c:273).
BT-22: 0x400e67f1 is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:713).
BT-23: 0x400e6806 is in mp_call_function_0 (/home/micropython/micropython-autobuild/py/runtime.c:687).
BT-24: 0x400f470e is in parse_compile_execute (/home/micropython/micropython-autobuild/shared/runtime/pyexec.c:123).
BT-25: 0x400f48a8 is in pyexec_raw_repl (/home/micropython/micropython-autobuild/shared/runtime/pyexec.c:280).
BT-26: 0x400d8092 is in mp_task (/home/micropython/micropython-autobuild/ports/esp32/main.c:154).

```

I've no idea what could be the cause but it seems to point `/home/micropython/esp-idf-v5.0/components/heap/tlsf/tlsf.c:567`

```C
static inline __attribute__((always_inline)) block_header_t* block_locate_free(control_t* control, size_t size)
{
	int fl = 0, sl = 0;
	block_header_t* block = 0;

	if (size)
	{
		mapping_search(control, size, &fl, &sl);
		
		/*
		** mapping_search can futz with the size, so for excessively large sizes it can sometimes wind up 
		** with indices that are off the end of the block array.
		** So, we protect against that here, since this is the only callsite of mapping_search.
		** Note that we don't need to check sl, since it comes from a modulo operation that guarantees it's always in range.
		*/
		if (fl < control->fl_index_count)
		{
			block = search_suitable_block(control, &fl, &sl);
		}
	}

	if (block)
	{
		tlsf_assert(block_size(block) >= size);
567             remove_free_block(control, block, fl, sl);
	}

	return block;
}
```

antirez, 2024-02-01:
@Carglglz Thanks for decoding the trace. The call stack makes sense... it crashes right away at BLE initialization.

Carglglz, 2024-02-01:
@antirez have you test this using e.g. `picocom` to enter the REPL and do 
```
>>> import bt
```
?
Maybe RAW REPL mode (the one `mpremote` uses) has anything to do? 🤔

antirez, 2024-02-01:
@Carglglz Thanks, definitely not related to mpremote. I was able to replicate issues with three different devices in different setups: just main.py, or in the context of FreakWAN, and and in the repl. Always same result.

antirez, 2024-02-01:
If @dpgeorge wants a LILYGO device like the one we use I can send one him ASAP btw.

dpgeorge, 2024-02-02:
> If @dpgeorge wants a LILYGO device to reproduce, like the ones we use here, I can send one him ASAP btw.

Yes, that would be useful, then I can try to reproduce this issue.

Or, is it now possible to reproduce it with another (standard) esp32 board?

antirez, 2024-02-02:
> > If @dpgeorge wants a LILYGO device to reproduce, like the ones we use here, I can send one him ASAP btw.
> 
> Yes, that would be useful, then I can try to reproduce this issue.
> 
> Or, is it now possible to reproduce it with another (standard) esp32 board?

Thank you, please could you give me your address sending an email to antirez@gmail.com?
I have only another ESP32 board and I can reproduce an issue but it's not the same: the code I provided in this issue kind works but there is a very long delay before the device starts sending the advertising beacons, yet it does not seem to crash like the LILYGO ESP32 and ESP32-S3 devices I worked with. Thanks.

Anyway given that these LILYGO boards are so widespread it would be cool if you had a few.

Long shot: I'll try to see if @lewisxhe is able to help with that. Otherwise I'll just send a board from Italy as I have a few. But if there was some official willingness from LILYGO to get involved, they should perhaps send you something more than a T3 with LoRa, but other devices of the line as well, like the T-DISPLAY, or some T-BEAM and so forth.

mattytrentini, 2024-02-03:
I have _many_ LilyGo boards (and live close to Damien) but none supporting LoRa...  Mine include: T4 S3, T-Track, T-Touch Bar, T-HMI, T7-S3, T-Display S3 AMOLED, T-Display, T-RGB, T-Dongle S3, T-QT Pro, T-Watch (ESP32). If any of those can be used I'm happy to loan them out.

I'll also be able to try and reproduce the issue, but not before Monday.

lewisxhe, 2024-02-03:
@antirez Hi, what do you need me to do? I'm happy to cooperate

antirez, 2024-02-03:
@lewisxhe Thank you, it would be totally awesome if Lilygo could send a few test devices to @dpgeorge in order to better track this issues. For instance a T3 v2 1.6 LoRa, which clearly shows this problem. Also I wanted to ask if you are aware of any BLE peculiarities in these devices that may cause issues compared to standard ESP32 devices.

@mattytrentini Appreciated. It could be very useful, even to know if this is somewhat related to all the Lilygo devices.

I'm currently trying to finish an ST77xx driver for MicroPython I would like to release ASAP and then I'll venture to build MicroPython myself from sources also because this would allow me to start contributing more actively, and maybe to track and fix this issue. I'm more of a C programmer than a Python one, but the toolchain to cross-compile stuff for the ESP32 discourages me a bit...

lewisxhe, 2024-02-03:
1. Of course, LilyGo is willing to provide hardware support to those who contribute to the community. However, it is the Chinese New Year now. If you need new equipment, you may need to wait until after the holiday. If possible, wait until the Spring Festival is over before contacting us.
2. I have not used BLE in depth and I cannot explain it.
3. Thank you for your contribution

andrewleech, 2024-02-03:
> the toolchain to cross-compile stuff for the ESP32 discourages me a bit...

If you'd prefer not installing the esp toolchain, docker can be used to quickly use the official toolchain and test different versions etc.

Once you've got docker installed, the following command will compile and flash your target esp board.
    
```
cd micropython
docker run -it --rm -e BOARD=ESP32_GENERIC -e UID=$(id -u) -v $(pwd):/micropython -v /sys/bus:/sys/bus -v /dev:/dev --net=host -w /micropython espressif/idf:v5.1.2 bash -c '\
useradd -ms /bin/bash -g root -G sudo -u $UID esp;\
git config --global --add safe.directory /opt/esp/idf;\
su esp -m -c ". /opt/esp/entrypoint.sh; export HOME=/tmp; \
make -C mpy-cross && \
make -C ports/esp32 BOARD=$BOARD submodules && \
make -C ports/esp32 BOARD=$BOARD deploy PORT=/dev/ttyACM0; \
"'
```

antirez, 2024-02-03:
@andrewleech this is *very* helpful. Thanks.

antirez, 2024-02-11:
Hi all, as agreed with @dpgeorge via email I tried to bisect the bug and found that indeed the issue is due to 5dbb822ca. e05d0a633 Runs ok, so as guessed by @jimmo the synchronous events is what started it. This is what I get (and I can easily reproduce):

```
***ERROR*** A stack overflow in task ble has been detected.

Backtrace:0x40093af2:0x3ffd0c90 0x40094269:0x3ffd0cb0 0x4009443a:0x3ffd0cd0 0x4009528c:0x3ffd0d40 0x40094530:0x3ffd0d70 0x400944e6:0x00000000 |<-CORRUPTED
```

I can provide further information if requested. Thank you.

dpgeorge, 2024-02-12:
> I tried to bisect the bug and found that indeed the issue is due to [5dbb822](https://github.com/micropython/micropython/commit/5dbb822ca4a809ac5cb4513afb0411b4eb8dc3cf). [e05d0a6](https://github.com/micropython/micropython/commit/e05d0a6335be6a45f29f5d987091b858ad378cfa) Runs ok, so as guessed by @jimmo the synchronous events is what started it.

OK, that's great that you could narrow it down.

> `***ERROR*** A stack overflow in task ble has been detected.`

Did you see this error before in your testing, ie previous to doing the git-bisect?  I don't see a mention of an explicit stack overflow message in the above conversion anywhere.

> Please note that the hardware where this happens is a ESP32-PICO-D4

I have some of these chips (on a TinyPICO board) and so can test with this SoC hardware.  What exactly are your build steps?  In particular what board do you board for and what ESP-IDF version did you use when you hit this stack overflow message?

(I did try a few different build configs on the PICO-D4 but couldn't yet get the stack oveflow.)

Can you also please try building with the following line added to the `ports/esp32/boards/sdkconfig.ble` file:
```
CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE=8192
```
That should double the BLE task stack size.

antirez, 2024-02-12:
@dpgeorge thanks for following up. Here are some more details:

1. The stack overflow is new using IDF4.4.2, with other versions and pre-compiled MicroPython .bin files the bug happens in identical conditions but sometimes it's Guru Meditation, sometimes is hard reset, or other times it bricks the device. Note that I can get the stack corruption only if I load the whole FreakWAN application and use bluetooth. If I run bt.py in isolation I have to reproduce in a different way: 1) I mpremote run bt.py and connect via BT, 2) I type a few commands and get the echo back. 3) With the repl I stop the program with CTRL+C, 4) I re-run the program and get OS error 16. 5) I hit CTRL+D for a software reset and instead the device bricks. I have to remove power to awake it again. Basically if the context is the one of a bigger loaded application it's like if there is more pre-existing stack or just the memory layout is different, so effects are different.

2. The bug happens with every precompiled image in the website or with any image I compile with different versions of IDF, as long as it is >= the offending commit. But the exact toolchain I'm using is: Docker image IDK4.4.2 (otherwise I can't compile MicroPython ~1.19/20 because of an error), then:

```
BOARD="GENERIC"
make -C mpy-cross || exit
make -C ports/esp32 BOARD=$BOARD submodules || exit
make -C ports/esp32 BOARD=$BOARD || exit

cd ports/esp32/build-GENERIC || exit
python ../makeimg.py sdkconfig bootloader/bootloader.bin partition_table/partition-table.bin micropython.bin micropython-full-image.bin micropython.uf2
```

3. About your reproduction attempt, beware that this is definitely an heisebug that depends a lot on the memory layout, I suggest doing this trick of re-running bt.py from FreakWAN, get the OS 16 error (EBUSY maybe? Not sure), hitting CTRL+D.

4. Doubling the stack was the first thing I tried:

```
        mp_stack_set_limit(4096*2);
```

No effects, I also tried to use 1024 bytes, and it bricks but before bricking it detects there is no space for a new stack frame and tells me (when firing the IRQ) max recursion depth reached. It's interesting that the state gets corrupted regardless of the fact it is detected that the IRQ handler can't be called, or maybe I got this error *after* the IRQ handler fired and tried to call another MP function. I'm not sure to be honest.

I can provide any other information, can run modified versions, can add some trap to see when this exact portion of the stack gets replaced with zeroes or anything else.

dpgeorge, 2024-02-12:
> 6\. Doubling the stack was the first thing I tried:
> 
> 
> ```
>         mp_stack_set_limit(4096*2);
> ```

This bit is not really doubling the stack, it's just setting a bigger limit before the MicroPython runtime detects a C stack overflow.  You probably want to **reduce** that value instead, so that MicroPython stops earlier with a `RuntimeError` before true corruption occurs.

> 1. Note that I can get the stack corruption only if I load the whole FreakWAN application and use bluetooth.

OK, then it could be that this is "simply" an out-of-memory problem, with the IDF running out of memory for low-level activities like BLE allocations.

To get a handle on that situation, can you please trigger the bug, and log the output of the following functions leading up to the bug being triggered:
```py
import micropython, esp32
print(esp32.idf_heap_info(esp32.HEAP_EXEC))
print(esp32.idf_heap_info(esp32.HEAP_DATA))
micropython.mem_info()
```
The output of that should help to give a picture of memory usage (IDF memory and MicroPython heap) as the system progresses.

antirez, 2024-02-12:
@dpgeorge thanks, I'll do, but I want to make sure you didn't miss that I said that only the stack corruption log is triggered in low memory condition. I can brick the device with just a couple of lines of code: simply mpremote run bt.py, then CTRL+D to reset the device. I'll try what you said and report back. Thank you.

antirez, 2024-02-12:
@dpgeorge two updates:

1. By properly (as you specified) setting a larger BLE stack, the bug continues to manifest without any noticeable differnece.

2. As long as I try to print the information you specified, the IRQ handler crashes immediately with this error:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401e05ef  PS      : 0x00060830  A0      : 0x800fae46  A1      : 0x3ffd17f0  
A2      : 0x00010002  A3      : 0x00000001  A4      : 0x3ffe5800  A5      : 0x0000022c  
A6      : 0x00000011  A7      : 0x3f40f12c  A8      : 0xffffffff  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffd17e0  
A14     : 0x00000001  A15     : 0x3f419d04  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0001001e  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x401e05ec:0x3ffd17f0 0x400fae43:0x3ffd1810 0x400dbe11:0x3ffd1860 0x400e2cee:0x3ffd1880 0x400e2e1d:0x3ffd18a0 0x40084779:0x3ffd18c0 0x400dbf1e:0x3ffd1960 0x400e2cee:0x3ffd1990 0x4010bf95:0x3ffd19b0 0x4010bfe5:0x3ffd19f0 0x400e2cee:0x3ffd1a10 0x400e2df2:0x3ffd1a30 0x400e6ce7:0x3ffd1a60 0x400e6d50:0x3ffd1ae0 0x400e6deb:0x3ffd1b80 0x400e6fcd:0x3ffd1be0 0x400eea53:0x3ffd1c20 0x40118b71:0x3ffd1c80 0x40118bad:0x3ffd1ca0 0x4011905d:0x3ffd1cc0 0x401154f4:0x3ffd1d30 0x40114fdd:0x3ffd1d80 0x40115551:0x3ffd1da0 0x40114b5d:0x3ffd1dc0 0x4011dc2a:0x3ffd1de0 0x400fa7eb:0x3ffd1e00

ELF file SHA256: 005815e3093c0623
```

So I tried to put the memory debugging calls in the main loop: same effect.

So I will downgrade from 1.20.0 to 1.19.1 and without changing any line of the program (debugging still in the main loop, not in the IRQ handler), I'll try again to see if I can collect this information.

dpgeorge, 2024-02-14:
> I want to make sure you didn't miss that I said that only the stack corruption log is triggered in low memory condition. I can brick the device with just a couple of lines of code: simply mpremote run bt.py, then CTRL+D to reset the device

I have tried to reproduce the issue using `bt.py` from above, and running `osx-bte-cli` on a Mac.  Using both ESP32-PICO-D4, and ESP32S3.  Unfortunately the Bluetooth is broken on the old Mac laptop that I have, so `osx-bte-cli` doesn't work.

Instead I wrote a small MicroPython BLE client that connects to the ESP32, discovers services, writes some data to the UART UUID, and reads back the echo.

I couldn't reproduce any errors.... until I tried doing a soft-reset on the ESP32 **while BLE was still connected**.  It turns out this completely locks up the ESP32.  And the reason for this is simple, it's because the GIL is held by the main thread and the BLE IRQ handler is trying to run (due to a disconnect event as part of soft reset) and trying to obtain the GIL.  So it deadlocks.

antirez, 2024-02-14:
@dpgeorge thank you @dpgeorge! I tested the fix, and one of the issues is gone, and the other persists. Details:

1. I can no longer brick the simple example bt.py. It works, and I can reset the device and re-run it without issues.
2. Now the larger example, which is exactly bt.py but called from a more complex example, with memory used, still exists for stack overflow:

```
***ERROR*** A stack overflow in task ble has been detected.

Backtrace:0x40093af2:0x3ffd0cd0 0x40094269:0x3ffd0cf0 0x4009443a:0x3ffd0d10 0x4009528c:0x3ffd0d80 0x40094530:0x3ffd0db0 0x400944e6:0x00000001 |<-CORRUPTED
```

Last word on the stack is not always 0x1, often 0x0. Once I saw it exiting for recursion depth reached instead, when calling the IRC. The puzzling thing is that even without using BLE at all, these calls generate a Guru Meditation:

```
esp32.idf_heap_info(esp32.HEAP_EXEC)
esp32.idf_heap_info(esp32.HEAP_DATA)
```

I call them from the main application look, and I get immediately:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40097c3f  PS      : 0x00060b30  A0      : 0x800fae2f  A1      : 0x3ffccd00  
A2      : 0xb5b100db  A3      : 0x3ffccd20  A4      : 0x3fffd120  A5      : 0x0000022c  
A6      : 0x00000011  A7      : 0x3f40f12c  A8      : 0x80097c3c  A9      : 0x3ffcccf0  
A10     : 0x3ffccd20  A11     : 0x00000000  A12     : 0x0000001c  A13     : 0x3ffccd3c  
A14     : 0x0000001e  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xb5b100db  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x40097c3c:0x3ffccd00 0x400fae2c:0x3ffccd20 0x400dbde1:0x3ffccd70 0x400e2cbe:0x3ffccd90 0x400e2ded:0x3ffccdb0 0x40084779:0x3ffccdd0 0x400dc2cd:0x3ffcce70 0x400dc32b:0x3ffcce90 0x400dc38e:0x3ffccec0 0x400dbdfe:0x3ffccee0 0x400e2cbe:0x3ffccf00 0x400e2ded:0x3ffccf20 0x40084779:0x3ffccf40 0x400dbeee:0x3ffccfe0 0x400e2cbe:0x3ffcd010 0x400846e6:0x3ffcd030 0x400dbeee:0x3ffcd0d0 0x400e2cbe:0x3ffcd120 0x400e2ded:0x3ffcd140 0x40084779:0x3ffcd160 0x400dbeee:0x3ffcd200 0x400e2cbe:0x3ffcd270 0x400e2ce6:0x3ffcd290 0x400f08cb:0x3ffcd2b0 0x400f0c9b:0x3ffcd340 0x400d4c5a:0x3ffcd360
```

At this point, I wonder if the two things are related...

What I'm trying to do now is to create a set of MicroPython files with my application but able to run without the LoRa chip or the display configured, so that could be loaded in your device as well, and see if I can still reproduce the issue: maybe this way you will be able to get the full issue as well.

I tried to setup a Linux box I have with the development environment needed for you to let you enter it via ssh and reproduce on a local device: I could not find any way on my (old) Linux box to let the controller power off/on the port in order to de-brick the device after the bug, but now that the race condition when switching off the BLE stack is gone, there is no longer need for it, so this now is a viable option as well, maybe the simplest: I'll open the port forwarding on my modem and send you via email how it is possible to enter the system. The missing part is the BLE trigger, but now that you have code to do it, I can just connect *two* devices on my Linux box, so that it is possible to use one as trigger.

I'll try to bring forward both the reproduction setups so that one or the other will work :) Thanks.

dpgeorge, 2024-02-15:
> I tried to setup a Linux box I have with the development environment needed for you to let you enter it via ssh and reproduce on a local device: I could not find any way on my (old) Linux box to let [...]

Thanks for doing this!  That helps a lot, I can reproduce it on your side now.

I think the problem is a C stack overflow in the FreeRTOS ble task.  If so, that should be relatively easy to reproduce without LoRa or any display connected.  Just call lots of deep Python functions from within the BLE IRQ handler.

antirez, 2024-02-15:
I hope I did the scripting right, but starting from now you can always see
my home ADSL IP address here:

http://antirez.com/ip.txt

dpgeorge, 2024-02-15:
Thanks again for the dev environment access.  I could fix the issue by increasing the stack size.  See #13680.

dpgeorge, 2024-02-15:
The BLE task stack overflow is fixed by cd66aa05cf2d11401876a342fad2425e766bd921

A test for the bug was added in 866fc3447c7a685c6c04d6a03ee94083b8d0dd72

closed 2024-02-15 by dpgeorge

lewisxhe, 2024-02-19:
Hi @antirez
We have finished our vacation, do we still need to provide the board?

antirez, 2024-02-19:
Thank you for your interest, @lewisxhe! Thanks to @dpgeorge, the issue is now fixed, so probably no longer needed. I had a BLE problem on the T-WATCH S3 as well, but I tested a couple of days ago and apparently with the nightly version of MicroPython things are working: I'll test better in the next days. Currently rewriting pure-MP drivers for almost all the components of the T-WATCH: I really hope you'll not stop to produce the T-WATCH S3! :D Cheers.

dpgeorge, 2024-02-28:
This bug is now fixed in the v1.22.2 release, and pre-built firmware is available at `https://micropython.org/download`.