micropython/micropython #12350: Sparkfun Micromod UART

SYNiMAK, 2023-09-01:
mpconfigboard.h for Sparkfun's micromod has the incorrect UART pin definitions. The current version is using A2 and A3, but on the micromod board these pins are pulled to ground. The pins should be A9 and A10.

SYNiMAK, 2023-09-01:
I've created a fork and am testing a fix

SYNiMAK, 2023-09-01:
So the issues seem to have been created by Sparkfun... there are different version of the micromod schematic. The 15FEB2021 version uses A2 and A3 as TX and RX and pulls A9 and A10 to 3V3. But the 6JAN2023 version uses A9 and A10 for TX and RX and pulls A2 and A3 to ground. ...why Sparkfun, why?

dpgeorge, 2023-09-01:
That's very annoying!

Is it possible to auto-detect the board revision and act accordingly?

SYNiMAK, 2023-09-01:
For reference...
[MicroMod_STM32_Processor_V10_20210215.pdf](https://github.com/micropython/micropython/files/12493896/MicroMod_STM32_Processor_V10_20210215.pdf)
[MicroMod_STM32_Processor_V20_20230106.pdf](https://github.com/micropython/micropython/files/12493897/MicroMod_STM32_Processor_V20_20230106.pdf)

SYNiMAK, 2023-09-01:
> That's very annoying!
> 
> Is it possible to auto-detect the board revision and act accordingly?

Possibly... you could read the pins and check to see if they are pulled low or high... and set the pins dynamically??

Maybe both UARTs need to be defined then only use the one you need?

There are no markings on the micromod boards to know which version they are.

SYNiMAK, 2023-09-01:
This branch works for V20