micropython/micropython #12385: Implement len() for dict_view

eleciawhite, 2023-09-06:
have a bit of code that works in Python 3.5 but not micropython:
```
d = {1:"hello", 2:"world"}
len(d)
len(d.items())

```
In Py3.5 and 2.7, both `len(d)` and `len(d.items())` return 2
In micropython, `len(d)` is 2 but `len(d.items())` throws:
```
len(d.items())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object of type 'dict_view' has no len()

```
Note that `len(d.items()) == len(d.keys()) == len(d.values()) == len(d)` should be true for consistency.

I don't think the version matters but I am running a variant of 1.20.0 on an STM32 L4:
```
MicroPython 724865b19-dirty on 2023-09-05; NUCLEO-L4A6ZG with STM32L4A6ZG
Type "help()" for more information.
```

jimmo, 2023-09-11:
At the very least we should document this difference -- see notes at https://github.com/micropython/micropython/pull/12386#issuecomment-1713073736

Obviously we aim to be as compatible as possible with CPython, and fixing this particular issue would be a very small code size increase, but that has to be weighed up against how useful it is.. "every byte counts" etc... Often we choose to enable these sorts of features on boards with more flash (e.g. this is the original intent of the `MICROPY_CPYTHON_COMPAT` config option).

capak07, 2024-03-18:
Hi, I would like to start working on this issue if no one's working on it.
Thanks

tadeubas, 2024-09-04:
Hey guys, any progress on this issue?

dpgeorge, 2025-11-19:
I've made a PR to implement this, see #18437.