micropython/micropython #12393: rp2: No support for named pins in Pico W build

Gadgetoid, 2023-09-08:
The named pins feature for supporting external pins requires the definition of `machine_pin_ext_init` and various other functions for configuring, setting and getting external pins.

This feature is entirely monopolized by the Pico W build, in such a way that it's impossible to add additional external pins on top of those defined for Pico W.

There should be no reason why we can't take a copy of [machine_pin_cyw43.c](https://github.com/micropython/micropython/blob/master/ports/rp2/machine_pin_cyw43.c) , and extend it as necessary.

This would require moving the hard-coded exception for Pico W in CMakeLists.txt into the board config -

https://github.com/micropython/micropython/blob/3637252b7bc3e85ea92038161e008a550991d1f4/ports/rp2/CMakeLists.txt#L300-L302

ESP32 handles additional sources with the `MICROPY_SOURCE_BOARD` variable, I think the RP2 port should mirror this behavior, and this is how `machine_pin_cyw43.c` should be included - https://github.com/micropython/micropython/blob/3637252b7bc3e85ea92038161e008a550991d1f4/ports/esp32/esp32_common.cmake#L105

robert-hh, 2023-09-08:
You wrote "named pins", but the examples show pins at external devices like the CYW43 or a NINAW10 module, which are supported by MicroPython's machine.Pin class like MICU pins. This support is highly dependent on the respective external module and linked to it's firmware.
So it's not clear to me what you want to achieve. Support for named pins in the sense that they could be addressed by their name on the board's silkscreen already exists. PR #12211 aims to harmonize that between ports.

Gadgetoid, 2023-09-08:
Corrected title to "external."

jimmo, 2023-09-11:
@Gadgetoid I think really the issue here is that the external pins feature doesn't compose nicely -- e.g. it should be possible to have external pins on the cyw43 as well as maybe an i2c port expander. `machine_pin_cyw43.c` isn't a feature of the board, it's a feature of all boards with cyw43 wifi on them (of which the Pico W is just one example).

It's not entirely clear to me though what you're after -- do you want to make more cyw43 gpio available (I thought this should just work?), or are you trying to add another source of external pins?