micropython/micropython #12406: Zero block size ?

one0410, 2023-09-11:
I used to use ESP32-S3-WROOM-1-N8R2. Everything works fine.
This time, I'm using ESP32-S3-WROOM-1-N4R2.
I can flash the latest firmware downloaded from micropython website.

```
c:\Downloads>esptool --chip esp32s3 --port COM6 --baud 115200 --after hard_reset write_flash -e --flash_size detect -z 0 ESP32_GENERIC_S3-20230906-unstable-v1.20.0-450-g3637252b7.bin
esptool.py v4.6.2
Serial port COM6
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:5e:cb:30
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.5s
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=4MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 1558640 bytes to 1036532...
Wrote 1558640 bytes (1036532 compressed) at 0x00000000 in 12.8 seconds (effective 977.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

After rebooting, I found I can't mkdir or mpremote cp to the board.
When I check statvfs, I got zero block size.

```
Connecting to COM7...

>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.20.0', version='v1.20.0-450-g3637252b7 on 2023-09-06', machine='Generic ESP32S3 module with ESP32S3')
>>> os.listdir()
[]
>>> os.mkdir('aa')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
>>> os.statvfs('/')
(0, 0, 0, 0, 0, 0, 0, 0, 0, 128)
>>>
```

Could anyone help me?
Thanks a lot.

jimmo, 2023-09-11:
We currently only support 8MiB flash for the generic S3 firmware -- _"The following files are firmware that should work on most ESP32-S3-based boards with 8MiB of flash, including WROOM and MINI modules."_

Your N4R2 module has 4MiB of flash.

We don't want to make the build server have to build every possible flash size for each board, so the plan is to add support for other flash sizes by adding a script to modify the default firmware for each flash size, and have those published alongside.

For now, you can get a working firmware image by building from source and updating `ESP32_GENERIC_S3/sdkconfig.board` to set the 4MiB options.

one0410, 2023-09-11:
OK, got it.
I'll try to build a 4MB firmware.
Thanks for your help.

closed 2023-09-11 by one0410