micropython/micropython #12410: RP2: UART parity gives wrong timing

climblinne, 2023-09-11:
OS: MicroPython v1.20.0 on 2023-04-26; Raspberry Pi Pico with RP2040

I use the Pico with a serial communication with parity settings 'parity=2' (on the PC it's 'even' parity setting).
When running it with sample code 1, there is one bit missing in the output (obviously the parity bit).

When doing the same with sample code 2, where I first init with 'parity=1' and then 'parity=2' it works correctly. I attached screenshot from my oscilloscope (the output driver put it to 24V).

Also I think the nomenclature of 'even' and 'odd' should be changed (in comparison to my PC serial connection).

Sample Code 1 (not working):

```python
from machine import Pin, UART
uart = UART(1, baudrate=115200, timeout=1000, tx=Pin(8), rx=Pin(9), parity=2, stop=1, invert=UART.INV_TX)
uart.write(b'\xC8\x37')
```
[image]

Sample Code 2 (work around, working!):

```python
from machine import Pin, UART
uart = UART(1, baudrate=115200, timeout=1000, tx=Pin(8), rx=Pin(9), parity=1, stop=1, invert=UART.INV_TX)
uart = UART(1, baudrate=115200, timeout=1000, tx=Pin(8), rx=Pin(9), parity=2, stop=1, invert=UART.INV_TX)
uart.write(b'\xC8\x37')
```
[image]

climblinne, 2023-09-11:
The fix from #10976 didn't.

robert-hh, 2023-09-11:
I cannot replicate the behavior with a RPI Pico using your code and the most recent nightly build.

MicroPython v1.20.0-450-g3637252b7 on 2023-09-11; Raspberry Pi Pico with RP2040

[image]

climblinne, 2023-09-11:
@robert-hh Thanks for the feedback. 

Yes I can confirm with the latest nightly from above it works. 
Looking again at the parity nomenclature I found that I used the wrong index.

closed 2023-09-11 by climblinne