micropython/micropython #12421: embedding example segfaults under gcc with no optimizations

Shootfast, 2023-09-11:
Hi,

I recently ran around in circles trying to find out why the embedded port was failing to run in my application, and eventually tracked it down to an issue when compiling with 0 optimizations in the library.

to reproduce:

    cd examples/embedding
    make -f micropython_embed.mk
    # Swap the optimization level to 0 in the Makefile
    sed -i s,-Og,-O0,g Makefile
    make

The embed application now segfaults on launch

    $ ./embed 
    Segmentation fault

This is from the released source tarball: micropython-1.20.0.tar.xz 
GCC 9.3.1 on Centos 7.9.2009 , under redhat's devtoolset-9

niftyhacking, 2023-09-11:
Consider trying compilers...  GCC, Clang, and LLVM
Imbedded hints that size can matter  -z and -s
You can search for the critical gcc optimization with some work.
https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html

Can you catch the segfault and map it to code?
Does  adding "-ansi -pedantic -Wall" illuminate anything? <-- first step?

projectgus, 2023-09-12:
Hi @Shootfast, 

Interesting corner case you've found! I built the embedded example on master branch with `-O0 -g` to include debug symbols, and it shows the segfault happens on return from from `nlr_push()` at embed_utils.c:47.

I'm pretty sure that dropping the optimisation level is changing the x86-64 calling convention in some way that the optimised NLR implementation (in `nlrx64.c`) no longer supports. In general, I don't believe that Micropython is tested or supported for arbitrary compiler flags and optimisation levels.

A workaround is to add `#define MICROPY_NLR_SETJMP    (1)` to `examples/embedding/mpconfigport.h`. This configures Micropython to use the un-optimised but architecture independent setjmp implementation instead, and the example seems to run fine with it.

It's possible `MICROPY_NLR_SETJMP` will also turn out to be the best solution for this (either that or always building with some particular compiler flag). However, I'm not particularly knowledgeable about x86-64 calling conventions so I couldn't say for sure - will wait for someone who is. :smile:

projectgus, 2023-09-12:
Oh, it's nothing exotic after all, - it's the old standby `fomit-frame-pointer`! If you build with `-O0 -fomit-frame-pointer` then the segfault goes away.

I don't believe there is a way to detect this at compile time (to either fail with an error or compile a modified version of the NLR implementation), so I think these might be the best workarounds available (either pass `-fomit-frame-pointer`, or set `MICROPY_NLR_SETJMP`).

Shootfast, 2023-09-12:
Thanks for looking into this and finding the answer so quickly! I wouldn't usually muck about too much with the optimizations, but because the embedded port builds to .c files, when I integrated them into my project (which uses CMake) and build with -DCMAKE_BUILD_TYPE=Debug, it was failing out of the box.
I guess that could be avoided by having the port build a .a/.so compiled with the required flags, or perhaps add a GCC specific pragma to ensure that optimization is enabled for the function?

projectgus, 2023-09-12:
> I guess that could be avoided by having the port build a .a/.so compiled with the required flags, or perhaps add a GCC specific pragma to ensure that optimization is enabled for the function?

Building the MicroPython code into a static library would work, or in CMake you could use `set_source_files_properties(<all micropython files> PROPERTIES COMPILE_FLAGS -fomit-frame-pointer)` to build only those files with `f-omit-frame-pointer`. Or you could apply it across your project with `add_compile_options(-fomit-frame-pointer)` or similar, the code should still be perfectly debuggable even without a frame pointer register.

`-fomit-frame-pointer`  has to be set on the *callers* of the nlr functions, not the nlr functions themselves, so it's unfortunately not as simple as setting a pragma in a single file.

projectgus, 2023-09-13:
> add a GCC specific pragma to ensure that optimization is enabled for the function?

Great suggestion, this appears to work! See linked PR.

commit fa685239 (2023-09-20): py/nlrx64: Mark nlr_push() as naked function when possible.

Supported from GCC 8 and up, and Compiler Explorer suggests it works as
expected with Clang since 3.6 (2014).

- Fixes situation where building embedded MicroPython with -O0 and
  MICROPY_NLR_X64 crashes at runtime (due to nlr_push pushing the
  frame pointer register EBP). Closes #12421.

- Allows removing the macOS tweak to undo pushing EBP onto the stack
  in the generated function prelude.
PR #12434 (2023-09-20): py/nlrx64: Mark nlr_push() as naked function when possible.
Alternative to #12433 that might also simplify the macOS builds, and makes nlrx64.c more similar to nlrx86.c.

I'm aware that naked functions have a history of being a nuisance in the NLR code though (#3484, etc) so maybe this is not a good option in some way.

- Fixes situation where building embedded MicroPython with -O0 and MICROPY_NLR_X64 crashes at runtime (due to nlr_push pushing the frame pointer register EBP). Closes #12421.

- Allows removing the macOS tweak to undo pushing EBP onto the stack in the generated function prelude [... cut]
closed 2023-09-20 by dpgeorge