micropython/micropython #12453: NameError: name 'errno' isn't defined

knemay, 2023-09-15:
Hey, raise OSError(errno.ETIMEDOUT) is working fine when I test it in the REPL or in an .py file, but is raising that error when ran in an .mpy file. I'm using mpy-cross 1.20.0 to compile the file. What could be happening and how can I fix it?
Thanks in advance,

Please, dont close the issue as it seems to be a compiltation error. The test script is `raise OSError(errno.ETIMEDOUT)`

robert-hh, 2023-09-15:
Just add `import errno` to you script.

knemay, 2023-09-15:
Perfect, thanks

closed 2023-09-15 by knemay

robert-hh, 2023-09-15:
It's still a valid question why `errno` is available in an interactive session like a global, but not visible in the global symbol table. You cannot type in REPL `errno`, but e.g. `errno.ETIMEDOUT`.

knemay, 2023-09-15:
Its also available in non interactive sessions as running a py file. Not sure why is undefined after compilation

jimmo, 2023-09-15:
> It's still a valid question why `errno` is available in an interactive session like a global, but not visible in the global symbol table. You cannot type in REPL `errno`, but e.g. `errno.ETIMEDOUT`.

https://github.com/micropython/micropython/blob/5473200aab492d64866511d96b6f4a7d4cb41393/py/parse.c#L639

Some modules (including errno) are registered as providing constants. The `stm` module is a good example. This allows `machine.mem32[stm.X_BASE + stm.X_FOO]` to be evaluated at runtime as just `machine.mem32[const]`.