micropython/micropython #12474: Arduino ESP32 Nano: Uploading multiple files is not reliable

JohannesMaierhofer, 2023-09-18:
Uploading multiple .py files using Thonny and the newest Micropython firmware v1.20.0-475-g174bb28d8 on 2023-09-14; Arduino Nano ESP32 with ESP32S3 is not stable. Often the files end up being corrupt on the flash of the esp32.
Also uploading only single files that changed is not possible. All other files become corrupt. 

It also happens when uploading the files using the Arduino uPython Lab software.

jimmo, 2023-09-19:
@JohannesMaierhofer Neither Thonny or Arduino uPython Lab are maintained by the micropython team.

Do you see the same issues with `mpremote`? https://docs.micropython.org/en/latest/reference/mpremote.html

You mentioned the issue is with "the newest MicroPython firmware". Do you mean it was working before, or is this the first version you've used.

I notice the Nano ESP32 uses FAT not LittleFS. I will try setting up the Nano filesystem layout on a regular ESP32S3 and see if it works for me. Maybe cc @pillo79 or @iabdalkader have thoughts?

jimmo, 2023-09-19:
I don't have an Arduino ESP32 Nano, but I flashed the Nano firmware onto a similar esp32s3 board, and can't see any issues with the filesystem with mpremote.

FYI @pillo79 `make BOARD=ARDUINO_NANO_ESP32 deploy` doesn't work because it's trying to flash the `factory` partition, where it should be flashing `app0`. I _think_ there just shouldn't be the `factory` line in the board's partitions.csv file?

pillo79, 2023-09-19:
Thanks for the heads-up @jimmo.

> FYI @pillo79 `make BOARD=ARDUINO_NANO_ESP32 deploy` doesn't work because it's trying to flash the `factory` partition, where it should be flashing `app0`. I _think_ there just shouldn't be the `factory` line in the board's partitions.csv file?

That would be an immediate fix, but would also hide the partition from apps (including uPython) looking for it. We need that to implement recovery from broken uploads, so it's not really a viable option. 

From my first glance, it looks like `idf.py` always [uses the "default partition"](https://github.com/espressif/esp-idf/blob/master/components/esptool_py/CMakeLists.txt#L6C5-L12C5) when deciding where to flash, and [`factory` overrides `app`](https://github.com/espressif/esp-idf/blob/master/components/partition_table/parttool.py#L141-L146) in the search (as expected). I don't see a way to change this either... :raised_eyebrow: 

I will ask Espressif for guidance and let you know!

JohannesMaierhofer, 2023-09-19:
Hi, I used mpremote today and faced the same issue. I copied the files either as batch or one by one. In the end some files are not in their directory, others are missing and again others are corrupt. 
Everything works well, if I have only 1 main.py file. 
I didn't try older firmwares yet. I am on macOS if this has any influence?

For curiosity I tried an Arduino RP2040 board and every method worked perfectly.

pillo79, 2023-09-20:
@JohannesMaierhofer thanks for the feedback. I am investigating this and can replicate on a Nano ESP32 - it appears to be an issue with the FAT file system handling in Micropython. I will update this thread with my findings.

pillo79, 2023-09-21:
Found out the root cause - the FAT was getting corrupted due to an issue in the implementation of partial writes in the `Partition.writeblocks()` function. 

FAT needs to update only areas of a hardware Flash block, but that is not possible at hardware level, so the implementation of `writeblocks` reads, erases and rewrites the block with new contents. There was a mistake in this logic that resulted in data being "offset" when a partial write was performed. I've opened a PR with the one-liner that fixes this.

JohannesMaierhofer, 2023-09-21:
yes, that's exactly what the issue looked like from the outside. Wow great, thanks for your effort!!

commit dd58be19 (2023-09-26): esp32: Fix Partition.writeblocks() partial write corruption.

To simulate a partial erase, the code reads a native block, erases it,
and writes back the data before and after the erased area. However, the
current logic was filling the area after the erased block with data
from the beginning of the native block-aligned data, instead of applying
the proper offset.

Fixes #12474.
PR #12493 (2023-09-26): esp32: fix Partition.writeblocks partial write corruption
To simulate a partial erase, the code reads a native block, erases it, and writes back the data before and after the erased area. However, the current logic was filling the area after the erased block with data from the beginning of the native block-aligned data, instead of applying the proper offset.

Fixes #12474.