micropython/micropython #12484: Nucleo H743ZI2 can not get to the REPL in thonny

nspsck, 2023-09-19:
I downloaded the firmware from [the official release](https://micropython.org/download/NUCLEO_H743ZI2/), both a few nightly releases and the stable 1.20 version. None of these could get me a REPL in thonny. I have tested these on 2 NUCLEO_H743ZI2 boards, neither of them worked. However, there is a serial port showing up in the windows device monitor.

I also own some NUCLEO F-Series boards (Nucleo F446RE and F411) and all of them works normally with the REPL. Could this be something ST-Link related? I have updated the ST-Link V3 on board to the newest firmware version, but before that, it didn't work neither.

dpgeorge, 2023-09-20:
Please try using a terminal program other than Thonny.  For example TeraTerm or https://pypi.org/project/mpremote/

nspsck, 2023-09-20:
Hi, I have installed `mpremote` through pip3 with the following command:
```shell
pip3 install mpremote
```
After that I tried:
```shell
mpremote /dev/ttyACM0
```
All the outputs I have gotten is:
```shell
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell
# And a blinking cursor
```
As well as the output of `lsusb` command:
```shell
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 138a:0097 Validity Sensors, Inc. 
Bus 001 Device 003: ID 13d3:5682 IMC Networks SunplusIT Integrated Camera
Bus 001 Device 002: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 005: ID 056a:50b5 Wacom Co., Ltd Pen and multitouch sensor
Bus 001 Device 006: ID 0483:374e STMicroelectronics STLINK-V3
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
As all "files" starts with `tty` under `/dev`:
```shell
tty        tty21      tty35      tty49      tty62      ttyS16     ttyS3
tty0       tty22      tty36      tty5       tty63      ttyS17     ttyS30
tty1       tty23      tty37      tty50      tty7       ttyS18     ttyS31
tty10      tty24      tty38      tty51      tty8       ttyS19     ttyS4
tty11      tty25      tty39      tty52      tty9       ttyS2      ttyS5
tty12      tty26      tty4       tty53      ttyACM0    ttyS20     ttyS6
tty13      tty27      tty40      tty54      ttyprintk  ttyS21     ttyS7
tty14      tty28      tty41      tty55      ttyS0      ttyS22     ttyS8
tty15      tty29      tty42      tty56      ttyS1      ttyS23     ttyS9
tty16      tty3       tty43      tty57      ttyS10     ttyS24     
tty17      tty30      tty44      tty58      ttyS11     ttyS25     
tty18      tty31      tty45      tty59      ttyS12     ttyS26     
tty19      tty32      tty46      tty6       ttyS13     ttyS27     
tty2       tty33      tty47      tty60      ttyS14     ttyS28     
tty20      tty34      tty48      tty61      ttyS15     ttyS29 
```

dpgeorge, 2023-09-20:
Which USB connector are you using?  You should try CN1 by itself.

nspsck, 2023-09-20:
Also, I have tried with v1.19 and v1.18. None of them worked... So most likely I have done something wrong.... again...

I flashed these boards using the [STM32 Cube Programmer](https://www.st.com/en/development-tools/stm32cubeprog.html), and tried to disconnect/reconnect, as well as pressing the reset button. 

Tho, when I am pressing that blue button, LD8 and LD2 blinks togather, so I assume that something is running on the board.