micropython/micropython #12494: CI/Ruff: Unexpected fail.

robert-hh, 2023-09-21:
When pushing a PR, Ruff now complains about a Python script called manifest.py. The error message is:
```
warning: Selection of nursery rule `PLC1901` without the `--preview` flag is deprecated.
warning: Selection of nursery rule `PLC1901` without the `--preview` flag is deprecated.
warning: Selection of nursery rule `PLC1901` without the `--preview` flag is deprecated.
Error: ports/samd/mcu/samd51/manifest_net.py:1:1: F821 Undefined name `include`
Error: ports/samd/mcu/samd51/manifest_net.py:2:1: F821 Undefined name `include`
Error: ports/samd/mcu/samd51/manifest_net.py:4:1: F821 Undefined name `require`
Error: ports/samd/mcu/samd51/manifest_net.py:5:1: F821 Undefined name `require`
Error: ports/samd/mcu/samd51/manifest_net.py:6:1: F821 Undefined name `require`
Error: ports/samd/mcu/samd51/manifest_net.py:8:1: F821 Undefined name `require`
```
The include() and require() calls are valid Python syntax, and the error was not raised before.

andrewleech, 2023-09-21:
Might be from an upgrade to the ruff version behind the scenes?

These are legitimately being marked as undefined because they're not imported / not standard builtins.

I'm also using ruff on my micropython projects and have this in pyproject.toml
```
[tool.ruff.per-file-ignores]
"manifest.py" = ["F821"]  # "Undefined name" for eg. freeze, include, etc.

```

robert-hh, 2023-09-22:
I just found the micropython pyproject.toml. For SAMD, there are MCU-specific manifest.py files, avoiding to replicate the for each board.  But that should be covered by the `**/manifest.py` rule.
But I think the problem is, that there is a second `manifest-net.py` file, which is  used with a different build variant. I did not find a way to evaluate `BOARD_VARIANT` in the manifest.py file.

closed 2023-11-21 by robert-hh