micropython/micropython #12508: rename usb_midi device

mrbbp, 2023-09-23:
soory for my newb question.

Is there a way to rename the usb_midi device name?
by default it named `CircuitPython usb_midi.ports[x]`

i saw a thread to modifing the `.uf2` file with an hex editor, but does not work.
i tried with a different length name and the uf2 is not working
i tried a second time with the same number of char (13), the uf2 seems to work but the device has the same name `CircuitPython...`.

i tried `supervisor.set_usb_identification("Herve Bi", "Herve Bi")`but it's not working.
i'm not sure about my syntax

any help appreciate
regards

circuitPython : 8.2.6
board: waveshare zero

dhalbert, 2023-09-23:
This is a CircuitPython issue, not MicroPython. and is probably covered by one of these: https://github.com/adafruit/circuitpython/issues?q=is%3Aissue+midi++name+is%3Aopen. Please close this issue in MicroPython.

mrbbp, 2023-09-23:
oups so sorry

mrbbp, 2023-09-23:
my apology

closed 2023-09-23 by mrbbp