micropython/micropython #12522: Null pointer dereference in mp_reader_new_file

gwangmu, 2023-09-26:
# Description

We found a null-dereference in the UNIX port version. All PoCs involve `builtins.execfile()` and crash in `mp_reader_new_file` eventually. We've attached three PoCs and their stack trace from ASAN.

[poc.zip](https://github.com/micropython/micropython/files/12729197/poc.zip)

# Proof of Concept

```
$ # build unix port with ASAN, at the root source code directory.
$ export CC=clang
$ export CXX=clang++
$ export CFLAGS="-fsanitize=address -fno-omit-frame-pointer"
$ export CXXFLAGS=$CFLAGS
$ export DEBUG=1
$ make -C mpy-cross -j
$ make -C ports/unix -j all lib
$
$ # run a poc.
$ ./ports/unix/build-standard/micropython <poc_file>
```

# Environment

Ubuntu 20.04
Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz
Memory: 64 GB

# Affected Version

v1.20.0 (commit a3862e726, latest as of 2023-09-26)
v1.20.0 (commit 813d559bc, 2023-06-19)

# Question

We actually found many more crashes with ASAN other than this (including buffer-overflow), but we'd need some advice from `micropython` maintainers before posting issues for all of them.

  1) Do you also consider the crashes in the UNIX port version relevant?
  2) How do you prefer accepting buffer-overflow cases? (via e-mail or via here) We haven't verified their exploitability.

Please let me report the rest of the crashes following your advice. Thank you very much for maintaining `micropython`.

jimmo, 2023-09-27:
Thanks for the detailed report @gwangmu.

This particular issue should be fixed by https://github.com/micropython/micropython/pull/12526

> Do you also consider ASAN crashes through the UNIX port version relevant?

Definitely. Please continue to report them.

> How do you prefer accepting buffer-overflow cases? (via e-mail or via here) We haven't verified their exploitability.

I understand why you're asking, but given the fixes are going to be public anyway we don't really have the resources to do a bigger process (and I'm not sure what that would achieve). I'm happy to just repeat exactly what you've done here. @dpgeorge WDYT?

dpgeorge, 2023-09-27:
> I'm happy to just repeat exactly what you've done here

Yes that's fine, to just open an issue with each of them.  But if there are many, I suggest starting with just a few and we'll see how they go.

gwangmu, 2023-09-27:
Thank you very much @jimmo @dpgeorge . We'll make sure to control the posting pace depending on how it develops.

PR #12526 (2023-10-12): py/builtinevex: Handle invalid filenames for execfile.
Fixes #12522.

If a non-string buffer was passed to execfile, then it would be passed as a non-null-terminated char* to mp_lexer_new_from_file.

This changes mp_lexer_new_from_file to take a qstr instead (as in almost all cases a qstr will be created from this input anyway).

This now makes execfile require a string (not generic buffer) argument, which is probably a good fix to make anyway.

_This work was funded through GitHub Sponsors._
closed 2023-10-12 by dpgeorge