micropython/micropython #12531: Uninitialized memory read in the lexer part

gwangmu, 2023-09-27:
# Description

We observed the UNIX port `micropython` crashes at startup when compiled with memory sanitizer.

# Proof of Concept

```
$ # build unix port with ASAN, at the root source code directory.
$ export CC=clang
$ export CXX=clang++
$ export CFLAGS="-fsanitize=memory -fno-omit-frame-pointer"
$ export CXXFLAGS=$CFLAGS
$ export LDFLAGS=$CFLAGS
$ export DEBUG=1
$ make -C mpy-cross -j
$ make -C ports/unix -j all lib
$
$ # run a poc.
$ export ASAN_OPTIONS="detect_leaks=0"
$ ./ports/unix/build-standard/micropython
```

Please notice that this crash happened without any argument to `micropython`, as long as it was compiled with memory sanitizer.

# Issue

In https://github.com/micropython/micropython/blob/master/py/lexer.c#L153, `lex->fstring_args_idx` was referenced before being initialized, causing a crash when compiled with memory sanitizer.
​
```c
    if (lex->fstring_args_idx) {
        // if there are saved chars, then we're currently injecting fstring args
        if (lex->fstring_args_idx < lex->fstring_args.len) {
            lex->chr2 = lex->fstring_args.buf[lex->fstring_args_idx++];
        } else {
            // no more fstring arg bytes
            lex->chr2 = '\0';
        }
    }
```

# Fix

Use `memset` to initialize the `lex` struct in https://github.com/micropython/micropython/blob/master/py/lexer.c#L837 .
​
```c
    memset(lex, 0, sizeof(*lex)); // insert this line
    lex->reader = reader;
    lex->line = 1;
    lex->column = (size_t)-2; // account for 3 dummy bytes
    lex->emit_dent = 0;
    lex->nested_bracket_level = 0;
    lex->alloc_indent_level = MICROPY_ALLOC_LEXER_INDENT_INIT;
    lex->num_indent_level = 1;
    lex->indent_level = m_new(uint16_t, lex->alloc_indent_level);
    //...
​
```

# Environment

Ubuntu 20.04
Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz
Memory: 64 GB

# Affected Version

v1.20.0 (commit a3862e726, latest as of 2023-09-26)
v1.20.0 (commit 813d559bc, 2023-06-19)

# Comment

The issue analysis and the possible fix were done and written by @chibinz.

jimmo, 2023-09-28:
Thanks @gwangmu and @chibinz! See https://github.com/micropython/micropython/pull/12535 for a fix.

I wasn't able to replicate this unless I also disabled `MICROPY_GC_CONSERVATIVE_CLEAR` (which makes sense to me, this effectively does the memset(0) that you're suggesting as the fix).

However, `MICROPY_GC_CONSERVATIVE_CLEAR` is about improving the precision of the GC, not to guard against missing initialisation (i.e. it should be possible to run correctly without it), so it needs to be fixed.

I've gone with the simpler option of just assigning it directly to zero as it's the same code size as using memset (even if I also remove the other zero-initialisation a few lines earlier).

gwangmu, 2023-09-28:
Thank you for the additional analysis and the quick fix! 

I think it's a better solution with fewer initialization operations.

PR #12535 (2023-09-29): py/lexer: Add missing initialisation for fstring_args_idx.
Fixes #12531 - see discussion there.

---

This was missed in 692d36d779192f32371f7f9daa845b566f26968d. Probably never noticed because everything enables `MICROPY_GC_CONSERVATIVE_CLEAR`, but found via ASAN thanks to @gwangmu & @chibinz.

_This work was funded through GitHub Sponsors._
closed 2023-09-29 by dpgeorge