micropython/micropython #12545: docs: Documentation version don't match image contents

erichiggins, 2023-09-28:
I'm developing with a Raspberry Pi Pico (RP2040) and a nightly image of v1.20.0

```
>>> sys.version
3.4.0; MicroPython v1.20.0-494-g72ef2e629 on 2023-09-27
```

Downloaded from:
- https://micropython.org/download/RPI_PICO/

I've come across two cases so far when the documentation does not match what's actually on the UF2 images.

Examples:

1. `sys.atexit`  
    This function is referenced by the documentation for 1.20.0 and latest
    - https://docs.micropython.org/en/v1.20.0/library/sys.html#sys.atexit
    - https://docs.micropython.org/en/latest/library/sys.html#sys.atexit
    However, it is not defined  
    ```
    import sys
    sys.__dict__
    {'__name__': 'sys', 'argv': [], 'version': '3.4.0; MicroPython v1.20.0-494-g72ef2e629 on 2023-09-27', 'version_info': (3, 4, 0), 'implementation': (name='micropython', version=(1, 20, 0), _machine='Raspberry Pi Pico with RP2040', _mpy=4358), 'platform': 'rp2', 'byteorder': 'little', 'maxsize': 2147483647, 'exit': <function>, 'stdin': <io.FileIO 0>, 'stdout': <io.FileIO 1>, 'stderr': <io.FileIO 2>, 'modules': {'font10': <module 'font10' from 'font10.py'>, 'rp2': <module 'rp2' from 'rp2.py'>}, 'print_exception': <function>}
    ```
2. `ustruct`  
    This module isn't defined in the documentation for 1.20.0 or latest, however, it can still be imported. My current assumption is that ustruct points to struct and they are not separate modules, but this is not reflected in the documentation.
    - https://docs.micropython.org/en/latest/search.html?q=ustruct&check_keywords=yes&area=default
    - https://docs.micropython.org/en/v1.20.0/search.html?q=ustruct&check_keywords=yes&area=default
    ```
    >>> import ustruct
    >>> print(ustruct.__dict__)
    {'__name__': 'struct', 'calcsize': <function>, 'pack': <function>, 'pack_into': <function>, 'unpack': <function>, 'unpack_from': <function>}
    >>> import struct
    >>> print(struct.__dict__)
    {'__name__': 'struct', 'calcsize': <function>, 'pack': <function>, 'pack_into': <function>, 'unpack': <function>, 'unpack_from': <function>}
    ```

robert-hh, 2023-09-28:
`struct` used to be named `ustruct`, but that has changed. For legacy purposes, `ustruct` can still be used.
`sys.atexit()` seems to be implemented only for the UNIX and Windows port. It is pretty common that not all features are supported by all ports. Somewhere in the documentation is a disclaimer telling that.
The documentation is under rework, so any hint for inconsistencies helps.

ricksorensen, 2023-09-28:
See:
[https://forum.micropython.org/viewtopic.php?t=10503](https://forum.micropython.org/viewtopic.php?t=10503)

Maybe some more documentation clarity is needed here.  There are several features of micropython which can be turned on or off for individual builds.

jonnor, 2024-07-13:
Does anyone have proposals for how to address this in the documentation, and can provide a merge request for that?

erichiggins, 2024-07-13:
> Does anyone have proposals for how to address this in the documentation, and can provide a merge request for that?

In my experience, it's best to make sure that URLs never 404, even between versions. If a module/feature is deprecated, the page should either a) redirect to the replacement which should include a note stating the change or b) still exist and document the deprecation, when (or which version) it was deprecated, link to the replacement and GitHub issue where it was discussed.

Another option would be to maintain a page specifically for deprecations like the u-modules, in this case documenting that they are now essentially just pointers to the stdlib names and when the changes were made. Commonly a release notes page would handle this, though those appear to be hosted on GitHub instead of the docs site.

This should also be true for the documentation search index. Searching for `ustruct` with an older version selected doesn't return results any longer, but it should given any of the above solutions.

jonnor, 2024-07-13:
Good points! Yeah, there are some issues in finding the. In fact when searching for "ustruct" on Google, one just gets results from the old MicroPython docs (1.19) - and various poorly maintained forks with quite official sounding names. None of which leads to any information about the renamed modules.

Also, none of the two announcements made mention all the `umodule` modules by their previous names. So they do not showing up in google searches either...
https://github.com/orgs/micropython/discussions/12602
https://github.com/orgs/micropython/discussions/12078

A single page about this change in the documentation would be quite effective in improving findability, and a big improvement over the situation today.

I think that stub/redirect pages for each module would also be good, but maybe more complicated to set up?

erichiggins, 2024-07-15:
Would it be easy to republish the release notes from GitHub into the documentation site?

At least then this release with information about the u-module remaining might appear in search results. 

https://github.com/micropython/micropython/releases/tag/v1.21.0