micropython/micropython #12557: Very odd memory allocation reporting on RP2 version with array.array() and gc.mem_free()

mendenm, 2023-09-30:
The following happens with both the 1.20.0 official build and the nightly build I used below:

```
MicroPython v1.20.0-489-ga3862e726 on 2023-09-20; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import gc
>>> from array import array
>>> gc.mem_free()
209936
>>> a=array('B', range(65536))
>>> gc.collect*)
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> gc.collect()
>>> gc.mem_free()
9904
>>> b=array('B', range(65536))
>>> gc.collect()
>>> gc.mem_free()
9648
>>> c=array('B', range(65536))
>>> gc.collect()
>>> gc.mem_free()
9408
>>> d=array('B', range(65536))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 65536 bytes
>>> del a,b,c
>>> gc.collect()
>>> gc.mem_free()
215904
>>> 
```

Apparently the numbers returned by gc.mem_free() (and similar numbers from micropython.mem_info())  have little to do with actual available memory.  Note that in most microcontroller applications, one should have figured out in advance if there is enough memory available, so this probably doesn't break very many things, but it is clearly wrong.

jimmo, 2023-10-05:
@mendenm Are you using Thonny?

I tried several times to repro this but couldn't, then I remembered an earlier issue that had been reported around gc behavior with Thonny.

This is what I get with mpremote:

```
>>> import gc
>>> from array import array
>>> gc.mem_free()
220768
>>> a=array('B', range(65536))
>>> gc.collect()
>>> gc.mem_free()
155296
>>> b=array('B', range(65536))
>>> gc.collect()
>>> gc.mem_free()
89664
```

This is what I get with Thonny (v 4.1.3 installed from PyPI). Seems to match what you're seeing. I have no idea what's going on, but maybe worth raising an issue at https://github.com/thonny/thonny. I know that Thonny's terminal is actually not a true terminal and it's doing a lot of behind-the-scenes magic.

```
>>> import gc
>>> from array import array
>>> gc.mem_free()
213408
>>> a=array('B', range(65536))
>>> gc.collect()
>>> gc.mem_free()
6800
>>> b=array('B', range(65536))
>>> gc.collect()
>>> gc.mem_free()
6576
>>> 
```

mendenm, 2023-10-05:
Wow, that's weird.  Yes, I was using Thonny.  Results using 'screen' on MacOS:

```
MicroPython v1.20.0-536-gabb44694d on 2023-09-29; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> from array import array
>>> from gc import *
>>> collect()
>>> mem_free()
169216
>>> a=array('B', range(65536))
>>> mem_free()
103264
>>> collect()
>>> mem_free()
103552
```

I wonder how Thonny manages to interfere with mem_free?

I may have to investigate further in free time.

Marcus

closed 2023-10-05 by mendenm