micropython/micropython #12563: Various null-dereference involving framebuf.FrameBuffer1()

gwangmu, 2023-10-02:
# Description

We also found multiple null-dereference cases that all involve `framebuf.FrameBuffer1()`. We suspect that the root cause may be shared with issue #12562 (as they all involved `framebuf.FrameBuffer1()`), but the stack traces of the PoCs below do not contain any mvlsb- or rgb565-related internal functions.

We've attached three sets of PoCs with all different crash locations.

[poc-mp_obj_bytes_hex.zip](https://github.com/micropython/micropython/files/12782454/poc-mp_obj_bytes_hex.zip)
[poc-mp_reader_mem_readbyte.zip](https://github.com/micropython/micropython/files/12782455/poc-mp_reader_mem_readbyte.zip)
[poc-stringio_write.zip](https://github.com/micropython/micropython/files/12782457/poc-stringio_write.zip)

# Proof of Concept

```
$ # build unix port with ASAN, at the root source code directory.
$ export CC=clang
$ export CXX=clang++
$ export CFLAGS="-fsanitize=address -fno-omit-frame-pointer"
$ export CXXFLAGS=$CFLAGS
$ export LDFLAGS=$CFLAGS
$ export DEBUG=1
$ make -C mpy-cross -j
$ make -C ports/unix -j all lib
$
$ # run a poc.
$ export ASAN_OPTIONS="detect_leaks=0"
$ ./ports/unix/build-standard/micropython <poc_file>
```

# Environment

Ubuntu 20.04
Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz
Memory: 64 GB

# Affected Version

v1.20.0 (commit a3862e726, latest as of 2023-09-26)
v1.20.0 (commit 813d559bc, 2023-06-19)
Discovered in the UNIX port version.

jimmo, 2023-10-04:
Thanks @gwangmu. The issue here is similar to #12562 but actually triggering a different bug in the way `framebuf_get_buffer` was calculating the length. See the second commit in #12585 for a fix.

gwangmu, 2023-10-04:
Thank you very much for the quick fix. :)

dpgeorge, 2023-10-16:
Fixed by 03a9fa227d323da7b239a1ef3fa84b0ef45a1621

closed 2023-10-16 by dpgeorge