micropython/micropython #12580: espnow: Resolve availability of aioespnow prior to v1.21 release

glenn20, 2023-10-04:
The [`aioespnow`](https://github.com/micropython/micropython-lib/blob/master/micropython/aioespnow/aioespnow.py) module is now available in `micropython-lib`. However, it is unclear how users are intended to access the `aioespnow` module once v1.21.0 is released. There are two options:

1. Uncomment `# require("aioespnow")` in `ports/esp32/boards/manifest.py` so that aioespnow.py is frozen into esp32 firmware by default (as suggested in PR #6515)
   - This should be done **after** updating the `micropython-lib` submodule reference in micropython from [c113611](https://github.com/micropython/micropython-lib/tree/c113611765278b2fc8dcf8b2f2c3513b35a69b39) and **before** releasing v1.21.0.
   - This requires no changes to the existing documentation

OR

2. Update `docs/library/espnow.html` to indicate that the `aioespnow` modules should be installed with `mip`.
    - **and** remove the commented line in `ports/esp32/boards/manifest.py`.

I am seeking guidance and will submit a PR accordingly. Without action, the documentation will be incomplete/inconsistent  with respect to use of `aioespnow` when v1.21.0 is released.

dpgeorge, 2023-10-04:
Let's go for option 1.  `micropython-lib` will be updated here shortly.

glenn20, 2023-10-04:
Great - thanks. I'll submit a one-liner PR (which will fail commit checks till micropython-lib is updated) - unless you prefer to handle it yourself ;).

PR #12582 (2023-10-05): esp32: Freeze aioespnow module into ESP32 firmware by default.
Uncomment the `# require("aioespnow")` line in `ports/esp32/boards/manifest.py` as proposed in #12580.

This PR can be merged **after** the `lib/micropython-lib` submodule is updated to a revision which includes `micropython/aioespnow/aioespnow.py`. This is expected to occur prior to the release of v1.21.0.

`aioespnow` extends the [`espnow`](https://docs.micropython.org/en/latest/library/espnow.html) module with support for asyncio methods: `arecv()` and `asend()` and is documented in the [espnow documentation](https://docs.microp [... cut]
closed 2023-10-05 by dpgeorge