micropython/micropython #12586: Something funny going on with ints and rp2 ticks_diff

mendenm, 2023-10-04:
In:

MicroPython v1.20.0-536-gabb44694d on 2023-09-29; Raspberry Pi Pico W with RP2040

I am trying to figure out the behavior of ticks_diff on the rp2, but  am getting clearly broken results.  It is not obvious if this is a problem with internal integer representations or or ticks_diff.

```python
#tick diff errors!

import time

for x in ("2**30 - 1", "2**30 - 1", "int(2**30 - 1)", "int(2**30 - 1)", "0x3fffffff", 
    
    ) :
    print(f"{x:15s}", eval(x), "{time.ticks_diff(10, eval(x)):08x}", f"{time.ticks_diff(10, eval(x)):08x}")
```

yields:

```
MicroPython v1.20.0-536-gabb44694d on 2023-09-29; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
2**30 - 1       1073741823 {time.ticks_diff(10, eval(x)):08x} -1000b366
2**30 - 1       1073741823 {time.ticks_diff(10, eval(x)):08x} -1000b3e6
int(2**30 - 1)  1073741823 {time.ticks_diff(10, eval(x)):08x} -1000b4be
int(2**30 - 1)  1073741823 {time.ticks_diff(10, eval(x)):08x} -1000b776
0x3fffffff      1073741823 {time.ticks_diff(10, eval(x)):08x} 0000000b
>>> 
```

Note that eval(x) returns the same value for all of these number representations, but ticks_diff gives different results for identical inputs.  Note that for the non-hex inputs, it is not only wrong, but not even a constant!  Note that this happens the same way if not put inside a format string, so the problem isn't in the format string processing.

```
>>> time.ticks_diff(10, int(2**30 - 1))
-268483614
>>> time.ticks_diff(10, int(2**30 - 1))
-268484430
>>> time.ticks_diff(10, int(2**30 - 1))
-268485966
>>> 
```

jimmo, 2023-10-04:
@mendenm Thanks for the detailed report! What's going on here is pretty subtle, but importantly, ticks_diff is only defined to work on values returned by `ticks_ms()` / `ticks_us()` / `ticks_cpu()`. See https://docs.micropython.org/en/latest/library/time.html#time.ticks_diff

Internally, MicroPython has a different representation for "small integers". 2**30 - 1 is the largest possible small integer value. However, when computing `2**30 - 1`, despite being representable as a small integer, because it came via a "big" integer (i.e. `2**30`) the subtraction operator keeps it internally as a big integer.

So passing the result of evaulating `2**30 - 1` to ticks_diff is not defined as `ticks_ms` (and friends) will never return a big integer, so `ticks_diff` never attempts to handle it being a big integer. This undefined behavior is what's leading to different results each time.

(FYI in #9531 I proposed that we should attempt to re-represent the result of doing arithmetic on big integers into a small integer if possible. This would fix what you're seeing).

> `>>> 0x3fffffff == (2*30 - 1)`
> `False     #OUCH!`

This should be `2**30` not `2*30`. All of this output agrees with CPython.

mendenm, 2023-10-04:
I think  the much bigger problem, now that I have had time to think about it, is the problem with 0x3fffffff == 2**30 - 1 being false. A small integer should still be equal (==)  to a large integer for the same value, shouldn't it? 

The ticks_diff issue might be a red herring here.

Thanks for the quick response.

mendenm, 2023-10-04:
Dang!  I missed that  I had a typo in the 2**30!

Sorry.

Marcus

closed 2023-10-04 by mendenm