micropython/micropython #12592: mem32 writes having no effect on ESP32-S3

pdg137, 2023-10-05:
I'm having trouble getting mem32 writes to show any effect on any registers. I am using a ESP32-S3-WROOM-1 with the v1.20.0 [ESP32-S3 firmware](https://micropython.org/download/ESP32_GENERIC_S3/). To test mem32, I am trying to run [the example from the docs](https://docs.micropython.org/en/v1.18/esp32/tutorial/peripheral_access.html):

```
from micropython import const
from machine import mem32

# Define the register addresses that will be used.
DR_REG_DPORT_BASE = const(0x3FF00000)
DPORT_PERIP_CLK_EN_REG = const(DR_REG_DPORT_BASE + 0x0C0)
DPORT_PERIP_RST_EN_REG = const(DR_REG_DPORT_BASE + 0x0C4)
DPORT_PWM0_CLK_EN = const(1 << 17)
MCPWM0 = const(0x3FF5E000)
MCPWM1 = const(0x3FF6C000)

# Enable CLK and disable RST.
print(hex(mem32[DPORT_PERIP_CLK_EN_REG] & 0xffffffff))
print(hex(mem32[DPORT_PERIP_RST_EN_REG] & 0xffffffff))
mem32[DPORT_PERIP_CLK_EN_REG] |= DPORT_PWM0_CLK_EN
mem32[DPORT_PERIP_RST_EN_REG] &= ~DPORT_PWM0_CLK_EN
print(hex(mem32[DPORT_PERIP_CLK_EN_REG] & 0xffffffff))
print(hex(mem32[DPORT_PERIP_RST_EN_REG] & 0xffffffff))

# Change the MCPWM0 prescaler.
print(hex(mem32[MCPWM0])) # read PWM_CLK_CFG_REG (reset value = 0)
mem32[MCPWM0] = 0x55      # change PWM_CLK_PRESCALE
print(hex(mem32[MCPWM0])) # read PWM_CLK_CFG_REG
```

I ran the code after power-up with no python files on the board. This is the output:

```
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
0x3121b837
0x2ade550
0x3121b837
0x2ade550
0x0
0x0
>>> 
```

The output indicates that none of the register values changed when written to.  Is there something wrong with mem32?

pdg137, 2023-10-05:
I figured it out. There was a hint in #7658 - the addresses are different on the S3.  I think this works:

```
from micropython import const
from machine import mem32

# Define the register addresses that will be used.
DR_REG_DPORT_BASE = const(0x600C_0000)
DPORT_PERIP_CLK_EN0_REG = const(DR_REG_DPORT_BASE + 0x0018)
DPORT_PERIP_RST_EN0_REG = const(DR_REG_DPORT_BASE + 0x0020)
DPORT_PWM0_CLK_EN = const(1 << 17)
MCPWM0 = const(0x6001_E000 + 0x0004)

# Enable CLK and disable RST.
print(hex(mem32[DPORT_PERIP_CLK_EN0_REG] & 0xffffffff))
print(hex(mem32[DPORT_PERIP_RST_EN0_REG] & 0xffffffff))
mem32[DPORT_PERIP_CLK_EN0_REG] |= DPORT_PWM0_CLK_EN
mem32[DPORT_PERIP_RST_EN0_REG] &= ~DPORT_PWM0_CLK_EN
print(hex(mem32[DPORT_PERIP_CLK_EN0_REG] & 0xffffffff))
print(hex(mem32[DPORT_PERIP_RST_EN0_REG] & 0xffffffff))

# Change the MCPWM0 prescaler.
print(hex(mem32[MCPWM0])) # read PWM_CLK_CFG_REG (reset value = 0)
mem32[MCPWM0] = 0x55      # change PWM_CLK_PRESCALE
print(hex(mem32[MCPWM0])) # read PWM_CLK_CFG_REG
```

pdg137, 2023-10-07:
Closing but probably someone should update the docs.

closed 2023-10-07 by pdg137

dpgeorge, 2023-10-09:
> Closing but probably someone should update the docs.

@pdg137 would you like to update the docs based on what you found?

pdg137, 2023-10-11:
@dpgeorge here you go: #12666