micropython/micropython #12611: Creating a directory named same as already frozen module makes the frozen module inaccessible.

megazhuk, 2023-10-07:
I made my own firmware (1.21.0) with frozen module (just folder frozened with `package("utils", base_path="/project/frozen")`) named `'utils'` with some py-files.
It works ok until i not upload new file `'somefile.py'` that was not frozened, to folder `'utils'`. Like `'utils/somefile.py'`
After this, the files in the frozen module become inaccessible: `ImportError: no module named 'utils.somefile'`
Only new file `'somefile.py'` is accessible.

Even if i delete all the files from this folder, but leave the folder to exist, they remain inaccessible.
If i delete the folder with the name of the frozen module, then everything returns to its place.

**I think this is bad behavior.** I think new files should coexist and work together with frozen ones. Until I write a file with the exact same name of an already frozen file in this module into this folder. Just like overwriting a file.
In other words, now if i create folder with name of forzened module, i turn off the frozened module with this name.

andrewleech, 2023-10-07:
Maybe check `sys.path`, if your filesystem is listed earlier than `.frozen` then this is expected behaviour and consistent with cpython.

megazhuk, 2023-10-07:
> Maybe check `sys.path`, if your filesystem is listed earlier than `.frozen` then this is expected behaviour and consistent with cpython.

Thanks! It has become clearer!

But the question remains - how to expand a frozen module by loading additional files? At the moment I understand that this is impossible. You need to change the folder name. Right?

dlech, 2023-10-07:
Changing the folder name seems the simplest, but since https://github.com/micropython/micropython/issues/9018, I think the way to do it is like this (in the not-frozen module with the same name, e.g. `my_module`):

```python
import sys

old_path = sys.path
sys.path = [".frozen"]
try:
    from my_module import *  # import everything from the frozen module
finally:
   sys.path = old_path
```

closed 2023-10-08 by megazhuk

jimmo, 2023-10-09:
@megazhuk see also https://github.com/micropython/micropython/pull/8922 for a proposed solution to this, but I'm not sure we're likely to go through with that. @dlech's workaround is probably the best option.

See also #8381 which hopes to provide an alternative to freezing for the development process. (i.e. most of the benefits of freezing, but with the convenience of the filesystem)