micropython/micropython #12617: RPI PICO W Bluetooth not working with self-compiled firmware

JH2k00, 2023-10-07:
Greetings,

I'm currently working on a project that requires bluetooth and ulab (numpy/scipy for micropython) on the Raspberry Pi Pico W. To this effect, I downloaded the most recent version of micropython with ulab integrated from https://github.com/v923z/micropython-builder/releases, which has been updated 8 hours ago, so it should contain the latest micropython version. With this firmware however, the following code snippet never gets to printing "Done" : 
```py
from micropython import const

import aioble
import bluetooth

# org.bluetooth.service.pulse_oximeter
_PLX_SENSE_UUID = bluetooth.UUID(0x1822)

# org.bluetooth.characteristic.heart_rate_measurement
_PLX_CHAR_HR_UUID = bluetooth.UUID(0x2A37)

# org.bluetooth.characteristic.gap.appearance.xml
_ADV_APPEARANCE_GENERIC_PULSE_OXIMETER = const(0x0C40)

# How frequently to send advertising beacons.
_ADV_INTERVAL_MS = 250000

# Register GATT server.
plx_service = aioble.Service(_PLX_SENSE_UUID)
plx_characteristic = aioble.Characteristic(
    plx_service, _PLX_CHAR_HR_UUID, read=True, notify=True
)
print("Registering")
aioble.register_services(plx_service)
print("Done")
```

While trying to resolve the issue, I downloaded the latest micropython release for the Raspberry Pi Pico W from https://micropython.org/download/RPI_PICO_W/ , and ran the same code with the new firmware. The code executed flawlessly and got to printing "Done". 

[all_firmware.zip](https://github.com/micropython/micropython/files/12837745/all_firmware.zip)

I then tried to compile my own firmware for the RPI PICO W using the instruction at https://github.com/micropython/micropython/tree/master/ports/rp2 and the v1.21.0 tag , in order to check if the problem persists, or if it's exclusive to ulab. The code unfortunately never got to printing "Done" with the new self-compiled firmware (which doesn't contain ulab), leading me to think that the instructions for compiling the firmware may be uncomplete, as the official released version works fine in contrast to the self-compiled one. 

I've zipped and attached all 3 versions of the firmware to this message. I hope that I've provided enough information, and thank you all for your time and help in advance. 

Best regards,
JH

dpgeorge, 2023-10-07:
Did you make sure to run `make BOARD=RPI_PICO_W submodules` to obtain the very latest libraries needed for BLE?  You can check that you have the latest by looking in `lib/pico-sdk` -- should be on commit 6a7db34ff63345a7badec79ebea3aaef1712f374 -- and `lib/cyw43-driver` -- should be on commit 2ab0db6e1fe9265fa9802a95f7f4d60b7f0d655a.

JH2k00, 2023-10-07:
Thank you for your speedy answer ! I went through the whole compiling process again and made sure to run `make BOARD=RPI_PICO_W submodules` . The result was unfortunately the same. I furthermore ran the command `git submodule status` (I hope that's the right one) and got the following output :

> -84f56af13292d8f32c40acbd949bde698ddd4507 ../../lib/asf4
-531cab9c278c947d268bd4c94ecab9153a961b43 ../../lib/axtls
-35aaec4418ad78628a3b935885dd189d41ce779b ../../lib/berkeley-db-1.xx
 77e752abd6a0992334047a48038a5a3960e5c6bc ../../lib/btstack (v1.4-2373-g77e752abd)
 2ab0db6e1fe9265fa9802a95f7f4d60b7f0d655a ../../lib/cyw43-driver (v1.0.1-9-g2ab0db6)
-e78939d32d1ccea9f0ba8bb42c51aceffd386b9b ../../lib/fsp
-e9de7e35f2339598b16cbb375f9992643ed81209 ../../lib/libffi
-5c5d513093075f7245ea522101b17c50aa579af2 ../../lib/libhydrogen
 6ca936f6b588cee702c638eee75c2436e6cf75de ../../lib/lwip (STABLE-2_1_3_RELEASE)
 981743de6fcdbe672e482b6fd724d31d0a0d2476 ../../lib/mbedtls (v2.28.3)
 e025c843b60e93689f0f991d753010bb5bd6a722 ../../lib/micropython-lib (v0.1-1207-ge025c84)
-42849560ba7906f023f61e5f7ff3709ba2c1dfca ../../lib/mynewt-nimble
-7a4c9d946cf1801771fc180acdbf7b878f270093 ../../lib/nrfx
-fa5a554c7944d2a196626f8d3631e44943f9abcc ../../lib/nxp_driver
 6a7db34ff63345a7badec79ebea3aaef1712f374 ../../lib/pico-sdk (1.5.1)
-abc67a11c6db271bedbb9f58be85d6f4e2ea8389 ../../lib/protobuf-c
-928df866e4d287ebc3c60726151513ebee609128 ../../lib/stm32lib
 1fdf29075d4e613eacfa881166015263797db0f6 ../../lib/tinyusb (0.15.0-765-g1fdf29075)
-0803fc519ad7227e841287fb3638d6c8b2f111a1 ../../lib/wiznet5k

Both `lib/pico-sdk` and `lib/cyw43-driver` appear to be at the latest version.

I hope that I've provided all the necessary information and thank you for your time and help !

robert-hh, 2023-10-07:
I tried you test script with a fresh built firmware image, and it seems to work. It prints:
```
>>> import ble_test
Registering
Done
```

Pushing Ctrl-B in REPL, the firmware comes up with the message:
```
MicroPython v1.22.0-preview.1.g523284777 on 2023-10-07; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
```
Do you get the same with your build?

robert-hh, 2023-10-07:
I get the same result with a plain v1.21.0:
```
MicroPython v1.21.0 on 2023-10-07; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import ble_test
Registering
Done
```

JH2k00, 2023-10-07:
I ran the script with Thonny and my build and got the following output :

> MicroPython v1.21.0 on 2023-10-07; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Registering

I furthermore ran the script with the RPI_PICO_W.uf2 firmware from https://github.com/v923z/micropython-builder/releases and got the following output : 

> MicroPython v1.22.0-preview.1.g523284777 on 2023-10-07; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Registering

Running the script with the firmware from https://micropython.org/download/RPI_PICO_W/ got me : 

> MicroPython v1.21.0 on 2023-10-06; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Registering
Done

Did you follow the instructions at https://github.com/micropython/micropython/tree/master/ports/rp2 exactly, or did you have an extra step ? 

Thank you for your time and help !

robert-hh, 2023-10-07:
> Did you follow the instructions at https://github.com/micropython/micropython/tree/master/ports/rp2 exactly, or did you have an extra step ?

I have set up my build environment quite a while ago. But the steps to build a firmware are the same as you mentioned. I use a rather old C compiler, which is `gcc-arm-none-eabi-10.3-2021.07`, but for no specific reason.

JH2k00, 2023-10-08:
I set up a new Ubuntu 22.04.3 LTS instance using wsl and went through the building process again. The result was unfortunately the same. The official ulab release at https://github.com/v923z/micropython-builder/releases having the same problem is also weird. Would it be possible for you to compile the latest micropython with ulab locally (https://github.com/v923z/micropython-builder) and try the script ? I hope I'm not taking up too much of your time. Thank you very much !

P.S. : 
Here is the output after running `make BOARD=RPI_PICO_W` in case it's helpful for resolving the issue. 

> [ -e build-RPI_PICO_W/Makefile ] || cmake -S . -B build-RPI_PICO_W -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=RPI_PICO_W -DMICROPY_BOARD_DIR=/home/jh/micropython/ports/rp2/boards/RPI_PICO_W
PICO_SDK_PATH is /home/jh/micropython/lib/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Build type is MinSizeRel
PICO target board is pico_w.
Using CMake board configuration from /home/jh/micropython/lib/pico-sdk/src/boards/pico_w.cmake
Using board configuration from /home/jh/micropython/lib/pico-sdk/src/boards/include/boards/pico_w.h
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
TinyUSB available at /home/jh/micropython/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/jh/micropython/lib/btstack
cyw43-driver available at /home/jh/micropython/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/jh/micropython/lib/lwip
Pico W Wi-Fi build support available.
Found User C Module(s):
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jh/micropython/ports/rp2/build-RPI_PICO_W
make -s -C build-RPI_PICO_W || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
[  0%] Creating directories for 'PioasmBuild'
[  0%] No download step for 'PioasmBuild'
[  0%] No update step for 'PioasmBuild'
[  1%] No patch step for 'PioasmBuild'
[  1%] Performing configure step for 'PioasmBuild'
loading initial cache file /home/jh/micropython/ports/rp2/build-RPI_PICO_W/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/PioasmBuild-cache-MinSizeRel.cmake
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jh/micropython/ports/rp2/build-RPI_PICO_W/pioasm
[  2%] Performing build step for 'PioasmBuild'
[ 10%] Building CXX object CMakeFiles/pioasm.dir/main.cpp.o
[ 20%] Building CXX object CMakeFiles/pioasm.dir/pio_assembler.cpp.o
[ 30%] Building CXX object CMakeFiles/pioasm.dir/pio_disassembler.cpp.o
[ 40%] Building CXX object CMakeFiles/pioasm.dir/gen/lexer.cpp.o
[ 50%] Building CXX object CMakeFiles/pioasm.dir/gen/parser.cpp.o
[ 60%] Building CXX object CMakeFiles/pioasm.dir/c_sdk_output.cpp.o
[ 70%] Building CXX object CMakeFiles/pioasm.dir/python_output.cpp.o
[ 80%] Building CXX object CMakeFiles/pioasm.dir/hex_output.cpp.o
[ 90%] Building CXX object CMakeFiles/pioasm.dir/ada_output.cpp.o
[100%] Linking CXX executable pioasm
[100%] Built target pioasm
[  2%] No install step for 'PioasmBuild'
[  2%] Completed 'PioasmBuild'
[  2%] Built target PioasmBuild
[... 519 lines cut ...]
[ 94%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/pico-sdk/src/rp2_common/pico_stdio_uart/stdio_uart.c.obj
[ 95%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/pico-sdk/src/rp2_common/pico_unique_id/unique_id.c.obj
[ 95%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/tusb.c.obj
[ 95%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/common/tusb_fifo.c.obj
[ 95%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c.obj
[ 95%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj
[ 96%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/device/usbd.c.obj
[ 96%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/device/usbd_control.c.obj
[ 96%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/class/audio/audio_device.c.obj
[ 96%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/class/cdc/cdc_device.c.obj
[ 96%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/class/dfu/dfu_device.c.obj
[... cut ...]
[ 98%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/tinyusb/src/class/video/video_device.c.obj
[ 98%] Building C object CMakeFiles/firmware.dir/home/jh/micropython/lib/pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c.obj
[ 99%] Linking CXX executable firmware.elf
   text    data     bss     dec     hex filename
 806108       0   41712  847820   cefcc /home/jh/micropython/ports/rp2/build-RPI_PICO_W/firmware.elf
[100%] Built target firmware

robert-hh, 2023-10-08:
That's quite a job. So I made another test:
If I build the firmware with the arm-none-eabi v12.2.1, that came with my debian 12 version, the test fails.
If I build the firmware with arm-none-eabi v12.3 or v10.3, that I downloaded from the ARM web site, the test succeeds.  Link: https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads, version `AArch32 bare-metal target (arm-none-eabi)`.
So it seems to be a compiler issue. I had similar effects with previous versions.

JH2k00, 2023-10-08:
That did the trick ! I compiled a version with ulab with the latest compiler and it worked flawlessly. Thank you and have a nice Sunday !

Josverl, 2023-10-08:
> If I build the firmware with the arm-none-eabi v12.2.1, that came with my debian 12 version, the test fails.\nIf I build the firmware with arm-none-eabi v12.3 or v10.3, that I downloaded from the ARM web site, the test succeeds.

Is this prerew dependency something that could/shoukd be checked in the makefile or elsewhere?

robert-hh, 2023-10-08:
There should be a note which compiler version is recommended. At least platform.python_compiler() tells you which one was used. For the build oneline, it's version 13.2.

jimmo, 2023-10-09:
Unless there's a known bug in gcc 12.2.1 (which we should assume is unlikely but not impossible *), MicroPython should work on all gcc versions, so I think we should possibly assume there's a latent issue here that needs investigating.

To put it another way, except for some LTO issues, we do not specify a minimum gcc version anywhere (e.g. in #8813 we were testing I think back to gcc 4.x), and so this would be a new thing for MicroPython.

* 12.3 fix list is here: https://gcc.gnu.org/bugzilla/buglist.cgi?bug_status=RESOLVED&resolution=FIXED&target_milestone=12.3

robert-hh, 2023-10-09:
I did actually not try, what happens. But I had similar observation already before with other C-Compiler versions. And in fact C-Compiler means the compiler, libs, and other code required for the build.
With the PICO W, I did not hook up the Debugger to tell the state of the board, when it seems to block. I just did the same with a SAMD board & BLE (easier to hook up JTAG & Ozone). It locks when used with the v12.2.1 compiler and works with the v10.3 compiler. In that case, not even the first REPL comes up. But the situation happens only if the code size exceeds 512k. In that 'lock' state the code is actual in the REPL loop, waiting for characters to arrive, but the USB device is not registered. I'll try further with the PICO board.

robert-hh, 2023-10-09:
With the Pico W, it's at that point.
```
(gdb) list 187
182 void __attribute__((noreturn)) __attribute__((weak)) _exit(__unused int status) {
183 #if PICO_ENTER_USB_BOOT_ON_EXIT
184     reset_usb_boot(0,0);
185 #else
186     while (1) {
187         __breakpoint();
188     }
189 #endif
190 }
191 

(gdb) bt
#0  _exit (status=status@entry=1)
    at /home/robert/Downloads/MicroPython/micropython/lib/pico-sdk/src/rp2_common/pico_runtime/runtime.c:187
#1  0x1005d2f2 in panic (fmt=0x100c194c "Out of memory")
    at /home/robert/Downloads/MicroPython/micropython/lib/pico-sdk/src/rp2_common/pico_runtime/runtime.c:325
#2  0x1005e740 in check_alloc (mem=<optimized out>, size=<optimized out>)
    at /home/robert/Downloads/MicroPython/micropython/lib/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c:26
#3  check_alloc (mem=<optimized out>, size=<optimized out>)
    at /home/robert/Downloads/MicroPython/micropython/lib/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c:23
#4  0x1005e76e in __wrap_malloc (size=600)
    at /home/robert/Downloads/MicroPython/micropython/lib/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c:44
#5  0x1004780e in cybt_fw_download_prepare (p_hex_buf=<synthetic pointer>, 
    p_write_buf=<synthetic pointer>)
    at /home/robert/Downloads/MicroPython/micropython/lib/pico-sdk/src/rp2_common/pico_cyw43_driver/cybt_shared_bus/cybt_shared_bus.c:70
#6  cyw43_btbus_init (self=self@entry=0x200092b4 <cyw43_state>)
    at /home/robert/Downloads/MicroPython/micropython/lib/pico-sdk/src/rp2_common/pico_cyw43_driver/cybt_shared_bus/cybt_shared_bus.c:120
#7  0x10046edc in cyw43_ensure_bt_up (self=0x200092b4 <cyw43_state>)
    at /home/robert/Downloads/MicroPython/micropython/lib/cyw43-driver/src/cyw43_ctrl.c:753
#8  0x100475a6 in cyw43_bluetooth_hci_init ()
    at /home/robert/Downloads/MicroPython/micropython/lib/cyw43-driver/src/cyw43_ctrl.c:764
#9  0x10039d92 in hci_transport_cyw43_open ()
    at /home/robert/Downloads/MicroPython/micropython/lib/pico-sdk/src/rp2_common/pico_cyw43_driver/btstack_hci_transport_cyw43.c:57
#10 0x10040c44 in hci_power_control_on ()
    at /home/robert/Downloads/MicroPython/micropython/lib/btstack/src/hci.c:4953
--Type <RET> for more, q to quit, c to continue without paging--
#11 0x1004300e in hci_power_control_state_off (power_mode=HCI_POWER_ON)
    at /home/robert/Downloads/MicroPython/micropython/lib/btstack/src/hci.c:5087
#12 hci_power_control (power_mode=power_mode@entry=HCI_POWER_ON)
    at /home/robert/Downloads/MicroPython/micropython/lib/btstack/src/hci.c:5212
#13 0x10030804 in mp_bluetooth_btstack_port_start ()
    at /home/robert/Downloads/MicroPython/micropython/ports/rp2/mpbtstackport.c:154
#14 0x1001a206 in mp_bluetooth_init ()
    at /home/robert/Downloads/MicroPython/micropython/extmod/btstack/modbluetooth_btstack.c:655
#15 0x1001c4e4 in bluetooth_ble_active (n_args=<optimized out>, args=<optimized out>)
    at /home/robert/Downloads/MicroPython/micropython/extmod/modbluetooth.c:295
#16 0x1000fc7a in fun_builtin_var_call (self_in=0x10069568 <bluetooth_ble_active_obj>, n_args=2, 
    n_kw=0, args=0x20041ac0) at /home/robert/Downloads/MicroPython/micropython/py/objfun.c:119
#17 0x1001770c in mp_call_function_n_kw (fun_in=0x10069568 <bluetooth_ble_active_obj>, n_args=2, 
    n_kw=0, args=0x20041ac0) at /home/robert/Downloads/MicroPython/micropython/py/runtime.c:712
#18 0x100177d0 in mp_call_method_n_kw (n_args=<optimized out>, n_kw=0, args=args@entry=0x20041abc)
    at /home/robert/Downloads/MicroPython/micropython/py/runtime.c:728
#19 0x20003150 in mp_execute_bytecode (code_state=code_state@entry=0x20041aa8, 
    inject_exc=inject_exc@entry=0x0) at /home/robert/Downloads/MicroPython/micropython/py/vm.c:1042
#20 0x1000fda4 in fun_bc_call (self_in=0x10073a5d <fun_data_aioble_server_register_services+24>, 
    n_args=0, n_kw=536974240, args=0x1001770d <mp_call_function_n_kw+56>)
    at /home/robert/Downloads/MicroPython/micropython/py/objfun.c:273
#21 0x1001770c in mp_call_function_n_kw (fun_in=0x200193a0, n_args=0, n_kw=0, 
    args=args@entry=0x2001a648) at /home/robert/Downloads/MicroPython/micropython/py/runtime.c:712
#22 0x200030be in mp_execute_bytecode (code_state=code_state@entry=0x2001a630, 
    inject_exc=inject_exc@entry=0x0) at /home/robert/Downloads/MicroPython/micropython/py/vm.c:957
#23 0x1000fda4 in fun_bc_call (self_in=0x20041c84, n_args=1, n_kw=536980400, 
    args=0x1001770d <mp_call_function_n_kw+56>)
    at /home/robert/Downloads/MicroPython/micropython/py/objfun.c:273
#24 0x1001770c in mp_call_function_n_kw (fun_in=0x2001abb0, n_args=1, n_kw=0, args=0x20041c8c)
    at /home/robert/Downloads/MicroPython/micropython/py/runtime.c:712
#25 0x100177d0 in mp_call_method_n_kw (n_args=<optimized out>, n_kw=0, args=args@entry=0x20041c84)
[... 18 lines cut ...]
#33 0x10001bbe in process_import_at_level (override_main=false, outer_module_obj=0x0, 
    level_mod_name=1972, full_mod_name=1972)
    at /home/robert/Downloads/MicroPython/micropython/py/builtinimport.c:505
#34 mp_builtin___import___default (n_args=n_args@entry=5, args=args@entry=0x20041e3c)
    at /home/robert/Downloads/MicroPython/micropython/py/builtinimport.c:600
#35 0x10017852 in mp_import_name (name=<optimized out>, fromlist=<optimized out>, 
    level=<optimized out>) at /home/robert/Downloads/MicroPython/micropython/py/runtime.c:1525
#36 0x2000331a in mp_execute_bytecode (code_state=code_state@entry=0x20041ec8, 
[... cut ...]
    at /home/robert/Downloads/MicroPython/micropython/py/runtime.c:686
#40 0x1002ba9c in parse_compile_execute (source=source@entry=0x20041fa0, 
    input_kind=<optimized out>, exec_flags=exec_flags@entry=22)
    at /home/robert/Downloads/MicroPython/micropython/shared/runtime/pyexec.c:123
#41 0x1002bd30 in pyexec_friendly_repl ()
    at /home/robert/Downloads/MicroPython/micropython/shared/runtime/pyexec.c:675
#42 0x1002ece4 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/robert/Downloads/MicroPython/micropython/ports/rp2/main.c:195
```

dpgeorge, 2023-10-09:
> 0x1005d2f2 in panic (fmt=0x100c194c "Out of memory")

Hmmm, that looks like it's running out of memory when trying to bring up BLE.

In `boards/RPI_PICO_W/mpconfigboard.cmake` can you try increasing `set(MICROPY_C_HEAP_SIZE 4096)` to a bigger number like 8192?

robert-hh, 2023-10-09:
Increasing the heap helps. Still unclear why other GCC stacks work. Seems like allocate less less RAM at other places.

robert-hh, 2023-10-09:
6144 or 5120 are fine as well.

dpgeorge, 2023-10-09:
It is possible to redefine `cyw43_malloc` so that it uses a statically allocated piece of memory, or memory from the GC heap.  Currently it uses the C-level system malloc and that's what's having out-of-memory errors.

robert-hh, 2023-10-09:
Using the defines below works, but I do not know if it has other side effects. If the allocated memory subject to garbage collection? Can the C heap be reduced if m_malloc is used?
```
#ifndef cyw43_malloc
#define cyw43_malloc m_malloc
#endif
#ifndef cyw43_free
#define cyw43_free m_free
#endif
```

dpgeorge, 2023-10-10:
Better to use `m_tracked_calloc` and `m_tracked_free`.  That automatically tracks the pointers so they aren't freed by the GC.

robert-hh, 2023-10-10:
That works as well. The simple test script works, and from the test suite all tests are either skipped or passed. Will you make a PR or should I do that?

dpgeorge, 2023-10-10:
> Will you make a PR or should I do that?

Feel free to make a PR, thanks.

dpgeorge, 2023-10-16:
Fixed by f84b4617cb6e15f943ea81efeeafa651a9553445

robofwy, 2023-11-10:
I also encountered this issue with v1.21.0 and can confirm that using GCC >= 12.3, increasing the stack size to 5120 or applying #12647 will fix it.

I think the official binary release of v1.21.0 built with GCC 13.2.0 could be a little unpredictable because of this. I am guessing with GCC 13.2.0 (as used by the official binary release) the stack grows to just less than 4096 but is probably very close so could there be other situations with Bluetooth/BLE that cause it to go over the limit?

Therefore, if wanting to use Bluetooth/BLE on a Pico W, I would suggest patching v1.21.0 with #12647 and building a custom firmware or using the latest master.

Would it not be an idea to build a v1.21.1 release as v1.21.0 with the #12647 patch applied?