micropython/micropython #12622: ESP32 port doesn't build with `MICROPY_PY_NETWORK_WLAN` set to `0` in `mpconfigboard.h`

agatti, 2023-10-08:
The ESP32 port for MicroPython 1.12.0 cannot be built with `MICROPY_PY_NETWORK_WLAN` set to `0` in my custom board's `mpconfigboard.h` file.  I'm disabling WiFi support to save flash space and with the following `mpconfigboard.h` file:

```c
#define MICROPY_ESPNOW                   (0)
#define MICROPY_PY_MACHINE_DAC           (0)
#define MICROPY_PY_MACHINE_I2S           (0)
#define MICROPY_PY_NETWORK_WLAN          (0)
#define MICROPY_HW_ENABLE_SDCARD         (0)
#define MICROPY_HW_ENABLE_MDNS_QUERIES   (0)
#define MICROPY_HW_ENABLE_MDNS_RESPONDER (0)
#define MICROPY_PY_NETWORK_LAN           (0)
```

MicroPython fails to link with the following error message (ESP-IDF 5.0.2 on Arch Linux, clean build):

```bash
[1207/1209] Linking CXX executable micropython.elf
FAILED: micropython.elf 
: && /home/agatti/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address  CMakeFiles/micropython.elf.dir/project_elf_src_esp32.c.obj -o micropython.elf [list of .a files being linked] -u esp_app_desc  -Wl,--wrap=vPortCleanUpTCB  -u pthread_include_pthread_impl  -u pthread_include_pthread_cond_impl  -u pthread_include_pthread_local_storage_impl  -u pthread_include_pthread_rwlock_impl  -u ld_include_highint_hdl  -u start_app  -u start_app_other_cores  -L "/home/agatti/tmp/micropython/ports/esp32/build-CUSTOM_BOARD/esp-idf/esp_system/ld"  -T memory.ld  -T sections.ld  -u __ubsan_include  -L "/home/agatti/tmp/esp-idf/components/esp_rom/esp32/ld"  -T esp32.rom.ld  -T esp32.rom.api.ld  -T esp32.rom.libgcc.ld  -T esp32.rom.newlib-data.ld  -T esp32.rom.syscalls.ld  -T esp32.rom.newlib-funcs.ld  -Wl,--wrap=longjmp  -u esp_dport_access_reg_read  -L "/home/agatti/tmp/esp-idf/components/soc/esp32/ld"  -T esp32.peripherals.ld  -Wl,--undefined=uxTopUsedPriority  -Wl,--undefined=FreeRTOS_openocd_params  -u app_main  -lc  -lm  esp-idf/newlib/libnewlib.a  -u newlib_include_heap_impl  -u newlib_include_syscalls_impl  -u newlib_include_pthread_impl  -u newlib_include_assert_impl  -Wl,--wrap=_Unwind_SetEnableExceptionFdeSorting  -Wl,--wrap=__register_frame_info_bases  -Wl,--wrap=__register_frame_info  -Wl,--wrap=__register_frame  -Wl,--wrap=__register_frame_info_table_bases  -Wl,--wrap=__register_frame_info_table  -Wl,--wrap=__register_frame_table  -Wl,--wrap=__deregister_frame_info_bases  -Wl,--wrap=__deregister_frame_info  -Wl,--wrap=_Unwind_Find_FDE  -Wl,--wrap=_Unwind_GetGR  -Wl,--wrap=_Unwind_GetCFA  -Wl,--wrap=_Unwind_GetIP  -Wl,--wrap=_Unwind_GetIPInfo  -Wl,--wrap=_Unwind_GetRegionStart  -Wl,--wrap=_Unwind_GetDataRelBase  -Wl,--wrap=_Unwind_GetTextRelBase  -Wl,--wrap=_Unwind_SetIP  -Wl,--wrap=_Unwind_SetGR  -Wl,--wrap=_Unwind_GetLanguageSpecificData  -Wl,--wrap=_Unwind_FindEnclosingFunction  -Wl,--wrap=_Unwind_Resume  -Wl,--wrap=_Unwind_RaiseException  -Wl,--wrap=_Unwind_DeleteException  -Wl,--wrap=_Unwind_ForcedUnwind  -Wl,--wrap=_Unwind_Resume_or_Rethrow  -Wl,--wrap=_Unwind_Backtrace  -Wl,--wrap=__cxa_call_unexpected  -Wl,--wrap=__gxx_personality_v0  -u __cxa_guard_dummy  -lstdc++  esp-idf/pthread/libpthread.a  -lgcc  esp-idf/cxx/libcxx.a  -u __cxx_fatal_exception  -L "/home/agatti/tmp/esp-idf/components/esp_phy/lib/esp32"  -u include_esp_phy_override  -lphy  -lrtc  esp-idf/esp_phy/libesp_phy.a  -lphy  -lrtc  esp-idf/esp_phy/libesp_phy.a  -lphy  -lrtc  -u vfs_include_syscalls_impl  -L "/home/agatti/tmp/esp-idf/components/esp_wifi/lib/esp32" && :
/home/agatti/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modespnow.c.obj):(.literal.espnow_config+0x10): undefined reference to `esp_initialise_wifi'
/home/agatti/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modespnow.c.obj): in function `espnow_config':
/home/agatti/tmp/micropython/ports/esp32/modespnow.c:256: undefined reference to `esp_initialise_wifi'
/home/agatti/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modespnow.c.obj): in function `espnow_init':
/home/agatti/tmp/micropython/ports/esp32/modespnow.c:192: undefined reference to `esp_initialise_wifi'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/agatti/tmp/micropython/ports/esp32/build-CUSTOM_BOARD/log/idf_py_stderr_output_124944 and /home/agatti/tmp/micropython/ports/esp32/build-CUSTOM_BOARD/log/idf_py_stdout_output_124944
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:62: all] Error 1
```

In the `mpconfigboard.h` file I mentioned earlier the only #define that triggers the issue is `MICROPY_PY_NETWORK_WLAN` - to make anything build it must be set to `1`.

Do I have to fork the whole ESP32 port for this and remove all traces of the ESP-NOW module or am I missing something else?  `MICROPY_PY_NETWORK_WLAN` in `mpconfigport.h` looks like it is meant to be overridable, but in practice it doesn't seem to be.

robert-hh, 2023-10-08:
What happens if you drop the definition of `MICROPY_PY_NETWORK_WLAN` instead of setting to 0. Maybe at some places `#ifdef` is used instead of `#if`.

agatti, 2023-10-08:
From what I wrote it looked like it must be present and set, my apologies.  The issue in question occurs when explicitly setting the #define to `0`.  If it is not set it defaults to `1` in `mpconfigport.h` and everything builds fine, although with WiFi support and all that it entails.

robert-hh, 2023-10-08:
The problem is in modespnow.h. It is missing a `#if MICROPY_ESPNOW`/`#endif` bracket. If you add
`#if MICROPY_ESPNOW` at the beginning and an `#endif` at the end of modespnow.c, and set **both**  `MICROPY_PY_NETWORK_WLAN` and ` MICROPY_ESPNOW` to 0, the firmware builds and runs.

agatti, 2023-10-08:
Ok, then it's a bug in MicroPython, thanks!  I guess a PR is in order then...

robert-hh, 2023-10-08:
Then, there is a lot more that can be skipped from the manifest.py, like the aioespnow, umqtt.*, the networking bundle, and espnow.py from esp32/modules.

agatti, 2023-10-08:
Right, for that I also have my own manifest with just the modules I need - I was planning to start messing with the SDK configuration as well (to remove LWIP and its dependencies) once I had the MicroPython side sorted out first.

Thanks for the help, I'll prepare a PR for this in the meantime!

robert-hh, 2023-10-08:
> Ok, then it's a bug in MicroPython

Let's say it's an inconsistency. It is not common to use an ESP32 without WLAN, because that's the major feature of this platform. If you do not need WLAN or Ethernet, there are other ports that perform much better.
But being able to omit ESPNOW should be possible.

agatti, 2023-10-08:
I agree, but for this project there are a few more ESP32s being used for other purposes so the natural choice is to stick with the same MCU to keep things simple (and to cross a MOQ threshold to reduce BOM costs :)).

glenn20, 2023-10-08:
Ah - thanks for picking up on that. In early drafts, the modespnow.c was conditionally included in the Makefile (as is still the case for esp8266), but that was lost in the transition to cmake for esp32 (due to my oversight).

dpgeorge, 2023-10-09:
It's best to try and keep all configuration within C, rather than having make-level config options, if possible.

commit 9f835df3 (2023-10-10): esp32,esp8266: Rename MICROPY_ESPNOW to MICROPY_PY_ESPNOW.

For consistency with other Python-level modules.

Also add the corresponding missing preprocessor guard to esp32/modespnow.c,
so that this port compiles if MICROPY_PY_ESPNOW and MICROPY_PY_NETWORK_WLAN
are set to 0.

Fixes #12622.
PR #12643 (2023-10-10): esp32/modespnow.c: Fix: Wrap code in MICROPY_PY_ESPNOW directive.
Bug: Micropython will fail to compile if `MICROPY_PY_ESPNOW` and `MICROPY_PY_NETWORK_WLAN` are set to 0 due to dependence on `esp_initialise_wifi`.

Fix: Code in modespnow.c should be  excluded from comilation if `MICROPY_PY_ESPNOW` is set to 0.

Also renames `MICROPY_ESPNOW` directive to `MICROPY_PY_ESPNOW` (see #12625).

Tests:

- `#define MICROPY_PY_ESPNOW (1)`: build and passed `tests/multi-espnow`.
- `#define MICROPY_PY_ESPNOW (0)`: Compiles for all board definitions and functional on ESP32/S2/S3.
- Compiles and tested on E [... cut]
closed 2023-10-10 by dpgeorge