micropython/micropython #12627: References to bound methods that are the same have different hash values (v1.21) (CPython incompatibility)

ned-pcs, 2023-10-08:
This is a sister issue to #5233 (which regards bound method equality).
I'm adding it to keep it on peoples' minds.

v1.21 added generic hashing for a number of classes that didn't previously have hashing.

So now you can make (say) a `set` of bound method references to use as callbacks, but discover that your method gets called multiple times:

```
>>> class A:
...     def a(self):
...         pass
...
...
...
>>> a = A()
>>> b = a.a
>>> c = a.a
>>> b == c
False
>>> hash(b)
5234552832
>>> hash(c)
5234553088
>>> s = set()
>>> s.add(a.a)
>>> s.add(b)
>>> s.add(c)
>>> s
{<bound_method 13800fc40 <A object at 13800ee40>.<function a at 0x13800ec00>>,
<bound_method 13800f000 <A object at 13800ee40>.<function a at 0x13800ec00>>,
<bound_method 13800f100 <A object at 13800ee40>.<function a at 0x13800ec00>>}
```
PR #9014 adds a `binary_op` method for equality testing, but doesn't add the corresponding fix for hashing.
@DvdGiessen offered a [commit of his ](https://github.com/micropython/micropython/pull/9014#issuecomment-1645432060)that added both equality and hashing.

dpgeorge, 2023-10-08:
Yes, this has the same issue and fix as with equality, #5233: hash the self and function objects of the bound method, and not the instance of the bound method (which changes with location on the heap).

ned-pcs, 2023-10-09:
@dpgeorge you were concerned about code size increase due to adding hashing to bound methods; do you feel that the usefulness of this fix outweighs the code size delta?

dpgeorge, 2023-10-10:
> you were concerned about code size increase due to adding hashing to bound methods

Yes.  Adding both equality and hashing to bound methods requires both unary and binary op functions, which adds a fair bit of code.

> do you feel that the usefulness of this fix outweighs the code size delta?

I'm not sure... but if we are to spend bytes implementing features, this is the sort of thing that it is good to spend bytes on, because it cannot (easily) be a runtime extension like a pure Python module can be.

dpgeorge, 2023-10-13:
Fixed by 4f5e165d0bf9ba338df3cdab4266556c4c5a70c7

closed 2023-10-13 by dpgeorge