micropython/micropython #12632: v1.21.0 Failed to use hostname

ruihua-code, 2023-10-09:
ip can be successfully used, but hostname cannot complete the request. May I ask if my method is wrong?

```py
    # TODO: Deprecated. Use network.hostname(name) instead.
    # wlan.config(dhcp_hostname=hostname)
    network.hostname(hostname)
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    print("hostname:", network.hostname())
    while not wlan.active(True):
        print("wait wlan")
```

### http api

> * url: /cmd

> * method: POST

### use ip _Success_
* postman image:
[image]

### use hostname _Fail_
* postman image:
[image]

### build info

* **hostname**: wood-chip
* **ifconfig**: ('192.168.188.131', '255.255.254.0', '192.168.188.1', '114.114.114.114')
* **uname**: (sysname='esp32', nodename='esp32', release='1.21.0', version='v1.21.0-dirty on 2023-10-09', machine='Generic ESP32S3 
module with ESP32S3')
* **platform**: MicroPython-1.21.0-xtensa-IDFv5.0.4-158-g3f55377ab4-with-newlib4.1.0

dpgeorge, 2023-10-09:
It looks like you're trying to use mDNS.  Do you have mDNS configured and working on your PC so that it can query the `wood-chip` name?  Are you able to use mDNS with a different (non-MicroPython) device?

ruihua-code, 2023-10-10:
> It looks like you're trying to use mDNS. Do you have mDNS configured and working on your PC so that it can query the `wood-chip` name? Are you able to use mDNS with a different (non-MicroPython) device?

I use print to output ```network.hostname()``` as ```wood-chip```

Do you still need to do mDNS configuration on the PC? I don't understand this

In v1.20.0, I used it like this: ```wlan.config(dhcp_hostname="esp")```

Apis can then be requested using ```esp.local``` on phones and computers

Seruf-Che, 2023-11-18:
Were you able to find a solution?

closed 2023-12-12 by ruihua-code