micropython/micropython #12638: Sporadic GuruMeditation, presumed related to TCP/IP LWIP

moefear85, 2023-10-09:
* For build issues, please include full details of your environment, compiler versions, command lines, and build output.

* Please provide as much information as possible about the version of MicroPython you're running, such as:
 - firmware file name
 - git commit hash and port/board
 - version information shown in the REPL (hit Ctrl-B to see the startup message)

* Remove all placeholder text above before submitting.

I'm running some simple code that just connects to wifi and sends out a udp heartbeat once per second. After random amounts of time (sometimes it takes long), GuruMeditation happens. I have no idea how to understand the crash dump, but it seems related to the LWIP subsystem, not my code. I'm on the following commit, running on a standard esp32cam:

micropython: 294baf52b346e400e2255c6c1e82af5b978b18f7 (grafted, HEAD, tag: v1.20.0)
esp-idf-v4.4 (release branch): b550086 cloned on 01-10-23

I compiled the running image myself after cloning the micropython repository. If micropython is not compatible with the mentioned esp-idf version, which branch (or what latest commit) is it definitely compatible with, and where is such information maintained (so one can avoid this pitfall in the future)?

The command used to build:
`make BOARD=GENERIC_SPIRAM USER_C_MODULES=/media/moefear/BACKUP/Projects/Programming/Micropython/Modules_Static/micropython.cmake && esptool.py -p /dev/ttyUSB1 -b 1152000 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x1000 build-GENERIC_SPIRAM/bootloader/bootloader.bin 0x8000 build-GENERIC_SPIRAM/partition_table/partition-table.bin 0x10000 build-GENERIC_SPIRAM/micropython.bin`

Any help is appreciated, as it is for the following project: [micropython_with_espnow_camera](https://github.com/moefear85/micropython_with_espnow_camera)

[sdkconfig.txt](https://github.com/micropython/micropython/files/12848877/sdkconfig.txt)
[code.zip](https://github.com/micropython/micropython/files/12848751/code.zip)
[crashdump.txt](https://github.com/micropython/micropython/files/12848771/crashdump.txt)

ned-pcs, 2023-10-09:
In `ports/esp32/README.md` it documents how to build Micropython for the ESP32. It says:
```
The ESP-IDF changes quickly and MicroPython only support certain versions.
Currently MicroPython supports only v5.0.2
```
I would try building your image with that version of the ESP-IDF, or using a pre-built image for your board.

projectgus, 2023-10-09:
Hi @moefear85,

>  If micropython is not compatible with the mentioned esp-idf version, which branch (or what latest commit) is it definitely compatible with, and where is such information maintained

In the `ports/esp32/README.md` there is some information about ESP-IDF versions. For the v1.20 release, it says:

> The ESP-IDF changes quickly and MicroPython only supports certain versions.
> Currently MicroPython supports v4.0.2, v4.1.1, v4.2.2, v4.3.2 and v4.4,
> although other IDF v4 versions may also work.

Note that `v4.4` here technically means the `v4.4` release tag (`git checkout v4.4`) not the "release branch" which is a pre-release commit snapshot.

I'm not 100% sure that this crash is caused by ESP-IDF version incompatibility, though.

---

Looking at `crashdump.txt`, this looks like a memory corruption bug or use-after-free.

The socket code is trying to lock a spinlock (thread stack shows that LWIP is calling `vPortEnterCritical`) and when doing that tries to read from address 0x7ffd56ca (the `vaddr` at the top). This isn't remotely close to a valid RAM ESP32 RAM region, and indicates most likely something else has come in an written over the memory which holds the spinlock.

From a very quick look at the Python code I don't see anything which looks too risky, memory-wise. "Timers" on the esp32 port don't run in interrupts they are "soft interrupts", so you can do things like socket operations from them (this is not guaranteed by MicroPython, but it's true in this case).

I see from your `make` line that you're compiling MicroPython with user C modules. I don't know what those do, but they have the potential to have corrupted memory in some way.

To narrow this down, I suggest downloading [ESP32_GENERIC-SPIRAM-20230426-v1.20.0.bin](https://micropython.org/resources/firmware/ESP32_GENERIC-SPIRAM-20230426-v1.20.0.bin) from [here](https://micropython.org/download/ESP32_GENERIC/), flash that, and see if you can reproduce the problem running that Python code. If you can, it's potentially a MicroPython bug (although v1.21 is out now so it may be fixed there). If you can't, it's something with the custom user modules or the ESP-IDF versions or something else in the custom build.

projectgus, 2023-10-09:
Oops @ned-pcs I didn't see your comment, posted while I was writing mine.

moefear85, 2023-10-09:
I'll check tomorrow, but I doubt it has anything to do with the User C modules, since as is obvious from the python code, none of the C modules are activated. They don't even do anything if imported. One would have to call specific initialization functions within them. I think there is a bug in the LWIP modules.

BTW, the release branches are not pre-release, (except for master). Rather they are running fixes to bugs reported in the issues. Nothing but the master branch gets any new features or anything that might break anything, at least according to espressif.

projectgus, 2023-10-09:
> BTW, the release branches are not pre-release, (except for master). Rather they are running fixes to bugs reported in the issues. Nothing but the master branch gets any new features or anything that might break anything, at least according to espressif.

They're bugfix pre-release branches ("beta quality or better"), refer to [this document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-to-a-release-branch). I wrote the first version of that document when I still worked at Espressif, which is why I'm sure. :grin:.

As I said in the previous reply, I don't think the ESP-IDF version is the most likely cause of this problem. Any commit taken from the release branch *probably* will be fine, but it's also possible that you've unluckily picked a commit that has some random regression in it and that's why this is crashing. Switching to a the stable release version that's supported by MicroPython removes this variable, so we don't have to worry about it.

> I think there is a bug in the LWIP modules, or something in mpy or esp-idf has overwritten the address. Could it be heap corruption? I'll try your suggestions as well as enabling heap poisioning detection etc. Could it be stack overflow?

It could be any of those things. Verifying if you can reproduce the problem with a release build of MicroPython that doesn't have custom C code compiled in, or possible other changes in the local build environment, will narrow things down substantially.

projectgus, 2023-10-09:
> esp-idf-v4.4 (release branch): b550086 cloned on 01-10-23

Ah so [b550086 is actually the merge commit for the v4.4.6 stable release](https://github.com/espressif/esp-idf/commit/b550086). So you basically have the ESP-IDF v4.4.6 stable release. That's different. However, I don't think anyone has specifically tested MicroPython with this version - trying with `v4.4` is probably worth doing to rule out a compatibility issue, anyway.

moefear85, 2023-10-10:
also the provided link says:

ˋA release branch is always beta quality or better, and receives bug fixes before they appear in each stable release.ˋ Moreover espressif themselves always insist that one try the release branch and master before filing a bug report because they consider it always likely that the bug has already been fixed there, which further implies that it is more stable than the stable release.

That's pretty much self-contradictory yet proves my point. The release branch has more fixes than the "stable" release it refers to in its name. Anyways I just woke up and intend to migrate to v1.21.0 & esp-idf-v5.0.2. I've missed the readme.md in the ports/esp32 folder but now I'll always check versions & stick to the stable commits.

projectgus, 2023-10-10:
> Moreover espressif themselves always insist that one try the release branch and master before filing a bug report because they consider it always likely that the bug has already been fixed there, which further implies that it is more stable than the stable release.

No, it implies that they don't want to fix the same bug twice. There will be bug fixes in those branches, but it's simultaneously possible for *regressions* to be introduced along with bug fixes. As much as we software developers would like every bug fix to be perfect, they aren't always perfect. Additional testing of a specific version is the difference between "beta quality or better" and "stable".

I'm interested to work with you and figure out if this is an issue in MicroPython, ESP-IDF, or a custom build issue. I know [quite a bit about ESP-IDF](https://github.com/espressif/esp-idf/graphs/contributors) and I'm a MicroPython maintainer, and I'm currently helping you in my spare time. However,  if you keep wasting both our time arguing about the meaning of a versioning scheme **that I myself helped create** then this will not happen.

Moving to v1.21.0 and the newer supported ESP-IDF seems like a great next move. If the bug still occurs then please also try running the same Python code on an official binary build for comparison.

>  I'll always check versions & stick to the stable commits

FWIW, I'm not saying you should always stick to the stable commits, or that it's never a good idea to update the ESP-IDF version used in MicroPython. I'm saying that if you have an unknown bug and you're not using matching supported versions of both MicroPython and ESP-IDF, then a good first step is to rule this out as a potential cause.

moefear85, 2023-10-11:
I see.... so

> No, it implies that they don't want to fix the same bug twice

Ahah... so the first bug fix would be.... in the "stable" commit? So your point is: The release branches are unstable in the sense that the "stable" commit means it is predictably unstable, whatever problems you are facing they are always the same cause, whereas with the release branches, you get the same amount of bugs, but in round-robin fashion with occasional sorting out of obsolete bugs and replacing them with fresh ones?

projectgus, 2023-10-11:
The [document I linked to two posts back](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html) explains the definition of "stable release" in the context of ESP-IDF. MicroPython is only tested with particular specific versions of the ESP-IDF.

If you want help finding the root cause of this bug, please perform some of the suggested steps to narrow it down and post the results. If you don't want to do that, please close this issue.

closed 2023-11-06 by dpgeorge